gazebosim/gz-gui #73: Add a QML code checker

chapulina, 2020-06-05:
When running `make codecheck` we run `cppcheck` for the C++ code. It would be nice to also check that our QML code is following some basic consistent style.

I haven't tried it, but maybe this tool is useful:

https://github.com/jesperhh/qmlfmt

ahcorde, 2021-01-14:
@chapulina, I was having a quick look to the `qmlfmt` tool. The down side it's that there is only binaries for MacOS. We should probably create our own binaries for Linux and Windows.

I also saw some weird indentation:

 - Original
```qml
    color: (Material.theme == Material.Light) ? Material.color(Material.Grey,Material.Shade200)
                                              : Material.color(Material.BlueGrey, 
```
 - Formatted
```qml
Material.Shade800)
    color: (Material.theme == Material.Light) ? Material.color(
                                                  Material.Grey,
                                                  Material.Shade200) : Material.color(
                                                  Material.BlueGrey,
                                                  Material.Shade800)
```

chapulina, 2021-01-14:
> I was having a quick look to the qmlfmt tool

Thanks!

> The down side it's that there is only binaries for MacOS. We should probably create our own binaries for Linux and Windows.

Since this would be a development tool, it would be ok not to distribute binaries. We could instruct users to use it from source and also use it from source on CI.

> I also saw some weird indentation:

Ouch yeah that looks weird. I'd be ok living with it to get the benefit of standardization and quick formatting though.