gazebosim/gz-gui #74: Minimize plugins

chapulina, 2020-06-10:
As we start having larger plugins, they start taking more and more real estate on the window. For example, the plotting tool (#66) and the resource spawner (https://github.com/ignitionrobotics/ign-gazebo/issues/31) will be too large to keep on the left panel, and too large to be overlapped on top of the 3D scene when undocked.

Some plugins could be launched on their own window using 'ign gui -s'. This should work for plotting if all the user needs are topics. But plugins like the spawner will need to be on the same process as the 3D scene. So we need a way to make large plugins more manageable.

The idea on this ticket is to let plugins be collapsed / minimized. This applies to both docked and undocked plugins.

The minimize / maximize buttons could be on the plugin toolbar next to dock / undock.

Sarath18, 2020-06-11:
[image]
Does this implementation meet a few of the ticket requirements?

I tried to add a button on the title bar to expand/collapse the plugin. I set the size of the plugin card rectangle to 50 (size of the title bar) to collapse the plugin. Is this correct? (feels kind of hacky way to do it)

chapulina, 2020-06-11:
Nice, that looks great! That sounds like a valid implementation to me. I believe it also works when the plugins are docked, right?

Looking at your gif, I also think it would make it super nice to apply a [NumberAnimation](https://doc.qt.io/qt-5/qml-qtquick-numberanimation.html) to the `height` when changing it, to make the transition smoother. See an example here:

https://github.com/ignitionrobotics/ign-gui/blob/75404e2ff46f4445a2f91b57f76033a0f6f6f2df/src/plugins/world_stats/WorldStats.qml#L123-L142

Sarath18, 2020-06-11:
> Nice, that looks great! That sounds like a valid implementation to me.

Great!

>  I believe it also works when the plugins are docked, right?

Not yet, will work on that.

> Looking at your gif, I also think it would make it super nice to apply a NumberAnimation to the height when changing it, to make the transition smoother. 

Yes, it is snappy at the moment. I'll try to add this feature as well.

closed 2020-08-01 by chapulina