gazebosim/gz-gui #122: levels.sdf trigger repeated entity errors

claireyywang, 2020-09-10:
ign-gui master branch, to reproduce, run `ign gazebo -v 4 levels.sdf`

console output
```
[GUI] [Err] [SceneManager.cc:129] Entity with Id: [38] already exists in the scene
[GUI] [Err] [SceneManager.cc:129] Entity with Id: [42] already exists in the scene
[GUI] [Err] [SceneManager.cc:129] Entity with Id: [64] already exists in the scene
[GUI] [Err] [SceneManager.cc:129] Entity with Id: [108] already exists in the scene
[GUI] [Err] [SceneManager.cc:129] Entity with Id: [130] already exists in the scene
[GUI] [Err] [SceneManager.cc:129] Entity with Id: [152] already exists in the scene
[GUI] [Err] [SceneManager.cc:129] Entity with Id: [174] already exists in the scene
[GUI] [Err] [SceneManager.cc:129] Entity with Id: [262] already exists in the scene
[GUI] [Err] [SceneManager.cc:129] Entity with Id: [284] already exists in the scene
[GUI] [Err] [SceneManager.cc:129] Entity with Id: [306] already exists in the scene
[GUI] [Err] [SceneManager.cc:129] Entity with Id: [328] already exists in the scene
[GUI] [Err] [SceneManager.cc:129] Entity with Id: [350] already exists in the scene
[GUI] [Err] [SceneManager.cc:129] Entity with Id: [372] already exists in the scene
[GUI] [Err] [SceneManager.cc:129] Entity with Id: [394] already exists in the scene
[GUI] [Err] [SceneManager.cc:129] Entity with Id: [438] already exists in the scene
[GUI] [Err] [SceneManager.cc:129] Entity with Id: [460] already exists in the scene
[GUI] [Err] [SceneManager.cc:181] Entity with Id: [29] already exists in the scene
[GUI] [Err] [SceneManager.cc:181] Entity with Id: [32] already exists in the scene
[GUI] [Err] [SceneManager.cc:181] Entity with Id: [39] already exists in the scene
[GUI] [Err] [SceneManager.cc:181] Entity with Id: [43] already exists in the scene
[GUI] [Err] [SceneManager.cc:181] Entity with Id: [65] already exists in the scene
[GUI] [Err] [SceneManager.cc:181] Entity with Id: [109] already exists in the scene
[GUI] [Err] [SceneManager.cc:181] Entity with Id: [131] already exists in the scene
[GUI] [Err] [SceneManager.cc:181] Entity with Id: [153] already exists in the scene
[GUI] [Err] [SceneManager.cc:181] Entity with Id: [175] already exists in the scene
[GUI] [Err] [SceneManager.cc:181] Entity with Id: [263] already exists in the scene
[GUI] [Err] [SceneManager.cc:181] Entity with Id: [285] already exists in the scene
[GUI] [Err] [SceneManager.cc:181] Entity with Id: [307] already exists in the scene
[GUI] [Err] [SceneManager.cc:181] Entity with Id: [329] already exists in the scene
[GUI] [Err] [SceneManager.cc:181] Entity with Id: [351] already exists in the scene
[GUI] [Err] [SceneManager.cc:181] Entity with Id: [373] already exists in the scene
[GUI] [Err] [SceneManager.cc:181] Entity with Id: [395] already exists in the scene
[GUI] [Err] [SceneManager.cc:181] Entity with Id: [439] already exists in the scene
[GUI] [Err] [SceneManager.cc:222] Entity with Id: [30] already exists in the scene
...
```

chapulina, 2020-09-10:
I've also been seeing these a lot lately, I think it's related to timing.

chapulina, 2021-11-15:
I can't reproduce this anymore

closed 2021-11-15 by chapulina