gazebosim/gz-gui #123: ignition-gui4 homebrew tests seg-faulting

scpeters, 2020-09-11:
There were 4 tests (`UNIT_Application_TEST`, `UNIT_Helpers_TEST`, `UNIT_MainWindow_TEST`, `UNIT_Plugin_TEST`) that failed to run in a recent home-brew Jenkins CI job of revision 40a4658bb7933f76bd90544401693acba1c19f6c on the master branch:

* https://build.osrfoundation.org/job/ignition_gui-ci-master-homebrew-amd64/25/testReport/

The console output show seg-faults:

~~~
1: [----------] 7 tests from ApplicationTest
1: [ RUN      ] ApplicationTest.Constructor
 1/30 Test  #1: UNIT_Application_TEST ..................***Exception: SegFault  0.03 sec
~~~

~~~
7: [----------] 6 tests from HelpersTest
7: [ RUN      ] HelpersTest.humanReadable
7: [       OK ] HelpersTest.humanReadable (0 ms)
7: [ RUN      ] HelpersTest.unitFromKey
7: [       OK ] HelpersTest.unitFromKey (0 ms)
7: [ RUN      ] HelpersTest.rangeFromKey
7: [       OK ] HelpersTest.rangeFromKey (0 ms)
7: [ RUN      ] HelpersTest.stringTypeFromKey
7: [       OK ] HelpersTest.stringTypeFromKey (0 ms)
7: [ RUN      ] HelpersTest.findFirstByProperty
 7/30 Test  #7: UNIT_Helpers_TEST ......................***Exception: SegFault  0.03 sec
~~~

~~~
11: [----------] 4 tests from MainWindowTest
11: [ RUN      ] MainWindowTest.Constructor
11/30 Test #11: UNIT_MainWindow_TEST ...................***Exception: SegFault  0.03 sec
~~~

~~~
15: [----------] 3 tests from PluginTest
15: [ RUN      ] PluginTest.DeleteLater
15/30 Test #15: UNIT_Plugin_TEST .......................***Exception: SegFault  0.03 sec
~~~

scpeters, 2020-09-11:
It seems to work when I use `lldb`, but I was able to get a backtrace by setting `ulimit -c unlimited` and then running without `lldb`. Here's the backtrace from `UNIT_Helpers_TEST` with relevant lines at [Helpers_TEST.cc:125](https://github.com/ignitionrobotics/ign-gui/blob/40a4658bb7933f76bd90544401693acba1c19f6c/src/Helpers_TEST.cc#L125) and [Application.cc:85:5](https://github.com/ignitionrobotics/ign-gui/blob/40a4658bb7933f76bd90544401693acba1c19f6c/src/Application.cc#L85):

~~~
$ bin/UNIT_Helpers_TEST 
Running main() from /Users/scpeters/ws/ignition/src/ign-gui/test/gtest/src/gtest_main.cc
[==========] Running 6 tests from 1 test case.
[----------] Global test environment set-up.
[----------] 6 tests from HelpersTest
[ RUN      ] HelpersTest.humanReadable
[       OK ] HelpersTest.humanReadable (0 ms)
[ RUN      ] HelpersTest.unitFromKey
[       OK ] HelpersTest.unitFromKey (0 ms)
[ RUN      ] HelpersTest.rangeFromKey
[       OK ] HelpersTest.rangeFromKey (0 ms)
[ RUN      ] HelpersTest.stringTypeFromKey
[       OK ] HelpersTest.stringTypeFromKey (0 ms)
[ RUN      ] HelpersTest.findFirstByProperty
Segmentation fault: 11 (core dumped)
$ lldb -c /cores/core.2
core.27906  core.27930  core.27933  core.28216  core.28484  
OSRF-macbook-pro-15:build scpeters$ lldb -c /cores/core.28484 
(lldb) target create --core "/cores/core.28484"
Core file '/cores/core.28484' (x86_64) was loaded.
(lldb) bt
* thread #1, stop reason = signal SIGSTOP
  * frame #0: 0x00007fff728a6fce libsystem_c.dylib`strrchr + 10
    frame #1: 0x000000010dacafeb QtCore`QCoreApplicationPrivate::appName() const + 203
    frame #2: 0x000000010dacd395 QtCore`QCoreApplicationPrivate::init() + 101
    frame #3: 0x000000010d302f79 QtGui`QGuiApplicationPrivate::init() + 57
    frame #4: 0x000000010cd29e1a QtWidgets`QApplicationPrivate::init() + 26
    frame #5: 0x000000010bb52c72 libignition-gui4.4.dylib`ignition::gui::Application::Application(this=0x00007ffee41426b8, _argc=<unavailable>, _argv=<unavailable>, _type=kMainWindow) at Application.cc:85:5 [opt]
    frame #6: 0x000000010baca0e0 UNIT_Helpers_TEST`HelpersTest_findFirstByProperty_Test::TestBody(this=<unavailable>) at Helpers_TEST.cc:125:15 [opt]
    frame #7: 0x000000010bade188 UNIT_Helpers_TEST`void testing::internal::HandleExceptionsInMethodIfSupported<testing::Test, void>(testing::Test*, void (testing::Test::*)(), char const*) [inlined] void testing::internal::HandleSehExceptionsInMethodIfSupported<testing::Test, void>(method=<unavailable>, location=<unavailable>)(), char const*) at gtest.cc:2447:10 [opt]
    frame #8: 0x000000010bade179 UNIT_Helpers_TEST`void testing::internal::HandleExceptionsInMethodIfSupported<testing::Test, void>(object=<unavailable>, method=<unavailable>, location="the test body")(), char const*) at gtest.cc:2483 [opt]
    frame #9: 0x000000010bade081 UNIT_Helpers_TEST`testing::Test::Run(this=0x00007fd473d27fb0) at gtest.cc:2521:5 [opt]
    frame #10: 0x000000010badf49c UNIT_Helpers_TEST`testing::TestInfo::Run(this=0x00007fd473d2ace0) at gtest.cc:2697:11 [opt]
    frame #11: 0x000000010badfe67 UNIT_Helpers_TEST`testing::TestCase::Run(this=0x00007fd473d2a820) at gtest.cc:2815:28 [opt]
    frame #12: 0x000000010baecb67 UNIT_Helpers_TEST`testing::internal::UnitTestImpl::RunAllTests(this=<unavailable>) at gtest.cc:5181:43 [opt]
    frame #13: 0x000000010baec618 UNIT_Helpers_TEST`bool testing::internal::HandleExceptionsInMethodIfSupported<testing::internal::UnitTestImpl, bool>(testing::internal::UnitTestImpl*, bool (testing::internal::UnitTestImpl::*)(), char const*) [inlined] bool testing::internal::HandleSehExceptionsInMethodIfSupported<testing::internal::UnitTestImpl, bool>(method=<unavailable>, location=<unavailable>)(), char const*) at gtest.cc:2447:10 [opt]
    frame #14: 0x000000010baec609 UNIT_Helpers_TEST`bool testing::internal::HandleExceptionsInMethodIfSupported<testing::internal::UnitTestImpl, bool>(object=<unavailable>, method=<unavailable>, location="auxiliary test code (environments or event listeners)")(), char const*) at gtest.cc:2483 [opt]
    frame #15: 0x000000010baec58c UNIT_Helpers_TEST`testing::UnitTest::Run(this=0x000000010bb0b8e8) at gtest.cc:4790:10 [opt]
    frame #16: 0x000000010baff6da UNIT_Helpers_TEST`main [inlined] RUN_ALL_TESTS() at gtest.h:2341:46 [opt]
    frame #17: 0x000000010baff6cd UNIT_Helpers_TEST`main(argc=1, argv=<unavailable>) at gtest_main.cc:36 [opt]
    frame #18: 0x00007fff728043d5 libdyld.dylib`start + 1
(lldb) 
~~~

chapulina, 2020-11-04:
These tests seem to be fixed, but could be slightly flaky:

* `UNIT_Helpers_TEST`: https://build.osrfoundation.org/job/ignition_gui-ci-ign-gui4-homebrew-amd64/lastCompletedBuild/testReport/(root)/HelpersTest/history/
* `UNIT_MainWindow_TEST`: https://build.osrfoundation.org/job/ignition_gui-ci-ign-gui4-homebrew-amd64/lastCompletedBuild/testReport/(root)/MainWindowTest/

The other tests seem to be failing at the `Application`'s constructor. The failure happens from `ign-gui4` onwards, but not on `ign-gui2` and `ign-gui3`:

* https://build.osrfoundation.org/job/ignition_gui-ci-ign-gui2-homebrew-amd64/43/testReport/(root)/ApplicationTest/Constructor/history/
* https://build.osrfoundation.org/job/ignition_gui-ci-ign-gui3-homebrew-amd64/27/testReport/(root)/ApplicationTest/Constructor/history/

The main change starting from `ign-gui4` is that `Application` inherits from `QApplication` instead of `QGuiApplication`. It's still strange to me why the application construction is not failing on other tests though.

jennuine, 2024-01-05:
Closing since dome is EOL.

closed 2024-01-05 by jennuine