gazebosim/gz-gui #131: Floating gui plugins don't respect minimum dimensions

JShep1, 2020-09-29:
Gui Plugins respect the layout sizes (such as [here](https://github.com/ignitionrobotics/ign-gui/blob/master/src/plugins/publisher/Publisher.qml#L24-L25) ) when they are docked (like within the right panel of the main GUI window), but these dimensions aren't respected in other scenarios, such as loading the plugin directly with `ign gui -s Publisher`, in this case, the window loads as a small box and it is up to the user to manually expand the window to the correct dimensions.

Sarath18, 2020-09-30:
I was going through the source code to get more insight into standalone GUI plugins. From what I understand, the plugin is attached to a new Dialog to create standalone plugins. The height and the width of the cardItem are being set after loading the plugin and then the dimensions of cardItem are used to set the size of the dialog. 

> the window loads as a small box and it is up to the user to manually expand the window to the correct dimensions.

This is happening because, in plugins, the minimum height and width of Layout are set instead of the plugin (for eg. TopicEcho, Publisher). So the height and width of the plugin fetched are 0.

Also, when the dialog is set as the parent item (before the dialog is resized), the cardItem's height and width property become 0.

Potential fix:
1. Set parent item of cardItem after the dialog is resized
2. Specify the height and width of each plugin
   - Manually by defining width and height (same as `Layout.minimumHeight` and `Layout.minimumWidth`)
   - Using the `Component.onCompleted` method. This way we need not maintain 2 variables for each, height and width.
   - Directly fetch the layout minimum width and height (I haven't found a way to do this) of the plugin if the height and width of the plugin are 0.

I would like to take up this issue if I am on the right path. Please correct me if I am wrong and there exits a better fix for standalone plugins.

chapulina, 2020-09-30:
Thanks for the analysis, @Sarath18 . I think you're on the right path.

I think potential fix 1 is good.

I think potential fix 2.a is inconvenient, but I think it would work for most cases. My worry with `onCompleted` is that it adds complexity. Maybe we could expose variables in `IgnCard` like `cardMinimumWidth` that set both `Layout.minimumWidth` and `width`?

Did you take a look at what would be needed to fix the floating plugins?

Sarath18, 2020-09-30:
Just to clarify, 1 and 2 are the two steps in the potential fix.
> potential fix 2.a is inconvenient

Agreed.
> My worry with onCompleted is that it adds complexity.

I am not a 100% sure what you mean by this. Isn't it just this block
```qml
  Component.onCompleted: {
    topicEcho.height = Layout.minimumHeight
    topicEcho.width = Layout.minimumWidth
  }
```

> Maybe we could expose variables in IgnCard like cardMinimumWidth that set both Layout.minimumWidth and width?

I need to look into this before I can comment on it.

> Did you take a look at what would be needed to fix the floating plugins?

I was mostly targeting standalone plugins so I didn't take a look into floating plugins. But I think we only need to ensure the minimumWidth as height is being changed for collapsing/expanding the plugin. On undock, the minimum sizes are respected for floating plugins. Since we are passing the IgnCard to IgnRuler as target, we might be able to control resizing by checking the minimumWidth on the drag callback.

chapulina, 2020-09-30:
> Isn't it just this block?

Oh, do you mean putting that block in `IgnCard`? Ok yeah I think that could work. I thought it would have to be in every plugin.

> we might be able to control resizing by checking the minimumWidth on the drag callback

+1

chapulina, 2020-10-02:
Closed by #135

closed 2020-10-02 by chapulina