gazebosim/gz-gui #137: Consolidate Scene3D with ign-gazebo's GzScene3D

chapulina, 2020-10-06:
We currently have 2 separate scene plugins in the Ignition ecosystem, one in `ign-gui` and one in `ign-gazebo`. The split into 2 scene plugins was something we did to move forward quickly, but the long-term goal is to combine those into a single plugin in a way that all Gazebo-independent features can be pushed into `ign-gui` so all downstream projects can take advantage of it and there's less duplication to maintain.

Some possible ways to achieve this:

* Move all Gazebo-specific features to separate plugins that interact with the 3D scene
* Make the `ign-gui` plugin extensible, similar to how we have [ignition::gui::TransportPlotting](https://github.com/ignitionrobotics/ign-gui/tree/main/src/plugins/plotting) and [ignition::gazebo::Plotting](https://github.com/ignitionrobotics/ign-gazebo/tree/main/src/gui/plugins/plotting).

scpeters, 2021-03-11:
we are using the `Scene3D` plugin in the delphyne project and have noticed some errors when loading mesh materials, but I believe this has already been fixed in ign-gazebo, so I'm cross-porting the fix from ign-gazebo's SceneManager in #191

ahcorde, 2021-04-29:
Related PRs

- MoveToHelper (ign-rendering) https://github.com/ignitionrobotics/ign-rendering/pull/311
- New Scene3D interface https://github.com/ignitionrobotics/ign-gui/pull/217
- New Scene3D plugin using the interface provided in ign-gui https://github.com/ignitionrobotics/ign-gazebo/pull/798

ahcorde, 2021-06-18:
@chapulina, I have been working in adding small plugins to ign-gazebo and ign-gui. I updated the ign-gazebo PR with some of the plugins.

 - ign-gui
   - view_control  https://github.com/ignitionrobotics/ign-gui/pull/231/
   - Move to / follow  https://github.com/ignitionrobotics/ign-gui/pull/226
 
- ign-gazebo
  - Select https://github.com/ignitionrobotics/ign-gazebo/pull/854
  - Transform controls https://github.com/ignitionrobotics/ign-gazebo/pull/854

ahcorde, 2021-06-18:
I have created some branches to test this (Fortress). Because it requires some cherry picks:

  - ign-gazebo https://github.com/ignitionrobotics/ign-gazebo/tree/ahcorde/6/scene_manager_test
  - ign-gui https://github.com/ignitionrobotics/ign-gui/tree/ahcorde/6/new_scene_test
  - ign-common https://github.com/ignitionrobotics/ign-common/tree/ahcorde/3_to_4

Run the demo:

```
ign gazebo --gui-config src/ign-gazebo/src/gui/gui.config shapes.sdf -v 4
```

chapulina, 2021-06-18:
> some branches to test this 

Nice, I see the 3D scene is updated! 

I assume all the plugins above should work with this combination of branches?

>  view_control #231

The `ViewControl` plugin is empty for me, no buttons

>  Move to / follow #226

Right-clicking the 3D scene doesn't give me a context menu to be able to choose these options.

Choosing them from the entity tree doesn't seem to work.

>   Select ignitionrobotics/ign-gazebo#854

I can select entities on the 3D scene and they get highlighted, but I don't see them highlighted on the entity tree. The same for the other direction.

> Transform controls ignitionrobotics/ign-gazebo#854

They work if I:

1. Deselect all entities
2. Choose translate mode
3. Select entity

But don't work if I do:

1. Select entity
2. Choose translate mode

ahcorde, 2021-06-21:
> The ViewControl plugin is empty for me, no buttons

I called `view_control` but it's the plugin to move the camera with the mouse. 

> Right-clicking the 3D scene doesn't give me a context menu to be able to choose these options.

The Scene3D it's on ign-gui, the context menu is part of ignition-gazebo, I didn't find a way (yet) to modify the qml of the scene3d from ign-gazebo and include the.

chapulina, 2021-07-06:
Outstanding issue:

- [ ] How to choose what camera plugins are being attached to, see https://github.com/ignitionrobotics/ign-gui/pull/231#discussion_r658353941)

chapulina, 2021-09-08:
Main PRs splitting the functionality:

* https://github.com/ignitionrobotics/ign-gui/pull/221
* https://github.com/ignitionrobotics/ign-gazebo/pull/813
* https://github.com/ignitionrobotics/ign-gui/pull/226
* https://github.com/ignitionrobotics/ign-gui/pull/231
* https://github.com/ignitionrobotics/ign-gui/pull/247
* https://github.com/ignitionrobotics/ign-gazebo/pull/854
* https://github.com/ignitionrobotics/ign-gazebo/pull/988
* https://github.com/ignitionrobotics/ign-gazebo/pull/900
* https://github.com/ignitionrobotics/ign-gazebo/pull/983
* https://github.com/ignitionrobotics/ign-gazebo/pull/1016
* https://github.com/ignitionrobotics/ign-gazebo/pull/1014

### TODO

- [ ] Drag & drop from Fuel / meshes
- [ ] Context menu
- [ ] Port rendering fixes from `GzScene3D` to `MinimalScene`, such as https://github.com/ignitionrobotics/ign-gazebo/pull/774
- [ ] Choose a render engine from the command line (this information needs to be sent to `MinimalScene` before it creates a scene.)

ahcorde, 2021-09-16:
Drag & drop
  - ign-gui PR https://github.com/ignitionrobotics/ign-gui/pull/282
  - Still a branch https://github.com/ignitionrobotics/ign-gazebo/tree/ahcorde/plugin%2Fdrop_from_fuel

 Context Menu
  - Added in this draft PR https://github.com/ignitionrobotics/ign-gazebo/pull/1014
 
Port rendering fixes from GzScene3D to MinimalScene
 - Added in this draft PR  https://github.com/ignitionrobotics/ign-gazebo/pull/1014

ahcorde, 2021-09-17:
Choose a render engine from the command line (this information needs to be sent to MinimalScene before it creates a scene.)
 - https://github.com/ignitionrobotics/ign-gazebo/pull/1034
 - https://github.com/ignitionrobotics/ign-gui/pull/286

ahcorde, 2021-09-22:
@chapulina can we consider this as done ?

chapulina, 2021-09-22:
Yes! :tada:

closed 2021-09-22 by chapulina