gazebosim/gz-gui #145: 3D scene failing to load on macOS

chapulina, 2020-11-05:
Ignition GUI builds on macOS, and plugins that don't involve rendering are loaded properly. But the `Scene3D` plugin, which uses Ignition Rendering, fails with both Ogre 1 and Ogre 1.

`ign gui -s Scene3D`

ahcorde, 2020-12-10:
I investigated this issue a little bit. Some of issues that I found:

 - This segfault https://github.com/ignitionrobotics/ign-gui/pull/155

I have been trying to create a simple QML + OGRE simple example, but I didn't have success until this moment. 

 - One of the failures that I found during this process, it's that `windowMovedOrResized()` is generating a segfault too.

I will update the simple demo to my Github account when it's ready to be tested by others.

----

ign gui -s Scene3D

```
	0   CoreFoundation                      0x00007fff204ae6af __exceptionPreprocess + 242
	1   libobjc.A.dylib                     0x00007fff201e63c9 objc_exception_throw + 48
	2   CoreFoundation                      0x00007fff204d6a9a -[NSException raise] + 9
	3   AppKit                              0x00007fff22c7697d -[NSWindow(NSWindow_Theme) _postWindowNeedsToResetDragMarginsUnlessPostingDisabled] + 321
	4   AppKit                              0x00007fff22c616d5 -[NSWindow _initContent:styleMask:backing:defer:contentView:] + 1375
	5   AppKit                              0x00007fff22c6116f -[NSWindow initWithContentRect:styleMask:backing:defer:] + 42
	6   RenderSystem_GL.1.9.0.dylib         0x000000014b08d572 _ZN4Ogre14OSXCocoaWindow15createNewWindowEjjNSt3__112basic_stringIcNS1_11char_traitsIcEENS1_9allocatorIcEEEE + 262
	7   RenderSystem_GL.1.9.0.dylib         0x000000014b08cd99 _ZN4Ogre14OSXCocoaWindow6createERKNSt3__112basic_stringIcNS1_11char_traitsIcEENS1_9allocatorIcEEEEjjbPKNS1_3mapIS7_S7_NS1_4lessIS7_EENS_12STLAllocatorINS1_4pairIS8_S7_EENS_22CategorisedAllocPolicyILNS_14MemoryCategoryE0EEEEEEE + 2391
	8   RenderSystem_GL.1.9.0.dylib         0x000000014b0928c0 _ZN4Ogre12OSXGLSupport9newWindowERKNSt3__112basic_stringIcNS1_11char_traitsIcEENS1_9allocatorIcEEEEjjbPKNS1_3mapIS7_S7_NS1_4lessIS7_EENS_12STLAllocatorINS1_4pairIS8_S7_EENS_22CategorisedAllocPolicyILNS_14MemoryCategoryE0EEEEEEE + 500
	9   RenderSystem_GL.1.9.0.dylib         0x000000014b0a45a9 _ZN4Ogre14GLRenderSystem19_createRenderWindowERKNSt3__112basic_stringIcNS1_11char_traitsIcEENS1_9allocatorIcEEEEjjbPKNS1_3mapIS7_S7_NS1_4lessIS7_EENS_12STLAllocatorINS1_4pairIS8_S7_EENS_22CategorisedAllocPolicyILNS_14MemoryCategoryE0EEEEEEE + 769
	10  libOgreMain.1.9.0.dylib             0x00000001193dfdca _ZN4Ogre4Root18createRenderWindowERKNSt3__112basic_stringIcNS1_11char_traitsIcEENS1_9allocatorIcEEEEjjbPKNS1_3mapIS7_S7_NS1_4lessIS7_EENS_12STLAllocatorINS1_4pairIS8_S7_EENS_22CategorisedAllocPolicyILNS_14MemoryCategoryE0EEEEEEE + 36
	11  libignition-rendering5-ogre.5.0.0~p 0x0000000117ed65cc _ZN8ignition9rendering2v516OgreRenderEngine18CreateRenderWindowERKNSt3__112basic_stringIcNS3_11char_traitsIcEENS3_9allocatorIcEEEEjjdj + 1484
	12  libignition-rendering5-ogre.5.0.0~p 0x0000000117ed4be6 _ZN8ignition9rendering2v516OgreRenderEngine18CreateRenderWindowEv + 86
	13  libignition-rendering5-ogre.5.0.0~p 0x0000000117ed2bb0 _ZN8ignition9rendering2v516OgreRenderEngine11LoadAttemptEv + 176
	14  libignition-rendering5-ogre.5.0.0~p 0x0000000117ed28d1 _ZN8ignition9rendering2v516OgreRenderEngine8LoadImplERKNSt3__13mapINS3_12basic_stringIcNS3_11char_traitsIcEENS3_9allocatorIcEEEESA_NS3_4lessISA_EENS8_INS3_4pairIKSA_SA_EEEEEE + 305
	15  libignition-rendering5.5.0.0~pre1.d 0x0000000118fa2cd0 _ZN8ignition9rendering2v516BaseRenderEngine4LoadERKNSt3__13mapINS3_12basic_stringIcNS3_11char_traitsIcEENS3_9allocatorIcEEEESA_NS3_4lessISA_EENS8_INS3_4pairIKSA_SA_EEEEEE + 192
	16  libignition-rendering5.5.0.0~pre1.d 0x0000000118f58b33 _ZN8ignition9rendering26RenderEngineManagerPrivate6EngineENSt3__114__map_iteratorINS2_15__tree_iteratorINS2_12__value_typeINS2_12basic_stringIcNS2_11char_traitsIcEENS2_9allocatorIcEEEEPNS0_2v512RenderEngineEEEPNS2_11__tree_nodeISF_PvEElEEEERKNS2_3mapISB_SB_NS2_4lessISB_EENS9_INS2_4pairIKSB_SB_EEEEEERSQ_ + 595
	17  libignition-rendering5.5.0.0~pre1.d 0x0000000118f587e2 _ZN8ignition9rendering2v519RenderEngineManager6EngineERKNSt3__112basic_stringIcNS3_11char_traitsIcEENS3_9allocatorIcEEEERKNS3_3mapIS9_S9_NS3_4lessIS9_EENS7_INS3_4pairISA_S9_EEEEEESB_ + 434
	18  libignition-rendering5.5.0.0~pre1.d 0x0000000118f66b4d _ZN8ignition9rendering2v56engineERKNSt3__112basic_stringIcNS2_11char_traitsIcEENS2_9allocatorIcEEEERKNS2_3mapIS8_S8_NS2_4lessIS8_EENS6_INS2_4pairIS9_S8_EEEEEESA_ + 45
	19  libScene3D.dylib                    0x0000000117912bcf _ZN8ignition3gui7plugins11IgnRenderer10InitializeEv + 191
	20  libScene3D.dylib                    0x0000000117914027 _ZN8ignition3gui7plugins12RenderThread10RenderNextEv + 119
	21  libScene3D.dylib                    0x000000011797a3ee _ZN8ignition3gui7plugins12RenderThread18qt_static_metacallEP7QObjectN11QMetaObject4CallEiPPv + 126
	22  QtCore                              0x000000010d5a3144 _ZN7QObject5eventEP6QEvent + 900
	23  QtWidgets                           0x000000010a938f56 _ZN19QApplicationPrivate13notify_helperEP7QObjectP6QEvent + 262
	24  QtWidgets                           0x000000010a93a302 _ZN12QApplication6notifyEP7QObjectP6QEvent + 466
	25  QtCore                              0x000000010d579de6 _ZN16QCoreApplication15notifyInternal2EP7QObjectP6QEvent + 166
	26  QtCore                              0x000000010d57af23 _ZN23QCoreApplicationPrivate16sendPostedEventsEP7QObjectiP11QThreadData + 803
	27  QtCore                              0x000000010d5d9459 _ZN20QEventDispatcherUNIX13processEventsE6QFlagsIN10QEventLoop17ProcessEventsFlagEE + 73
	28  QtCore                              0x000000010d5761e7 _ZN10QEventLoop4execE6QFlagsINS_17ProcessEventsFlagEE + 471
	29  QtCore                              0x000000010d3b48c3 _ZN7QThread4execEv + 131
	30  QtCore                              0x000000010d3b57d9 _ZN14QThreadPrivate5startEPv + 329
	31  libsystem_pthread.dylib             0x00007fff2033c950 _pthread_start + 224
	32  libsystem_pthread.dylib             0x00007fff2033847b thread_start + 15
```

chapulina, 2020-12-14:
Thanks for the update. I see you're using Ogre1. Just curious if you're running into the same exact issue with Ogre2.

ahcorde, 2020-12-15:
I can't execute Ogre2 in MacOS because of the following error:

```
objc[37280]: Class OgreConfigWindowDelegate is implemented in both /usr/local/opt/ogre1.9/lib/libOgreMain.1.9.0.dylib (0x14b388dc0) and /usr/local/opt/ogre2.1/lib/libOgreMain.2.1.0.dylib (0x1539eb838). One of the two will be used. Which one is undefined.
[GUI] [Err] [Ogre2RenderEngine.cc:436] Unable to find Ogre Plugin[/usr/local/opt/ogre2.1/lib/OGRE-2.1/RenderSystem_GL3Plus]. Rendering will not be possible.Make sure you have installed OGRE properly.
```

Anyhow If I hacked my system I got the same error

chapulina, 2020-12-15:
Oush I see, mind ticketing an issue for this on `ign-rendering`? It may need to be fixed there or on https://github.com/ignition-forks/ogre-2.1-release.

I don't see anything about that on `ign-rendering`: https://github.com/ignitionrobotics/ign-rendering/issues?q=is%3Aissue+is%3Aopen+label%3AmacOS

chapulina, 2021-01-05:
Getting the 3D scene to work on macOS has proved to demand more effort than we can put into this for now. So Open Robotics is not planning to work on this in the near future, but we welcome contributions from the community in case someone is interested in that platform.

srmainwaring, 2021-02-11:
I am seeing the same problem on macOS 10.15.4 in `ignition-gui`. I've restricted a build of `ign-rendering` to `ogre1.9` to avoid the issue mentioned above. Interestingly most of the `ignition-rendering` examples work:

- actor_animation: FAIL (segfault)
- camera_tracking: OK
- custom_scene_viewer: FAIL - no window
- custom_shaders: OK (files produced, no window)
- gazebo_scene_viewer: FAIL - no window
- lidar_visual: OK
- mesh_viewer: OK
- mouse_picking: OK
- ogre2_demo: Not tested
- particles_demo: FAIL (runs with ogre, needs ogre2 for particles)
- render_pass: OK
- simple_demo: OK
- text_geom: OK
- thermal_camera: OK (runs with ogre if mod, default needs ogre2)
- transform_control: OK
- view_control: OK

I'll continue to look into this to understand why the rendering is not working in this package first for `ogre1.9`, and if I make any headway `ogre2.1`. I'd really like to use the physics engine plugin features and not having a the UI on macOS is a stopper for that.

In the meanwhile it might be worth placing a note on the ignition installation page that macOS is not fully supported at this stage, as it gives the impression that it is.

chapulina, 2021-02-12:
> I'll continue to look into this to understand why the rendering is not working 

Thanks!

> it might be worth placing a note on the ignition installation page that macOS is not fully supported at this stage, as it gives the impression that it is.

We have an issue for that, planning to tackle it soon: https://github.com/ignitionrobotics/docs/issues/116

srmainwaring, 2021-02-15:
@chapulina  It looks like there is an initialisation ordering issue in `Scene3D.cc`:

In the following we only build for ogre (1) so the potential conflict between ogre and ogre2 is not present.

System: macOS 10.15.7
Build Type: RelWithDebInfo
Component: ign-gui
Example: examples/standalone/window

Start by replicating the issue using the command line tools to load the Scene3D plugin:

```bash
$ ign gui -v -c examples/config/scene3d.config
...
[GUI] [Msg] Loading config [examples/config/scene3d.config]
[GUI] [Msg] Added plugin [View 1] to main window
[GUI] [Msg] Loaded plugin [Scene3D] from path [/usr/local/lib/ign-gui-4/plugins/libScene3D.dylib]
[GUI] [Msg] Loading plugin [ignition-rendering-ogre]
[GUI] [Err] [OgreRenderEngine.cc:290] Failed to load render-engine
[GUI] [Err] [BaseRenderEngine.cc:55] Render-engine must be loaded first
[GUI] [Err] [BaseRenderEngine.cc:215] Render-engine has not been initialized
[BUG] Segmentation fault at 0x0000000000000000
ruby 3.0.0p0 (2020-12-25 revision 95aff21468) [x86_64-darwin19]
...
(Stack trace)
```

Next we try to replicated the error using the window standalone example modified to load the "Scene3D" plugin:

```diff
- Load plugins / config
- if (!app.LoadPlugin("Publisher"))
- {
-     return 1;
- }
+ if (!app.LoadPlugin("Scene3D"))
+ {
+     return 1;
+ }
```

Running the example yields the same type of error:

```bash
% ./window
Hello, GUI!
[Dbg] [Application.cc:87] Initializing application.
[GUI] [Dbg] [Application.cc:407] Create main window

... (various Qt font and deprecated behaviour warnings)

[GUI] [Dbg] [Application.cc:305] Loading plugin [Scene3D]
[GUI] [Msg] Added plugin [3D Scene] to main window
[GUI] [Msg] Loaded plugin [Scene3D] from path [/usr/local/lib/ign-gui-4/plugins/libScene3D.dylib]
[GUI] [Msg] Loading plugin [ignition-rendering-ogre]
[GUI] [Err] [OgreRenderEngine.cc:290] Failed to load render-engine
[GUI] [Err] [BaseRenderEngine.cc:55] Render-engine must be loaded first
[GUI] [Dbg] [Scene3D.cc:941] Create scene [scene]
[GUI] [Err] [BaseRenderEngine.cc:215] Render-engine has not been initialized
zsh: segmentation fault  ./window
```

Stepping through the example we find:

- Scene is loaded before the rendering engine is initialised. 
- Attempt to deref nullptr in  `Scene3D.cc` line 943

Details:

In the initialise function the engine is used to create the scene and then set lights etc.

```c++
/////////////////////////////////////////////////
void IgnRenderer::Initialize()
{
  if (this->initialized)
    return;

  std::map<std::string, std::string> params;
  params["useCurrentGLContext"] = "1";
  auto engine = rendering::engine(this->engineName, params);
  if (!engine)
  {
    ignerr << "Engine [" << this->engineName << "] is not supported"
           << std::endl;
    return;
  }

  // Scene
  auto scene = engine->SceneByName(this->sceneName);
  if (!scene)
  {
    igndbg << "Create scene [" << this->sceneName << "]" << std::endl;
    scene = engine->CreateScene(this->sceneName);
    scene->SetAmbientLight(this->ambientLight);
    scene->SetBackgroundColor(this->backgroundColor);
  }

  ...
}
```

However we see that that `BaseRenderEngine::CreateScene` will return a `nullptr`
because the engine is not initialised

```c++
ScenePtr BaseRenderEngine::CreateScene(const std::string &_name)
{
  unsigned int sceneId = this->NextSceneId();
  return this->CreateScene(sceneId, _name);
}
```

```c++
//////////////////////////////////////////////////
ScenePtr BaseRenderEngine::CreateScene(unsigned int _id,
    const std::string &_name)
{
  if (!this->IsInitialized())
  {
    ignerr << "Render-engine has not been initialized" << std::endl;
    return nullptr;
  }

  ...
}
```

This causes the plugin to fail.

I haven't confirmed this is exactly what is going on with the command line tools as I'm still trying to figure out where the entry point for those are, but it seems like the culprit. Not clear why this is only a macOS issue though?

chapulina, 2021-03-29:
> Attempt to deref nullptr in Scene3D.cc line 943

Nice catch, I think we should check the pointer there.

> Not clear why this is only a macOS issue though?

As you mention later, the issue is earlier in the code path. So it doesn't manifest for Linux because the engine is properly loaded, but it manifests for macOS because that fails to load.

> If all exceptions are caught then the function will exit reporting an error, but crucially OgreRenderEngine::LoadImpl marks the engine as loaded as it is relying on the same exceptions caught lower in the stack to determine success.

I'm not sure I follow, so there's no way catch that exception and make `LoadImpl` mark it as failed?

srmainwaring, 2021-03-29:
In `OgreRenderEngine::LoadAttempt` the function returns `false` only if an exception is caught.

https://github.com/ignitionrobotics/ign-rendering/blob/d98ea2abeef5c1927ff01b7ed2715a33c493491a/ogre/src/OgreRenderEngine.cc#L278-L292

The issue is that

- `=> OgreRenderEngine::LoadAttempt`
- `=> OgreRenderEngine::CreateRenderWindow()`
- `=> OgreRenderEngine::CreateRenderWindow(const std::string &_handle, ...)`

which contains this code:

https://github.com/ignitionrobotics/ign-rendering/blob/d98ea2abeef5c1927ff01b7ed2715a33c493491a/ogre/src/OgreRenderEngine.cc#L667-L688

The exception is logged, but the function exits without the exception being propagated. So the calling functions have no exception to catch and `OgreRenderEngine::LoadAttempt` reports the attempt as successful even though it wasn't.

srmainwaring, 2021-07-29:
@ahcorde and @chapulina. I've been digging into this and think I may be getting to the reason why Scene3D is not working for macOS.

It seems to come down to issues with threading and the OpenGL context. In what follows I've only been focussing on Ogre2.1 - but similar arguments apply for Ogre2.2 (which now has some support for OpenGL on macOS!) and probably Ogre 1.9 as well.

In the Ogre2 GLPlus render system the Cocoa window that is eventually created on the call to create a render window is very fussy about being on the main thread. An attempt to initialise a Cocoa window or set a view on an NSOpenGLContext will result in a hard crash if you are not in the main thread. If you wrap the offending calls in Ogre with objective-c try catch blocks you get some information - but mostly it just exits and throws the core dump at you. It appears that prior to macOS Catalina the behaviour was to emit warnings and carry on, now it's a crash. By default ignition hides the Ogre logging so you may not have seen the warnings in earlier incarnations.

Now Scene3D winds up initialising Ogre on a render thread which is a problem; because either you pass the Cocoa window handle in and hit the problem with trying to set the view on a NSOpenGLContext on a non-main thread, or you pass a null window handle in (current behaviour) and try to create a Cocoa window on a non-main thread. Both crash.

So I think part of the solution is to ensure that Ogre is initialised on the main thread in Scene3D (in the RenderThread object) before it is moved to the render thread and started. There is also some weirdness going on with the contexts - and I think it might be necessary to capture the context directly from Ogre before switching back to the QML OpenGL context. I haven't got that all nailed down just yet.

By way of progress I've managed to get the simple_demo example from ignition-rendering running in a Qt (not QML) application on macOS Big Sur. It is simpler than the QML case - no separate render thread or competing QML rendering, but it does capture some of the problems of the more complex case, highlights where some of the issues are and how to tackle them. It's in a private repo at present but could be added as an example to either ignition-rendering or ignition-gui if that would help. The objective is to get the stand alone QML version running but that's still wip.

iche033, 2021-07-30:
> So I think part of the solution is to ensure that Ogre is initialised on the main thread in Scene3D (in the RenderThread object) before it is moved to the render thread and started. 

I did some tests in the [ogre_qt_thread](https://github.com/ignitionrobotics/ign-gazebo/compare/ign-gazebo4...ogre_qt_thread) branch, which initializes ogre first before moving to the rendering thread. I don't have a working mac machine to test though but ign-gazebo seems to launch fine with the shapes world on ubuntu bionic. The changes are done against ign-gazebo4 but should be portable to other versions. You can also ignore the GuiRunner changes as those are unrelated but I had to make those changes for ign-gazebo to run.

srmainwaring, 2021-07-30:
Thanks @iche033. I applied your changes from [ogre_qt_thread](https://github.com/ignitionrobotics/ign-gazebo/compare/ign-gazebo4...ogre_qt_thread)  to `Scene3D.cc` in `ign_gui` and ran the standalone `window` example.

The Scene3D plugin loads through the Ogre initialisation stage in `RenderThread::RenderNext()`, and succeeds on the first call to `ignRenderer.Render()` (because the OpenGL context isn't switched). It then fails on the next call to `RenderThread::RenderNext()`. This suggests that the Ogre OpenGL context is not being correctly reset after the QML rendering stage.

Here is the full log including the Ogre messages. The shader compilation warnings are expected on macOS with its supported version of OpenGL. The OpenGL GL_INVALID_ENUM error is because of the incorrect context.

```console
Process exited with code 9.
Launching: /Volumes/MacPro2_DV1/Code/robotics/gazebo/ign_ediface_ws/src/ign-gui/examples/standalone/window/build/window
Launched process 75392
Hello, GUI!
2021-07-30 09:19:43.163480+0100 window[75392:848329] SecTaskLoadEntitlements failed error=22 cs_flags=20, pid=75392
2021-07-30 09:19:43.163538+0100 window[75392:848329] SecTaskCopyDebugDescription: window[75392]/0#-1 LF=0
[1;36m[Dbg] [Application.cc:87] [1;36mInitializing application.[1;36m
[1;36m[GUI] [Dbg] [Application.cc:415] [1;36mCreate main window[1;36m
[1;33m[GUI] [Wrn] [Application.cc:657] [1;33m[QT] Populating font family aliases took 325 ms. Replace uses of missing font family "Roboto" with one that exists to avoid this cost. [1;33m
2021-07-30 09:19:43.801211+0100 window[75392:848380] SecTaskLoadEntitlements failed error=22 cs_flags=20, pid=75392
2021-07-30 09:19:43.801301+0100 window[75392:848380] SecTaskCopyDebugDescription: window[75392]/0#-1 LF=0
2021-07-30 09:19:43.820744+0100 window[75392:848380] SecTaskLoadEntitlements failed error=22 cs_flags=20, pid=75392
2021-07-30 09:19:43.820813+0100 window[75392:848380] SecTaskCopyDebugDescription: window[75392]/0#-1 LF=0
2021-07-30 09:19:44.079353+0100 window[75392:848329] SecTaskLoadEntitlements failed error=22 cs_flags=20, pid=75392
2021-07-30 09:19:44.079440+0100 window[75392:848329] SecTaskCopyDebugDescription: window[75392]/0#-1 LF=0
2021-07-30 09:19:44.246255+0100 window[75392:848329] SecTaskLoadEntitlements failed error=22 cs_flags=20, pid=75392
2021-07-30 09:19:44.246346+0100 window[75392:848329] SecTaskCopyDebugDescription: window[75392]/0#-1 LF=0
[1;33m[GUI] [Wrn] [Application.cc:657] [1;33m[QT] qrc:/qml/StyleDialog.qml:112:3: QML Connections: Implicitly defined onFoo properties in Connections are deprecated. Use this syntax instead: function onFoo(<arguments>) { ... }[1;33m
[1;33m[GUI] [Wrn] [Application.cc:657] [1;33m[QT] qrc:/qml/StyleDialog.qml:105:3: QML Connections: Implicitly defined onFoo properties in Connections are deprecated. Use this syntax instead: function onFoo(<arguments>) { ... }[1;33m
[1;33m[GUI] [Wrn] [Application.cc:657] [1;33m[QT] qrc:/qml/StyleDialog.qml:98:3: QML Connections: Implicitly defined onFoo properties in Connections are deprecated. Use this syntax instead: function onFoo(<arguments>) { ... }[1;33m
[1;33m[GUI] [Wrn] [Application.cc:657] [1;33m[QT] qrc:qml/Main.qml:83:3: QML Connections: Implicitly defined onFoo properties in Connections are deprecated. Use this syntax instead: function onFoo(<arguments>) { ... }[1;33m
[1;33m[GUI] [Wrn] [Application.cc:657] [1;33m[QT] qrc:/qml/PluginMenu.qml:23:3: QML Connections: Implicitly defined onFoo properties in Connections are deprecated. Use this syntax instead: function onFoo(<arguments>) { ... }[1;33m
[1;36m[GUI] [Dbg] [Application.cc:313] [1;36mLoading plugin [[1;36mPublisher[1;36m][1;36m
[1;32m[GUI] [1;32m[Msg] [1;32mAdded plugin [[1;32mPublisher[1;32m] to main window[1;32m
[1;32m[GUI] [1;32m[Msg] [1;32mLoaded plugin [[1;32mPublisher[1;32m] from path [[1;32m/Volumes/MacPro2_DV1/Code/robotics/gazebo/ign_ediface_ws/install/lib/ign-gui-5/plugins/libPublisher.dylib[1;32m][1;32m
[1;36m[GUI] [Dbg] [Application.cc:313] [1;36mLoading plugin [[1;36mScene3D[1;36m][1;36m
[1;32m[GUI] [1;32m[Msg] [1;32mAdded plugin [[1;32m3D Scene[1;32m] to main window[1;32m
[1;32m[GUI] [1;32m[Msg] [1;32mLoaded plugin [[1;32mScene3D[1;32m] from path [[1;32m/Volumes/MacPro2_DV1/Code/robotics/gazebo/ign_ediface_ws/install/lib/ign-gui-5/plugins/libScene3D.dylib[1;32m][1;32m
[1;32m[GUI] [1;32m[Msg] [1;32mIgnRenderer::Initialize: externalWindowHandle = [1;32m13086727840[1;32m
[1;32m[GUI] [1;32m[Msg] [1;32mLoading plugin [[1;32mignition-rendering-ogre2[1;32m][1;32m
Creating resource group General
Creating resource group Internal
Creating resource group Autodetect
SceneManagerFactory for type 'DefaultSceneManager' registered.
Registering ResourceManager for type Material
Registering ResourceManager for type Mesh
Registering ResourceManager for type Mesh2
Registering ResourceManager for type OldSkeleton
MovableObjectFactory for type 'ParticleSystem' registered.
ArchiveFactory for archive type FileSystem registered.
ArchiveFactory for archive type Zip registered.
ArchiveFactory for archive type EmbeddedZip registered.
DDS codec registering
FreeImage version: 3.18.0
This program uses FreeImage, a free, open source image library supporting all common bitmap formats. See http://freeimage.sourceforge.net for details
[... cut ...]
WARNING: Could not find vertex shader attribute 'uv3' to match BindAttributeLocation request.
WARNING: Could not find vertex shader attribute 'uv7' to match BindAttributeLocation request.
OpenGL error 0x0500 GL_INVALID_ENUM in virtual void *Ogre::GL3PlusBufferInterface::map(size_t, size_t, Ogre::MappingState, bool) at line 92 for glBindBuffer
OpenGL error 0x0502 GL_INVALID_OPERATION in void *Ogre::GL3PlusDynamicBuffer::map(size_t, size_t, size_t &) at line 91 for mMappedPtr = glMapBufferRange
Stop reason: EXC_BAD_ACCESS (code=1, address=0x171c00)
```

iche033, 2021-07-30:
> params["useCurrentGLContext"] = "1";
> is treated differently on Linux (and is effectively ignored on macOS and windows).

ah! that param is important as we are sharing GL context and don't want OGRE to create its own. Looking at [this bit of code](https://github.com/ignition-forks/ogre-2.1-release/blob/master/RenderSystems/GL3Plus/src/windowing/OSX/OgreOSXCocoaWindow.mm#L236-L247) in ogre 2.1's cocoa window class, it tries to grab an existing opengl context by handle if `externalGLContext` is specified. So it maybe worth a try setting [that param](https://github.com/ignitionrobotics/ign-rendering/blob/ign-rendering5/ogre2/src/Ogre2RenderEngine.cc#L731-L733) in ign-rendering's ogre engine setup, i.e. move it out of the `if` check. The `_handle` arg there is incorrect as it's just setting it to a `dummyWindow`. You'll probably need to get the current [opengl handle through Qt](https://doc.qt.io/qt-5/qopenglcontext.html#nativeHandle).

Alternatively, the other option is to tweak ogre's cocoa window code to support `currentGLContext` param. I found [this post](https://forums.ogre3d.org/viewtopic.php?t=58981) here with suggested changes but I don't see those changes in the main ogre code base.

srmainwaring, 2021-08-01:
I attempted to pass the native handle from Qt to Ogre in an earlier attempt but without success. The Qt headers for the native context `<QCocoaWindowFunctions>` do not seem to be available as standard so I'll need to revisit and look into that further.

In the meanwhile I've finally got a standalone ignition-rendering / QML sample up and running. It uses the same render to FBO on a thread approach as Scene3D but instead of sharing the FBO context with Ogre the ignition camera image is extracted and then rendered as a textured quad. This is similar to the simple demo example but rendering via a FBO to the Qt SceneGraph instead of a GLUT window. It's not pretty but it does work and at least provides a baseline to try and resolve the context issues.

srmainwaring, 2021-08-04:
With similar changes `ign gazebo` does run on macOS but a number of issues remain:

- The application gets stuck in the call to `_renderSync->WaitForQtThreadAndBlock(lock)` in Scene3D.cc because the `condition_value::wait` is never met. I disabled the check to get the gui running but it needs a proper solution. 
- The server and gui need to be run separately. Running them combined: i.e. `ign gazebo -v4 -r shapes.sdf` will segfault
- Some of the gui plugins cause the application to crash - probably best tracked in separate issues.

[image]

xlla, 2022-05-20:
@srmainwaring how to get ogre2.2 work on macOS?
I use macOS 10.13.6, it support openGL 4.1 and Metal 2 with GTX 1060 6GB.
I have installed CUDA 10.2 and Optix 5.0

<pre>
[Msg] Ignition Gazebo Server v7.0.0~pre1
[Msg] Loading SDF world file[/usr/local/Cellar/ignition-gazebo7/6.999.999~0~20220412/share/ignition/ignition-gazebo7/worlds/shapes.sdf].
[Msg] Loaded level [3]
[Msg] No systems loaded from SDF, loading defaults
[Dbg] [ServerConfig.cc:970] Loaded (3) plugins from file [/Users/xlla/.ignition/gazebo/7/server.config]
[Dbg] [Physics.cc:824] Loaded [ignition::physics::dartsim::Plugin] from library [/usr/local/Cellar/ignition-physics6/5.999.999~0~20220414/lib/ign-physics-6/engine-plugins/libignition-physics-dartsim-plugin.dylib]
[Dbg] [SystemManager.cc:49] Loaded system [ignition::gazebo::systems::Physics] for entity [1]
[Msg] Create service on [/world/shapes/create]
[Msg] Remove service on [/world/shapes/remove]
[Msg] Pose service on [/world/shapes/set_pose]
[Msg] Pose service on [/world/shapes/set_pose_vector]
[Msg] Light configuration service on [/world/shapes/light_config]
[Msg] Physics service on [/world/shapes/set_physics]
[Msg] SphericalCoordinates service on [/world/shapes/set_spherical_coordinates]
[Msg] Enable collision service on [/world/shapes/enable_collision]
[Msg] Disable collision service on [/world/shapes/disable_collision]
[Msg] Material service on [/world/shapes/visual_config]
[Msg] Material service on [/world/shapes/wheel_slip]
[Dbg] [SystemManager.cc:49] Loaded system [ignition::gazebo::systems::UserCommands] for entity [1]
[Dbg] [SystemManager.cc:49] Loaded system [ignition::gazebo::systems::SceneBroadcaster] for entity [1]
[Msg] Serving world controls on [/world/shapes/control], [/world/shapes/control/state] and [/world/shapes/playback/control]
[Msg] Serving GUI information on [/world/shapes/gui/info]
[Msg] World [shapes] initialized with [default_physics] physics profile.
[Msg] Serving world SDF generation service on [/world/shapes/generate_world_sdf]
[Msg] Serving world names on [/gazebo/worlds]
[Msg] Resource path add service on [/gazebo/resource_paths/add].
[Msg] Resource path get service on [/gazebo/resource_paths/get].
[Msg] Resource paths published on [/gazebo/resource_paths].
[Msg] Server control service on [/server_control].
[Msg] Found no publishers on /stats, adding root stats topic
[Msg] Found no publishers on /clock, adding root clock topic
[Dbg] [SimulationRunner.cc:494] Creating PostUpdate worker threads: 2
[Dbg] [SimulationRunner.cc:505] Creating postupdate worker thread (0)
[Msg] Serving scene information on [/world/shapes/scene/info]
[Msg] Serving graph information on [/world/shapes/scene/graph]
[Msg] Serving full state on [/world/shapes/state]
[Msg] Serving full state (async) on [/world/shapes/state_async]
[Msg] Publishing scene information on [/world/shapes/scene/info]
[Msg] Publishing entity deletions on [/world/shapes/scene/deletion]
[Msg] Publishing state changes on [/world/shapes/state]
[Msg] Publishing pose messages on [/world/shapes/pose/info]
[Msg] Publishing dynamic pose messages on [/world/shapes/dynamic_pose/info]
[Dbg] [EntityComponentManager.cc:1617] Updated state thread iterators: 8 threads processing around 4 entities each.
[Wrn] [Component.hh:144] Trying to serialize component with data type [N3sdf3v135WorldE], which doesn't have `operator<<`. Component will not be serialized.
</pre>

<pre>
[Msg] Ignition Gazebo GUI    v7.0.0~pre1
[Dbg] [Application.cc:94] Initializing application.
[Dbg] [Application.cc:120] Qt use opengl 4.1
[GUI] [Dbg] [Application.cc:566] Create main window
[GUI] [Dbg] [PathManager.cc:66] Requesting resource paths through [/gazebo/resource_paths/get]
[GUI] [Wrn] [Application.cc:815] [QT] file::/Gazebo/GazeboDrawer.qml:147:3: QML Dialog: Binding loop detected for property "implicitHeight"
[GUI] [Wrn] [Application.cc:815] [QT] file::/Gazebo/GazeboDrawer.qml:147:3: QML Dialog: Binding loop detected for property "implicitHeight"
[GUI] [Dbg] [Gui.cc:156] GUI requesting list of world names. The server may be busy downloading resources. Please be patient.
[GUI] [Dbg] [PathManager.cc:55] Received resource paths.
[GUI] [Dbg] [Gui.cc:215] Requesting GUI from [/world/shapes/gui/info]...
[GUI] [Dbg] [GuiRunner.cc:145] Requesting initial state from [/world/shapes/state]...
[GUI] [Msg] Loading config [/Users/xlla/.ignition/gazebo/7/gui.config]
[GUI] [Dbg] [Application.cc:429] Loading plugin [MinimalScene]
[GUI] [Dbg] [MinimalScene.cc:634] Creating ign-rendering interface for OpenGL
[GUI] [Dbg] [MinimalScene.cc:634] Creating ign-rendering interface for OpenGL
[GUI] [Dbg] [MinimalScene.cc:788] Creating render thread interface for OpenGL
[GUI] [Msg] Added plugin [3D View] to main window
[GUI] [Msg] Loaded plugin [MinimalScene] from path [/usr/local/Cellar/ignition-gui7/6.999.999~0~20220414/lib/ign-gui-7/plugins/libMinimalScene.dylib]
[... 79 lines cut ...]
[GUI] [Msg] Loading plugin [ignition-rendering-ogre2]
[GUI] [Msg] Loading plugin [ignition-rendering-ogre2]
[GUI] [Dbg] [Spawn.cc:289] Spawn plugin is using camera [scene::Camera(65527)]
[GUI] [Msg] Loading plugin [ignition-rendering-ogre2]
[GUI] [Dbg] [SelectEntities.cc:452] SelectEntities plugin is using camera [scene::Camera(65527)]
[GUI] [Msg] Loading plugin [ignition-rendering-ogre2]
[GUI] [Dbg] [MarkerManager.cc:173] Advertise /marker/list service.
[GUI] [Dbg] [MarkerManager.cc:183] Advertise /marker/list.
[GUI] [Dbg] [MarkerManager.cc:193] Advertise /marker_array.
[... cut ...]
[GUI] [Msg] Follow service on [/gui/follow]
[GUI] [Msg] Move to pose service on [/gui/move_to/pose]
[GUI] [Msg] Camera pose topic advertised on [/gui/camera/pose]
[GUI] [Msg] Follow offset service on [/gui/follow/offset]
[GUI] [Msg] Loading plugin [ignition-rendering-ogre2]
[GUI] [Dbg] [InteractiveViewControl.cc:130] InteractiveViewControl plugin is moving camera [scene::Camera(65527)]
[GUI] [Msg] Loading plugin [ignition-rendering-ogre2]
[GUI] [Msg] Loading plugin [ignition-rendering-ogre2]

</pre>

it only shows black area.
[image]

srmainwaring, 2022-05-20:
Hi @xlla, Gazebo Sim on macOS won't be officially supported until this PR is merged https://github.com/gazebosim/gz-sim/pull/1225 (and this is a handy reminder that I need to get back to that).

You will need a build of Gazebo Sim (Garden) that pulls in the branch in the PR rather than main, otherwise Gazebo will attempt to run with OpenGL which is not going to work on macOS as ogre2.2 requires and OpenGL version > 4.2. This is the most likely explanation for the black screen. All the other dependencies required for Metal in gz-rendering etc. are already merged.

If you try to use the branch in the PR and run into difficulties can you please add a note to the PR, any additional feedback from macOS users would be helpful.

You can verify the Metal rendering is working by building one of the examples in gz-rendering, say simple_demo, which you run as `simple_demo ogre2 metal` to target Metal on macOS.

xlla, 2022-05-20:
Hi @srmainwaring , thanks for your advice, I have merge that PR into my branch, then sensors_demo.sdf get work.
[image]

I have tested some examples of under layer libs.
for ogre2.2.7 
 Sample_PbsMaterials is work on both engine.
metal(59.9 fps) 
[image]

 OpenGL3+ (300 fps)
[image]

for gz-rendering , 2.7
simple_demo is work on ogre and ogre2 metal,  but keep black for ogre2
[image]

for gz-gui , 7.7
only 2d UI elements work, scene3d.config  totally black.
[image]

srmainwaring, 2022-05-20:
@xlla looks great, Metal rendering seems to be working for you. OpenGL will not work for Gazebo, even though the PBS sample for ogre2.2 works, other samples that use OpenGL 4.2+ features will not. Gazebo uses these features => black screen when using OpenGL. 

Scene3d is deprecated in favour of MinimalScene, and only the latter has the Metal support.

Btw I use this patch for gz-gui to fix the gamma on MinimalScene: https://github.com/srmainwaring/ign-gui/tree/feature/ign-gui7-minimalscene-gamma.

xlla, 2022-05-23:
@srmainwaring , after merged all mentioned PRs and commits, 
simple_demo_qml get work.
[image]

shapes.sdf is work too after apply latest gamma patch.

[image]

but camera_sensor.sdf didn't.
I'll post the issues to the related PR.

srmainwaring, 2022-05-23:
> but camera_sensor.sdf didn't.

You need to ensure the sensors plugin in `camera_sensor.sdf` is using ogre2 for rendering, it uses ogre by default.

```diff
    <plugin
      filename="ignition-gazebo-sensors-system"
      name="ignition::gazebo::systems::Sensors">
-      <render_engine>ogre</render_engine>
+      <render_engine>ogre2</render_engine>
    </plugin>
```

xlla, 2022-05-23:
I got it! thank you very much!