gazebosim/gz-gui #162: `ign gui` results in a seg fault

acxz, 2020-12-26:
Running `ign gui` results in a seg fault:

<details>
<summary> Log <\summary>

  ```
/usr/lib/ruby/ignition/cmdgui4.rb:189: [BUG] Segmentation fault at 0x00000005653a88e0
ruby 2.7.2p137 (2020-10-01 revision 5445e04352) [x86_64-linux]

-- Control frame information -----------------------------------------------
c:0005 p:---- s:0038 e:000037 CFUNC  :call
c:0004 p:0016 s:0034 e:000033 METHOD /usr/lib/ruby/ignition/cmdgui4.rb:189
c:0003 p:0391 s:0028 e:000027 METHOD /usr/lib/ruby/ignition/cmdgui4.rb:189
c:0002 p:0675 s:0019 E:000f78 EVAL   /usr/bin/ign:273 [FINISH]
c:0001 p:0000 s:0003 E:001de0 (none) [FINISH]

-- Ruby level backtrace information ----------------------------------------
/usr/bin/ign:273:in `<main>'
/usr/lib/ruby/ignition/cmdgui4.rb:189:in `execute'
/usr/lib/ruby/ignition/cmdgui4.rb:189:in `cmdEmptyWindow'
/usr/lib/ruby/ignition/cmdgui4.rb:189:in `call'

-- Machine register context ------------------------------------------------
 RIP: 0x00007fa580f7f917 RBP: 0x00000005653a88ea RSP: 0x00007ffdbb40ea38
 RAX: 0x00005653a88ea810 RBX: 0x00007ffdbb40ed70 RCX: 0x000000000000000a
 RDX: 0x00000000000003e0 RDI: 0x00000005653a88e0 RSI: 0x000000000000002f
  R8: 0x0000000000000000  R9: 0x0000000000000000 R10: 0x00005653a82d2340
 R11: 0x0000000000000020 R12: 0x00007ffdbb40eaa0 R13: 0x00005653a82d4080
 R14: 0x00005653a82d45d0 R15: 0x00005653a864d7c0 EFL: 0x0000000000010202

-- C level backtrace information -------------------------------------------
/usr/lib/libruby.so.2.7(0x7fa58121fc04) [0x7fa58121fc04]
/usr/lib/libruby.so.2.7(0x7fa58108a019) [0x7fa58108a019]
/usr/lib/libruby.so.2.7(0x7fa58119ab3a) [0x7fa58119ab3a]
/usr/lib/libc.so.6(__restore_rt+0x0) [0x7fa580e5a6a0]
/usr/lib/libc.so.6(0x162917) [0x7fa580f7f917]
/usr/lib/libQt5Core.so.5(_ZNK23QCoreApplicationPrivate7appNameEv+0x43) [0x7fa57b051c63]
/usr/lib/libQt5Core.so.5(_ZN23QCoreApplicationPrivate4initEv+0x577) [0x7fa57b059147]
/usr/lib/libQt5Gui.so.5(_ZN22QGuiApplicationPrivate4initEv+0x30) [0x7fa57b426b20]
/usr/lib/libQt5Widgets.so.5(_ZN19QApplicationPrivate4initEv+0xa) [0x7fa57ccf26aa]
/usr/lib/libignition-gui4.so(_ZN8ignition3gui11ApplicationC2ERiPPcNS0_10WindowTypeE+0x36) [0x7fa57d250786]
/usr/lib/libignition-gui4.so(cmdEmptyWindow+0x3e) [0x7fa57d26ddde]
/usr/lib/libffi.so.7(0x7fa57d2baacd) [0x7fa57d2baacd]
/usr/lib/libffi.so.7(0x7fa57d2ba03a) [0x7fa57d2ba03a]
/usr/lib/ruby/2.7.0/x86_64-linux/fiddle.so(0x7fa57d2c4979) [0x7fa57d2c4979]
/usr/lib/libruby.so.2.7(rb_nogvl+0x94) [0x7fa5811d44f4]
/usr/lib/ruby/2.7.0/x86_64-linux/fiddle.so(0x7fa57d2c4d16) [0x7fa57d2c4d16]
/usr/lib/libruby.so.2.7(0x7fa5811fee92) [0x7fa5811fee92]
/usr/lib/libruby.so.2.7(0x7fa5811f9a96) [0x7fa5811f9a96]
/usr/lib/libruby.so.2.7(0x7fa581209bf3) [0x7fa581209bf3]
/usr/lib/libruby.so.2.7(rb_vm_exec+0x1bb) [0x7fa58120f43b]
/usr/lib/libruby.so.2.7(0x7fa58108e2c1) [0x7fa58108e2c1]
/usr/lib/libruby.so.2.7(ruby_run_node+0x4a) [0x7fa58109283a]
ruby(0x5653a7688081) [0x5653a7688081]
/usr/lib/libc.so.6(__libc_start_main+0xf2) [0x7fa580e45152]
ruby(_start+0x2e) [0x5653a76880ce]

-- Other runtime information -----------------------------------------------

* Loaded script: /usr/bin/ign

* Loaded features:

    0 enumerator.so
    1 thread.rb
    2 rational.so
    3 complex.so
    4 ruby2_keywords.rb
    5 /usr/lib/ruby/2.7.0/x86_64-linux/enc/encdb.so
    6 /usr/lib/ruby/2.7.0/x86_64-linux/enc/trans/transdb.so
[... 594 lines cut ...]
7fa580fe6000-7fa581013000 r--p 00000000 103:04 6223939                   /usr/lib/libruby.so.2.7.2
7fa581013000-7fa58122a000 r-xp 0002d000 103:04 6223939                   /usr/lib/libruby.so.2.7.2
7fa58122a000-7fa581322000 r--p 00244000 103:04 6223939                   /usr/lib/libruby.so.2.7.2
7fa581322000-7fa58132b000 r--p 0033b000 103:04 6223939                   /usr/lib/libruby.so.2.7.2
7fa58132b000-7fa58132c000 rw-p 00344000 103:04 6223939                   /usr/lib/libruby.so.2.7.2
7fa58132c000-7fa581348000 rw-p 00000000 00:00 0 
7fa581348000-7fa58134a000 r--p 00000000 103:04 6163799                   /usr/lib/ld-2.32.so
7fa58134a000-7fa58136b000 r-xp 00002000 103:04 6163799                   /usr/lib/ld-2.32.so
7fa58136b000-7fa581374000 r--p 00023000 103:04 6163799                   /usr/lib/ld-2.32.so
7fa581374000-7fa581375000 r--p 0002b000 103:04 6163799                   /usr/lib/ld-2.32.so
7fa581375000-7fa581377000 rw-p 0002c000 103:04 6163799                   /usr/lib/ld-2.32.so
7ffdbac13000-7ffdbb412000 rw-p 00000000 00:00 0                          [stack]
7ffdbb578000-7ffdbb57c000 r--p 00000000 00:00 0                          [vvar]
7ffdbb57c000-7ffdbb57e000 r-xp 00000000 00:00 0                          [vdso]
ffffffffff600000-ffffffffff601000 --xp 00000000 00:00 0                  [vsyscall]
  ```

<\details>

Sys info:
OS: Arch Linux
GCC: 10.2.0
Qt: 5.15.2
ruby: 2.7.2
protobuf: 3.12.4

chapulina, 2020-12-28:
Are you compiling the `ign-gui4` branch from source? That crash seems to be coming from Qt's internals, are you able to compile and run some simple Qt example using `QApplication`?

acxz, 2020-12-29:
> Are you compiling the ign-gui4 branch from source?

yep, just a regular `mkdir build && cd build && cmake .. && make && make install && ign gui`

> That crash seems to be coming from Qt's internals, are you able to compile and run some simple Qt example using QApplication?

I tried this example: That crash seems to be coming from Qt's internals, are you able to compile and run some simple Qt example using QApplication?
and it compiles and runs just fine.

chapulina, 2020-12-29:
> I tried this example: 

Woops, it looks like you pasted the wrong clipboard :sweat_smile: 

---

I thought this could be a GCC 10.2.0 issue like https://github.com/ignitionrobotics/ign-physics/issues/176, but I was able to compile and run on Ubuntu Focal using that. So I think this may be related to the Qt version, I'm using 5.12.8.

acxz, 2020-12-30:
yikes my bad, fixed the above comment with: https://riptutorial.com/qt/example/15047/simple-example

acxz, 2020-12-30:
> So I think this may be related to the Qt version, I'm using 5.12.8.

yeah it is most def the qt version, but I just don't even know where to start debugging the error. Like what cpp file should I be looking at?

I was trying to use gdb, but then I got a bit lost since everything is wrapped in a ruby script. If you have any tips on trying to debug this that would be much appreciated.

chapulina, 2020-12-30:
> I was trying to use gdb, but then I got a bit lost since everything is wrapped in a ruby script

There are tips on the [Debugging tutoria](https://ignitionrobotics.org/api/gazebo/4.0/debugging.html).

> Like what cpp file should I be looking at?

The crash is coming from `gui::Application`'s constructor:

https://github.com/ignitionrobotics/ign-gui/blob/016cd325fbabb42bb2ea8e1b33bafdc478f17bf8/src/Application.cc#L84-L85

A simpler example that you could try debugging is this executable, which you can run gdb directly on:

https://github.com/ignitionrobotics/ign-gui/blob/ign-gui4/examples/standalone/window/window.cc

Hope that helps

acxz, 2020-12-30:
> There are tips on this debugging tutorial.

Thx so much, should have done a better job searching...

> A simpler example that you could try debugging is this executable

The window.cc example works perfectly.

acxz, 2020-12-31:
I've resolved the issue on my system on #165 but not sure if it works on older QTs.

PR #165 (2021-01-05): use default value of double ptr
this fix allows QApplication to properly instantiate on qt-5.15

resolves #162 

Seems like QT doesn't like the default value we are passing to QApplication constructor. This PR just uses the default **char value to be passed in.
closed 2021-01-05 by chapulina