gazebosim/gz-gui #183: Display grid - configure from launch file

peci1, 2021-02-19:
## Desired behavior
I saw there is the GridConfig plugin, which is nice, but I haven't found a way to configure a grid at startup time, automatically. Ideally, even have the option to override `<world><scene><grid>false` is specified in the world SDF.

chapulina, 2021-02-19:
Note that the `GridConfig` plugin lives in `ign-gazebo`. [Here](https://github.com/ignitionrobotics/ign-gazebo/blob/4c2f498344b0316aca689271015951a23ad6cf9e/src/gui/plugins/grid_config/GridConfig.cc#L78)'s where the SDF would be parsed.

The `Grid3D` plugin on the other hand lives in `ign-gui` and [accepts SDF parameters](https://github.com/ignitionrobotics/ign-gui/blob/56bb6b3fac106e898a64d0461466958c6170a2ec/src/plugins/grid_3d/Grid3D.cc#L130).

We should unify these 2 plugins into one.

peci1, 2021-02-20:
Thanks for the clarification. However, Grid3d doesn't work for me with ogre2. I get a flickering in the 3D window, and the Qt widget shows nothing but the titlebar. Is that a known issue?

chapulina, 2021-02-23:
> Grid3d doesn't work for me with ogre2 on Dome.

It's been fixed for Citadel on #172 and is being ported forward in #180

peci1, 2021-03-16:
I've now tried adding Grid3D into gui.config, but Ign Gazebo 4 GUI crashes right on startup. 

It's sufficient to add

```XML
<plugin filename="Grid3D" name="Grid">
    <engine>ogre2</engine>
    <scene>scene</scene>
</plugin>
```

to e.g. gazebo_plugins.ign from ign-launch examples. It doesn't even have to contain any `<insert>` section.

Is that still expected and waiting for a fix to be merged, or should I report it as a new issue?

Adding the grid at runtime now works.

chapulina, 2021-12-07:
> We should unify these 2 plugins into one.

#324 consolidates `Grid3D` with `GridConfig` and should work from startup or when added at runtime.

closed 2021-12-08 by chapulina