gazebosim/gz-gui #186: Add a teleop plugin that sends twist commands

chapulina, 2021-03-03:
A GUI plugin that has arrow buttons and publishes twist messages to a custom topic to control robots would be nice to test vehicles easily.

## Desired behavior

The user loads a world in `ign-gazebo` that has a vehicle in it, like `ign gazebo diff_drive.sdf`.

Then the user can insert the GUI twist teleop plugin, set the desired topic, and start sending commands by pressing buttons or keys, or with sliders.

## Alternatives considered

The approach currently taken for teleop is using `KeyPublisher` + `TriggeredPublisher` as described on the [moving robot tutorial](https://ignitionrobotics.org/docs/citadel/moving_robot). This can be inconvenient because:

* There's no way to load and configure `TriggeredPublisher` at runtime yet (https://github.com/ignitionrobotics/ign-gazebo/issues/190), so it needs to be hardcoded in the SDF file.
* The combination of controls that can be made with that approach is very limited, because the speed has to be hardcoded. It isn't possible for example to increment the speed at each click.

## Implementation suggestion

The [rqt_robot_steering](http://wiki.ros.org/rqt_robot_steering) plugin for ROS has vertical and horizontal sliders to control throttle and steering.

LolaSegura, 2021-06-11:
I am tackling this on my branch: on the latest commit I added the button functionalities 
https://github.com/ignitionrobotics/ign-gui/commit/31c4a4bf4bb1bfbb489a7af97b29acf9a4db70f7

[](https://user-images.githubusercontent.com/48759425/121741005-47b73a80-cad4-11eb-9b0c-d44f65dbccf2.mp4)

You can set the desire topic and change the module of the linear and angular velocities. 

For now it can only be controlled clicking on the buttons. So the next steps would be :
* Allow moving the robot using keys.
* Adding a switch to enable-disable the plugin.

jennuine, 2021-06-11:
Looks good @LolaSegura ! A suggestion would be to add a stop button. Also, it would be nice to turn while driving forward/backward (for example, driving forward and turning left at the same time)

chapulina, 2021-06-14:
> I am tackling this on my branch

Nice! Looking good!

> it would be nice to turn while driving forward/backward (for example, driving forward and turning left at the same time)

+1, that would be very helpful! I believe that's why the robot steering plugin linked above uses sliders instead of buttons.

But I think we can also achieve the same with buttons. For example, when the user presses "forward", we increment the forward velocity instead of setting it. So the more the user presses forward, the faster the robot goes. Then the same number of backward presses results in the robot slowing down until it stops. Pressing right while there's some forward velocity would make the robot go forward while rotating, etc.

---

Another idea is to use unicode characters with arrows on the buttons instead of the words "Forward", etc.

francocipollone, 2021-06-15:
> Looks good @LolaSegura ! A suggestion would be to add a stop button. Also, it would be nice to turn while driving forward/backward (for example, driving forward and turning left at the same time)

That's a great idea. Actually, Lola has implemented these buttons too:

[image]
This is wip so the buttons could be replaced by something else to be more user-friendly.

> Another idea is to use unicode characters with arrows on the buttons instead of the words "Forward", etc.

+1000. 

> But I think we can also achieve the same with buttons. For example, when the user presses "forward", we increment the forward velocity instead of setting it. So the more the user presses forward, [...]

I think that approach is interesting, it is definitely good to mix forward/backward movement and clockwise rotation at the same time however I feel like probably it would be a bit confusing for the user.

What about the following:

- Having buttons to move in 8 directions + stop as it is shown in the previous image. Pressing these buttons will send a twist message with the angular and linear velocity that is described in the text fields.
- Allow pressing keys from the keyboard. Replicate functionality with the buttons in the visualizer
 Tipically W: forward

francocipollone, 2021-06-15:
(Sorry for the noise, I hit some key combination by mistake)

chapulina, 2021-06-16:
> W: forward, A: left, X: backward, D: right., S: stop. (And Q, E, Z and C for the combinations)

+1 to having WASD+ keys. The advantage of keys over buttons is that the user can press more than one at the same time. So we could achieve the linear + angular combinations by combining keys instead of using new keys. For example, front + right would be `W` + `D` instead of `Q`.

> Having buttons to move in 8 directions + stop 

I'm ok with that. That's the approach in [turtlebot_teleop](https://learn.turtlebot.com/2015/02/03/5/), for example.

Another idea that came to mind is to make just 4 buttons that are togglable. So the user could enable front and right at the same time. The advantage would be that:
* we have less buttons
* the user knows the current state of the controller by looking at the widget and seeing which buttons are checked

> Sliders

If it's not too difficult, I think they'd be a nice addition. Similar to the idea above, they allow throttle + steer combinations while providing visual feedback of the current command.

> You certainly have more insight about what could be more useful to the users than me smile

Not sure about that :wink: In the end of the day, the best approach is to write something, then have someone else try to use it. In the end of the day, there's no ideal GUI control for 2D movement - that's why we have joysticks! :smile:

LolaSegura, 2021-06-18:
> Another idea that came to mind is to make just 4 buttons that are togglable. So the user could enable front and right at the same time.

I implemented this idea so now there are only 5 togglable buttons. 4 for direction and 1 for stopping the vehicle. So now for turning while moving forward you have to push both 'forward' and 'right' buttons for example.
I also change the `forward`, `left`, `right` and `backward` text of the buttons for black triangles pointing to the direction. I couldn't think of any symbol for the `stop` button so I left it as it was. But any suggestions are welcome. 
I added some visual feature so that it shows which buttons are toggled at the moment.

> The advantage of keys over buttons is that the user can press more than one at the same time. So we could achieve the linear + angular combinations by combining keys instead of using new keys.

I implemented this as well. So now the keys available for moving the vehicle are `W`, `A`, `D` and `X`. I didn't include any key for stopping because the vehicle only moves while the keys are pressed. 

Here is a video showing the plugin with all that I mentioned before.
[Demo](https://user-images.githubusercontent.com/48759425/122595556-09270000-d03f-11eb-8d2f-969c4e61b685.mp4)

The next steps to follow would be:
- Adding sliders.
- Solve a bug that appears when you first use the buttons and then the keyboard.

chapulina, 2021-06-18:
That looks great, @LolaSegura ! Feel free to open a PR once you're ready!

The last recommendation I have is to use Material Design controls so the UI looks a bit more modern. I think that changing to these imports should do the trick:

```
import QtQuick 2.9
import QtQuick.Controls 2.2
```

See for example how the `Publisher` plugin's `TextField` and `Switch` have a more modern look and make use of the UI's color scheme for highlights:

[image]

PR #245 (2021-07-19): New teleop plugin implementation.

# 🎉 New feature

Closes #186 

## Summary
The user can insert the GUI teleop plugin in a world that has a vehicle and setting a desire topic, can send commands by pressing buttons, using the keyboard or with sliders. 

## Test it
You can test it loading a world in ignition gazebo that has a vehicle like `ign gazebo diff_drive_skid.sdf`, selecting the Teleop plugin, configuring the desired topic to send the twist commands, `/model/vehicle_blue/cmd_vel`.

https://user-images.githubusercontent.com/48759425/124180840-2621f100-da8b-11eb-9674-7226f0f86b3e.mp4

To t [... cut]
closed 2021-07-20 by chapulina