gazebosim/gz-gui #198: check_test_ran.py requires grep and xsltproc

traversaro, 2021-03-24:
## Desired behavior
The `check_test_ran.py` requires grep and xsltproc, that may not be present in some machine (especially Windows), leading to `check_` test failures.

## Implementation suggestion
It seems that other ignition projects use a different version of check_test_ran.py that does not use grep and xsltproc (see https://github.com/ignitionrobotics/ign-math/commits/ign-math6/tools/check_test_ran.py, grep and xsltproc usage removed in https://github.com/ignitionrobotics/ign-math/commit/69d881d51c78929f096791a31f23b87f6032ee88#diff-2d6a060081b4c12efdbe4f12e846ead21821236e834f41885a04e41e884687c2 ). Could it make sense to use the same version here, or instead it is better to stick to the version that uses grep and xsltproc ?

## Additional context
Related to https://github.com/conda-forge/libignition-gui-feedstock/issues/2 .

scpeters, 2021-03-29:
This is a leftover from osrf/gazebo, which uses the [qtest_to_junit.xslt](https://github.com/osrf/gazebo/blob/gazebo11/tools/qtest_to_junit.xslt) file to translate the xml output files produced by QTest into junit files that can be read by Jenkins. We removed this functionality from other ignition packages, as you noted, but I hadn't removed it here as I thought we were still using QTest. It turns out we stopped using it as of [bitbucket pull request 70](https://osrf-migration.github.io/ignition-gh-pages/#!/ignitionrobotics/ign-gui/pull-requests/70/page/1) (13d314eb2b6a9fc3ebb4ae723bad990546d557aa), so we can remove the `grep` and `xsltproc` portions from `check_test_ran.py` as well.

scpeters, 2021-03-29:
see #203

commit bd6f2e2f (2021-03-29): check_test_ran.py: remove grep/xsltproc

We aren't using QTest anymore in ign-gui, so remove
the parts of check_test_ran.py that translated QTest
xml files into junit, since they don't work easily
on Windows and aren't needed any longer.

Fixes #198.
PR #203 (2021-04-05): check_test_ran.py: remove grep/xsltproc
# 🦟 Bug fix

Fixes #198 

## Summary

We aren't using QTest anymore in ign-gui, so remove
the parts of check_test_ran.py that translated QTest
xml files into junit, since they don't work easily
on Windows and aren't needed any longer.

## Checklist
- [X] Signed all commits for DCO
- [ ] Added tests
- [ ] Updated documentation (as needed)
- [ ] Updated migration guide (as needed)
- [ ] `codecheck` passed (See [contributing](https://ignitionrobotics.org/docs/all/contributing#contributing-code))
- [ ] All tests passed (See [test coverage](https://ignitionro [... cut]
commit 064f8156 (2021-04-05): check_test_ran.py: remove grep/xsltproc (#203)

We aren't using QTest anymore in ign-gui, so remove
the parts of check_test_ran.py that translated QTest
xml files into junit, since they don't work easily
on Windows and aren't needed any longer.

Fixes #198.
commit 8e8f51cc (2021-06-22): 3 ➡️ 4 (#234)

* Release 3.5.1 (#195)

* check_test_ran.py: remove grep/xsltproc (#203)

We aren't using QTest anymore in ign-gui, so remove
the parts of check_test_ran.py that translated QTest
xml files into junit, since they don't work easily
on Windows and aren't needed any longer.

Fixes #198.

* Fixed material specular in scene3D (#218)

* Remove tools/code_check and update codecov (#222)

* Removed duplicated code with rendering::sceneFromFirstRenderEngine (#223)

* Emit more events from Scene3D (#213)

* Start porting events from ign-gazebo

* Remove test that fails due to #216

* Move  [... cut]
closed 2021-04-05 by chapulina