gazebosim/gz-gui #216: Failing to load render engine plugin in tests

chapulina, 2021-04-29:
## Environment
* OS Version: Ubuntu Bionic and Focal
* Source or binary build? Source, `ign-gui3` onwards
 
## Description
* Expected behavior: Loading `Scene3D` within a test should successfully load a rendering engine
* Actual behavior: `Scene3D` fails to load the rendering engine during the test, but it successfully loads it during regular execution

## Steps to reproduce

Create a test that loads `Scene3D` and executes the Window, like this one:

https://github.com/ignitionrobotics/ign-gui/blob/2eb2ee054c984b6573c6fbf4f7edcf84f2fc29de/src/plugins/scene3d/Scene3D_TEST.cc#L71-L94

And the rendering engine will fail to load:

```
[ RUN      ] Scene3DTest.Config                                                                                                   
[GUI] [Dbg] [Application.cc:87] Initializing application.
[GUI] [Dbg] [Application.cc:407] Create main window      
[GUI] [Dbg] [Application.cc:305] Loading plugin [Scene3D]                                    
[GUI] [Msg] Added plugin [3D Scene] to main window                                                                                       
[GUI] [Msg] Loaded plugin [Scene3D] from path [/home/chapulina/dev_bionic/ws_citadel/build/ignition-gui3/lib/libScene3D.so]   
[GUI] [Msg] Loading plugin [ignition-rendering-ogre]                                              
[GUI] [Err] [RenderEngineManager.cc:468] Failed to load plugin [ignition-rendering-ogre] : couldn't load library on path [/home/chapulina/dev_bionic/ws_citadel/install/lib/ign-rendering-3/engine-plugins/libignition-rendering-ogre.so].
/home/chapulina/dev_bionic/ws_citadel/src/ign-gui/src/plugins/scene3d/Scene3D_TEST.cc:98: Failure        
Expected: (nullptr) != (engine), actual: (nullptr) vs NULL
```

The main difference I can think of between the test executable and running the GUI is that the test links against `Scene3D` instead of just loading it. So I tried just linking the test against `ign-rendering` and not against the plugin, and this time the plugin fails to load:

```
[GUI] [Err] [Application.cc:333] Failed to load plugin [Scene3D] : couldn't load library on path [/home/chapulina/dev_bionic/ws_citadel/buil
d/ignition-gui3/lib/libScene3D.so].
```

I did some digging through `ign-plugin` to try finding out what's happening but I didn't find anything worth documenting.

## Additional info

This came up on #170 and #213

scpeters, 2021-05-06:
I've been looking into this a bit, and since `Scene3d_TEST.cc` isn't yet testing any methods of the Scene3D plugin, I've made it an integration test instead in 899b97a0ce15efbbd93640669592a367c68a510c (branch `scpeters/3/events_debug`). Now the plugins load, but the test fails to get a scene pointer from the `RenderEngine`. I suspect this is because the rendering system has not had time to initialize in the test. Is there a way to execute the window for a second or so in order to initialize the app?

chapulina, 2021-05-07:
> Is there a way to execute the window for a second or so in order to initialize the app?

I've been using `win->QuickWindow()->show();` like it was done on the [removed test](https://github.com/ignitionrobotics/ign-gui/pull/213/commits/8299a15bf5459adb05810afccd6254e9b61c12b5). 

This works for non-rendering plugins, for example:

https://github.com/ignitionrobotics/ign-gui/blob/ffbdb26ce519f85b273d57527a11470dae3838e1/src/plugins/world_control/WorldControl_TEST.cc#L89-L90

> Now the plugins load

Are you building from source? Without https://github.com/ignitionrobotics/ign-rendering/pull/236, which hasn't been released yet, the `[GUI] [Msg] Loading plugin [ignition-rendering-ogre]` message can be misleading and give the impression that the plugin was successfully loaded, but it may not have.

scpeters, 2021-05-07:
> Are you building from source? Without [ignitionrobotics/ign-rendering#236](https://github.com/ignitionrobotics/ign-rendering/pull/236), which hasn't been released yet, the `[GUI] [Msg] Loading plugin [ignition-rendering-ogre]` message can be misleading and give the impression that the plugin was successfully loaded, but it may not have.

yes, I was building `ign-plugin` and `ign-rendering` from source in order to add a bunch of print statements. I'll run it again and paste the console output from my branch...

scpeters, 2021-05-07:
I get the following as of https://github.com/ignitionrobotics/ign-gui/commit/899b97a0ce15efbbd93640669592a367c68a510c when building ign-rendering3 from source:

~~~
$ bin/INTEGRATION_scene3d 
Running main() from /home/scpeters/maliput_ws/src/ign-gui3/test/gtest/src/gtest_main.cc
[==========] Running 2 tests from 1 test case.
[----------] Global test environment set-up.
[----------] 2 tests from Scene3DTest
[ RUN      ] Scene3DTest.Load
QStandardPaths: XDG_RUNTIME_DIR not set, defaulting to '/tmp/runtime-scpeters'
[Dbg] [Application.cc:87] Initializing application.
[GUI] [Dbg] [Application.cc:407] Create main window
[GUI] [Dbg] [Application.cc:305] Loading plugin [Scene3D]
[GUI] [Msg] Added plugin [3D Scene] to main window
[GUI] [Msg] Loaded plugin [Scene3D] from path [/home/scpeters/maliput_ws/build/ignition-gui3/lib/libScene3D.so]
[GUI] [Dbg] [Application.cc:130] Terminating application.
[       OK ] Scene3DTest.Load (473 ms)
[ RUN      ] Scene3DTest.Config
[GUI] [Dbg] [Application.cc:87] Initializing application.
[GUI] [Dbg] [Application.cc:407] Create main window
[GUI] [Dbg] [Application.cc:305] Loading plugin [Scene3D]
[GUI] [Msg] Added plugin [3D Scene] to main window
[GUI] [Msg] Loaded plugin [Scene3D] from path [/home/scpeters/maliput_ws/build/ignition-gui3/lib/libScene3D.so]
[GUI] [Msg] Loading plugin [ignition-rendering-ogre]
/home/scpeters/maliput_ws/src/ign-gui3/test/integration/scene3d.cc:98: Failure
Expected equality of these values:
  1u
    Which is: 1
  engine->SceneCount()
    Which is: 0
/home/scpeters/maliput_ws/src/ign-gui3/test/integration/scene3d.cc:100: Failure
Expected: (nullptr) != (scene), actual: (nullptr) vs 16-byte object <00-00 00-00 00-00 00-00 00-00 00-00 00-00 00-00>
[GUI] [Dbg] [Application.cc:130] Terminating application.
[  FAILED  ] Scene3DTest.Config (740 ms)
~~~

Having added some additional print statements, I think the Scene is not initialized by the time the expectation fails

chapulina, 2021-05-07:
> I get the following

Is this with the `show` call?

scpeters, 2021-05-07:
> > I get the following
> 
> Is this with the `show` call?

yes

https://github.com/ignitionrobotics/ign-gui/blob/899b97a0ce15efbbd93640669592a367c68a510c/test/integration/scene3d.cc#L92

chapulina, 2021-05-07:
> yes

Nice! I was just checking. Why do you think it behaves differently as an integration test, is it because it's not linked against the `Scene3D` plugin? I thought I had tried that.

> I think the Scene is not initialized by the time the expectation fails

That sounds familiar to me. You can try processing events in a loop until the scene is available to give the plugin some time to load. Like this:

https://github.com/ignitionrobotics/ign-gui/blob/ffbdb26ce519f85b273d57527a11470dae3838e1/src/plugins/publisher/Publisher_TEST.cc#L131-L138

chapulina, 2021-05-07:
> the test now works in my branch 

That's great! Feel free to commit to that branch directly :tada:

scpeters, 2021-05-08:
ok, I've pushed to that branch. I'll close this now

closed 2021-05-08 by scpeters