gazebosim/gz-gui #248: Gazebo doesn't exit cleanly when clicking the "x" button

adlarkin, 2021-07-06:
## Environment
* OS Version: Ubuntu 18.04
* Source or binary build? Binary, Fortress

## Description
_Note: I'm not sure if this issue belongs here, or in https://github.com/ignitionrobotics/ign-gazebo_

When using `ign-gazebo`, I get buggy behavior when trying to exit the GUI by pressing the "x" in the top-right menu. When I click the red "x" button, I am prompted with the following message box:

[image]

When I click "ok", the GUI usually freezes and doesn't exit/close (things work as expected when I click "cancel"). When the GUI seizes up, I have to go to the terminal and kill Gazebo with `ctrl-c`. When I exit with `ctrl-c`, I sometimes see error messages in the terminal. Here are the error messages I have seen:
```
terminate called after throwing an instance of 'std::system_error'
  what():  Invalid argument
```

```
ign gazebo gui: ../nptl/pthread_mutex_lock.c:433: __pthread_mutex_lock_full: Assertion `INTERNAL_SYSCALL_ERRNO (e, __err) != ESRCH || !robust' failed.
```

I have also seen the following messages after force-quitting with `ctrl-c` that are probably more so "debug/info messages" than they are "error messages":

```
Escalating to SIGKILL on [Ignition Gazebo GUI]
```

```
Escalating to SIGKILL on [Ignition Gazebo GUI]
Traceback (most recent call last):
	3: from /usr/bin/ign:275:in `<main>'
	2: from /usr/lib/ruby/ignition/cmdgazebo6.rb:473:in `block in execute'
	1: from /usr/lib/ruby/ignition/cmdgazebo6.rb:170:in `killProcess'
/usr/lib/ruby/ignition/cmdgazebo6.rb:170:in `kill': No such process (Errno::ESRCH)
```

I was able to get the GUI to shut down successfully twice by clicking "x" -> "ok" with simulation running, but never had success with this if simulation was paused. Even with simulation running, the GUI would seize up most of the time, and I'd have to force-kill the GUI by pressing `ctrl-c` in the terminal. On one of the successful GUI shutdowns, I was given the following backtrace in my terminal window (the other successful GUI shutdown was clean and gave no error output in my terminal window):

<details>

<summary>Backtrace</summary>

```
[BUG] Segmentation fault at 0x00007f2d1d164ed8
ruby 2.5.1p57 (2018-03-29 revision 63029) [x86_64-linux-gnu]

-- Machine register context ------------------------------------------------
 RIP: 0x00007f2d6945ecd7 RBP: 0x00007f2d5000baa0 RSP: 0x00007fff01638170
 RAX: 0x00007f2d1d164ed8 RBX: 0x00007f2d5000baa0 RCX: 0x0000000000000000
 RDX: 0x00007f2d5000baa0 RDI: 0x00007f2d5000ba90 RSI: 0x0000557cd2bf44e0
  R8: 0x000000000000007e  R9: 0x00007f2d4803ba50 R10: 0x0000000000000065
 R11: 0x00007f2d6b30cbd0 R12: 0x00007f2d5000ba90 R13: 0x0000000000000001
 R14: 0x0000000000000001 R15: 0x00007f2d69558cb0 EFL: 0x0000000000010217

-- C level backtrace information -------------------------------------------
/usr/lib/x86_64-linux-gnu/libruby-2.5.so.2.5(0x7f2d6ecd4a15) [0x7f2d6ecd4a15]
/usr/lib/x86_64-linux-gnu/libruby-2.5.so.2.5(0x7f2d6ecd4c4c) [0x7f2d6ecd4c4c]
/usr/lib/x86_64-linux-gnu/libruby-2.5.so.2.5(0x7f2d6eb9e874) [0x7f2d6eb9e874]
/usr/lib/x86_64-linux-gnu/libruby-2.5.so.2.5(0x7f2d6ec64a32) [0x7f2d6ec64a32]
/lib/x86_64-linux-gnu/libc.so.6(0x7f2d6e765040) [0x7f2d6e765040]
/usr/lib/x86_64-linux-gnu/libQt5Qml.so.5(0x7f2d6945ecd7) [0x7f2d6945ecd7]
/usr/lib/x86_64-linux-gnu/libQt5Qml.so.5(_ZN17QQmlPropertyCacheD1Ev+0x6e) [0x7f2d69654fce]
/usr/lib/x86_64-linux-gnu/libQt5Qml.so.5(_ZN17QQmlPropertyCacheD0Ev+0x9) [0x7f2d69655419]
/usr/lib/x86_64-linux-gnu/libQt5Qml.so.5(0x7f2d69621462) [0x7f2d69621462]
/usr/lib/x86_64-linux-gnu/libQt5Qml.so.5(0x7f2d69621489) [0x7f2d69621489]
/lib/x86_64-linux-gnu/libc.so.6(0x7f2d6e769161) [0x7f2d6e769161]
/lib/x86_64-linux-gnu/libc.so.6(0x4325a) [0x7f2d6e76925a]
/usr/lib/x86_64-linux-gnu/libruby-2.5.so.2.5(ruby_run_node+0x10) [0x7f2d6eba6400]
/usr/lib/x86_64-linux-gnu/libruby-2.5.so.2.5(0x7f2d6ec2a9a7) [0x7f2d6ec2a9a7]
/usr/lib/x86_64-linux-gnu/libruby-2.5.so.2.5(0x7f2d6ecbd339) [0x7f2d6ecbd339]
[... 1450 lines cut ...]
ffffffffff600000-ffffffffff601000 --xp 00000000 00:00 0                  [vsyscall]

[NOTE]
You may have encountered a bug in the Ruby interpreter or extension libraries.
Bug reports are welcome.
For details: http://www.ruby-lang.org/bugreport.html
```

</details>

## Steps to reproduce

1. Start `ign-gazebo`, either paused or running:
```
ign gazebo
```
or
```
ign gazebo -r
```
2. Click the "x" button in the top bar of the GUI
3. When asked, "Do you really want to exit?", click "ok"

## Output
Example output has been given above, including a backtrace.

francocipollone, 2021-07-30:
I tried it several times in order to force this behavior but I wasn't able to reproduce it.

I am using fortress binary in ubuntu. I am using `rocker` to set up this container.

https://user-images.githubusercontent.com/53065142/127712305-0ccb7017-dfd1-484c-a349-82a4a40a08ca.mp4

chapulina, 2021-07-30:
Thanks for looking into it, @francocipollone . It could be related to issues like https://github.com/ignitionrobotics/ign-gazebo/issues/725, which crash the GUI on exit. I imagine that the GUI crashing during destruction could cause a frozen state on the dialog.

adlarkin, 2021-07-30:
> It could be related to issues like ignitionrobotics/ign-gazebo#725

I was seeing this issue without using the transform tool, but I did not see this issue with citadel if that helps at all.

adlarkin, 2021-08-03:
I'm not longer seeing this issue :thinking: :sweat_smile: I'll go ahead and close it.

closed 2021-08-03 by adlarkin