gazebosim/gz-gui #263: gui3-3.7.0 broken, missing sceneFromFirstRenderEngine

mboisson, 2021-08-03:
## Environment
* OS Version: 
* Source or binary build?
Source, gui3-3.7.0 with rendering3-3.4.0
  
## Description
This is a compilation bug. Compiling gui3-3.7.0, with rendering3-3.4.0 yields : 
``` 
25301 /tmp/mboisson/avx2/ignition/citadel/GCC-9.3.0/ign-gui-ignition-gui3_3.7.0/src/plugins/screenshot/Screenshot.cc:185:27: error: sceneFromFirstRenderEngine is not a member of ignition::rendering
25302   185 |   auto scene = rendering::sceneFromFirstRenderEngine();
25303       |                           ^~~~~~~~~~~~~~~~~~~~~~~~~~
```

mboisson, 2021-08-03:
Ok, looks like I need `rendering3-3.5.0`, rather than `3.4.0.

closed 2021-08-03 by mboisson

chapulina, 2021-08-03:
See https://github.com/ignitionrobotics/ign-gui/pull/264 for bumping the required version in the code and https://github.com/ignition-release/ign-gui3-release/pull/5 for the debian packages. Ideally the installation / compilation should end early when a new dependency version is needed, instead of resulting in failures in the middle of the compilation.