gazebosim/gz-gui #283: GuiRunner error: failed to get plugin

ggiulian, 2021-09-15:
Whenever I run any gazebo command which is for example `ign gazebo empty.sdf` or i`gn gazebo shapes.sdf` this error appears. For now, it is not giving too many problems because the simulation is not affected. But for future applications I don't know if it will be an inconvenience

## Environment
* OS Version: Ubuntu 20.04 
* Source or binary build?
  source, Ignition Edifice and Galactic
* If this is a GUI or sensor rendering bug, describe your GPU and rendering system. Otherwise delete this section. 
    - Rendering plugin: [ogre | ogre2]. 
      - [ ] Sensor rendering error. 
      - [ ] GUI rendering error. 
    - EGL headless mode:
      - [ ] Running in EGL headless mode 
    - Generally, mention all circumstances that might affect rendering capabilities: 
      - [ ] running on a dual GPU machine (integrated GPU + discrete GPU)
      - [ ] running on a multi-GPU machine (it has multiple discrete GPUs)
      - [ ] running on real hardware
      - [ ] running in virtual machine
      - [ ] running in Docker/Singularity
      - [ ] running remotely (e.g. via SSH)
      - [ ] running in a cloud
      - [ ] using VirtualGL, XVFB, Xdummy, XVNC or other indirect rendering utilities
      - [ ] GPU is concurrently used for other tasks
        - [ ] desktop acceleration
        - [ ] video decoding (i.e. a playing Youtube video)
        - [ ] video encoding
        - [ ] CUDA/ROCm computations (Tensorflow, Torch, Caffe running)
        - [ ] multiple simulators running at the same time
      - [ ] other...
    - Rendering system info:
      -Linux:
         VGA: 00:02.0 VGA compatible controller [0300]: Intel Corporation UHD Graphics 630 (Mobile) [8086:3e9b]
          GLX: direct rendering: Yes
                   Preferred profile: core (0x1)
                   Max core profile version: 4.6
                   Max compat profile version: 4.6
                   Max GLES1 profile version: 1.1
                   Max GLES[23] profile version: 3.2
                  OpenGL vendor string: Intel
                  OpenGL renderer string: Mesa Intel(R) UHD Graphics 630 (CFL GT2)
                  OpenGL core profile version string: 4.6 (Core Profile) Mesa 21.0.3
                  OpenGL core profile shading language version string: 4.60
                  OpenGL core profile context flags: (none)
                  OpenGL core profile profile mask: core profile
                  OpenGL version string: 4.6 (Compatibility Profile) Mesa 21.0.3
                  OpenGL shading language version string: 4.60
                  OpenGL context flags: (none)
                  OpenGL profile mask: compatibility profile
                  OpenGL ES profile version string: OpenGL ES 3.2 Mesa 21.0.3
                  OpenGL ES profile shading language version string: OpenGL ES GLSL ES 3.20
    
<details>
  
```
[GUI] [Err] [GuiRunner.cc:158] Failed to get plugin [plugin75964]
[GUI] [Err] [GuiRunner.cc:158] Failed to get plugin [plugin72555]
[GUI] [Err] [GuiRunner.cc:158] Failed to get plugin [plugin82393]
[GUI] [Err] [GuiRunner.cc:158] Failed to get plugin [plugin35474]
[GUI] [Err] [GuiRunner.cc:158] Failed to get plugin [plugin55424]
[GUI] [Err] [GuiRunner.cc:158] Failed to get plugin [plugin41785]
[GUI] [Err] [GuiRunner.cc:158] Failed to get plugin [plugin9093]
[GUI] [Err] [GuiRunner.cc:158] Failed to get plugin [plugin16262]
```

</details>

## Description
* Expected behavior: No Gui error
* Actual behavior: Gui error, failed to get plugin

## Output

chapulina, 2021-09-15:
That error should be harmless and will disappear once we release https://github.com/ignitionrobotics/ign-gazebo/pull/944 for Fortress. I'll keep this issue open until we do so.

chapulina, 2021-11-17:
Ignition Gazebo 6.2 has been released.

closed 2021-11-17 by chapulina