gazebosim/gz-gui #289: Backporting TransportPlotting plugin to citadel

francocipollone, 2021-09-23:
### Context

The plotting feature was implemented for `ign-gui4` (dome) and then forward-ported to newer releases, however, it was never backported to `ign-gui3`(citadel) which is a LTS.

### Implementation

I tried backporting it to ign-gui3. Here is the commit/branch: https://github.com/ignitionrobotics/ign-gui/commit/1a4d323cebff8d5b2e1eca2e4ef47f0eb8eaaacc

And it seems to be working correctly:

[image]

Note: I had started following the instructions as it is explained in the `contributions` section about what is the correct way to backport a feature, which is using cherry-pick and fixing the conflicts. However, this feature was added in several commits and given it was added a year ago, some changes/bug-fixes affected those files afterwards so that method wasn't that smooth.

If you give me the thumbs up and there is nothing that prevent us to backporting it I will go ahead and create PR based on the branch I shared.

chapulina, 2021-09-23:
Thanks for looking into it! The problematic bit is this change:

```diff
-    class IGNITION_GUI_VISIBLE Application : public QGuiApplication
+    class IGNITION_GUI_VISIBLE Application : public QApplication
```

This breaks ABI and can't be done in a stable distribution.

francocipollone, 2021-09-23:
> Thanks for looking into it! The problematic bit is this change:
> 
> ```diff
> -    class IGNITION_GUI_VISIBLE Application : public QGuiApplication
> +    class IGNITION_GUI_VISIBLE Application : public QApplication
> ```
> 
> This breaks ABI and can't be done in a stable distribution.

Thanks for the quick reply! 
Oh you are right it can't be ported! Ok then, downstream we'll probably need to migrate to a newer release then.
(`Fortress` is candidate given that it is LTS, good timing).

Thanks again. Feel free to close the issue.

closed 2021-09-23 by chapulina