gazebosim/gz-gui #291: Non desired windows opening alongside ignition GUI

matosinho, 2021-09-25:
## Environment
* OS Version: Ubuntu 20.04
* Source or binary build?
  * Source
  * main (Fortress)
  * commit: 653b0fb58abd6ea8b3e3937e43d5aee509e125fc
* If this is a GUI or sensor rendering bug, describe your GPU and rendering system. Otherwise delete this section. 
    - Rendering plugin: default.
      - [ ] Sensor rendering error.
      - [x] GUI rendering error. 
    - EGL headless mode:
      - [ ] Running in EGL headless mode 
    - Generally, mention all circumstances that might affect rendering capabilities: 
      - [x] running on a dual GPU machine (integrated GPU + discrete GPU)
      - [ ] running on a multi-GPU machine (it has multiple discrete GPUs)
      - [ ] running on real hardware
      - [ ] running in virtual machine
      - [ ] running in Docker/Singularity
      - [ ] running remotely (e.g. via SSH)
      - [ ] running in a cloud
      - [ ] using VirtualGL, XVFB, Xdummy, XVNC or other indirect rendering utilities
      - [ ] GPU is concurrently used for other tasks
        - [ ] desktop acceleration
        - [ ] video decoding (i.e. a playing Youtube video)
        - [ ] video encoding
        - [ ] CUDA/ROCm computations (Tensorflow, Torch, Caffe running)
        - [ ] multiple simulators running at the same time
      - [ ] other...
    - Rendering system info:
      - On Linux, provide the outputs of the following commands:
      0a:00.0 VGA compatible controller [0300]: NVIDIA Corporation TU106 [GeForce RTX 2060 SUPER] [10de:1f06] (rev a1)
      direct rendering: Yes
      OpenGL vendor string: NVIDIA Corporation
      OpenGL renderer string: GeForce RTX 2060 SUPER/PCIe/SSE2
      OpenGL core profile version string: 4.6.0 NVIDIA 460.91.03
      OpenGL core profile shading language version string: 4.60 NVIDIA
      OpenGL core profile context flags: (none)
      OpenGL core profile profile mask: core profile
      OpenGL version string: 4.6.0 NVIDIA 460.91.03
      OpenGL shading language version string: 4.60 NVIDIA
      OpenGL context flags: (none)
      OpenGL profile mask: (none)
      OpenGL ES profile version string: OpenGL ES 3.2 NVIDIA 460.91.03
      OpenGL ES profile shading language version string: OpenGL ES GLSL ES 3.20

<details>  

```
19:29:01: Creating resource group General
19:29:01: Creating resource group Internal
19:29:01: Creating resource group Autodetect
19:29:01: SceneManagerFactory for type 'DefaultSceneManager' registered.
19:29:01: Registering ResourceManager for type Material
19:29:01: Registering ResourceManager for type Mesh
19:29:01: Registering ResourceManager for type Skeleton
19:29:01: MovableObjectFactory for type 'ParticleSystem' registered.
19:29:01: ArchiveFactory for archive type FileSystem registered.
19:29:01: ArchiveFactory for archive type Zip registered.
19:29:01: ArchiveFactory for archive type EmbeddedZip registered.
19:29:01: DDS codec registering
19:29:01: FreeImage version: 3.18.0
19:29:01: This program uses FreeImage, a free, open source image library supporting all common bitmap formats. See http://freeimage.sourceforge.net for details
19:29:01: Supported formats: bmp,ico,jpg,jif,jpeg,jpe,jng,koa,iff,lbm,mng,pbm,pbm,pcd,pcx,pgm,pgm,png,ppm,ppm,ras,tga,targa,tif,tiff,wap,wbmp,wbm,psd,psb,cut,xbm,xpm,gif,hdr,g3,sgi,rgb,rgba,bw,exr,j2k,j2c,jp2,pfm,pct,pict,pic,3fr,arw,bay,bmq,cap,cine,cr2,crw,cs1,dc2,dcr,drf,dsc,dng,erf,fff,ia,iiq,k25,kc2,kdc,mdc,mef,mos,mrw,nef,nrw,orf,pef,ptx,pxn,qtk,raf,raw,rdc,rw2,rwl,rwz,sr2,srf,srw,sti,x3f,webp,jxr,wdp,hdp
19:29:01: PVRTC codec registering
19:29:01: ETC codec registering
19:29:01: Registering ResourceManager for type HighLevelGpuProgram
19:29:01: Registering ResourceManager for type Compositor
19:29:01: MovableObjectFactory for type 'Entity' registered.
19:29:01: MovableObjectFactory for type 'Light' registered.
19:29:01: MovableObjectFactory for type 'BillboardSet' registered.
[... 228 lines cut ...]
19:29:01: All done
19:29:01: Added resource location '/home/matosinho/dev/ROS2/brov/ignition/install/share/ignition/ignition-rendering5/ogre/media/rtshaderlib150' of type 'FileSystem' to resource group 'General'
19:29:01: Creating resource group ShaderGeneratorResourceGroup
19:29:01: Added resource location '/home/matosinho/.ignition/rendering/ogre-rtshader/matosinho-rtshaderlibcache/' of type 'FileSystem' to resource group 'ShaderGeneratorResourceGroup'
19:29:01: Texture: spot_shadow_fade.png: Loading 1 faces(PF_R8G8B8,128x128x1) with 5 hardware generated mipmaps from Image. Internal format is PF_X8R8G8B8,128x128x1.
```

</details>

## Description
* Expected behavior: 
  * Upon running `ign gazebo`, the GUI should open with the entity tree and component on the right and manipulating tools on top.
* Actual behavior: 
  * Upon running `ign gazebo`, the GUI opens correctly, but it also open another window, which is very small and hard to notice.

## Steps to reproduce
1. Run `ign gazebo` in any scene or just without loading scenes
2. DON'T maxmize 
3. Search for the easter egg VERY thin wirndow close to GUI title

## Output
* Screenshot of the non desired window upon GUI opening, it is REALLY thin.
[image]
* Screenshot of the non desired window after expanding it
[image]

ahcorde, 2021-09-28:
these three PR should fix your issue:

 - https://github.com/ignitionrobotics/ign-gui/pull/292
 - https://github.com/ignitionrobotics/ign-rendering/pull/436
 - https://github.com/ignitionrobotics/ign-gazebo/pull/1063

ahcorde, 2021-10-01:
All PRs are merged, this should be fixed with the new releases. Feel free to reopen it if the issue persists

closed 2021-10-01 by ahcorde

matosinho, 2021-10-01:
Everything works! I've commented in the PRs. I tested them.