gazebosim/gz-gui #298: Make camera tracking configurable from startup

chapulina, 2021-10-01:
When we migrated camera tracking functionality from `GzScene3D` to `CameraTracking`, we forgot to port the ability of configuring it at startup.

## Desired behavior

Camera tracking is configurable from XML.

## Alternatives considered

N/A

## Implementation suggestion

Here's the code that needs to be ported:

https://github.com/ignitionrobotics/ign-gazebo/blob/fc2a98ec5e1b149d773eeb6cae4a407bd82a81a2/src/gui/plugins/scene3d/Scene3D.cc#L2726-L2770

## Additional context

Camera tracking added in #226

GauravKumar9920, 2024-02-12:
can I work on this issue ?

nick4fake, 2024-02-27:
How can this be achieved with the current state?

samialperen, 2026-09-25:
I am gonna handle this  @azeey