gazebosim/gz-gui #308: MinimalScene: configurable near/far clip planes

scpeters, 2021-11-02:
## Desired behavior

Configuration of the clipping distance for the near/far clip planes was requested in https://github.com/osrf/gazebo/issues/2064. Support was added for configuring these parameters via the gzclient GUI in [BitBucket pull request 2470](https://osrf-migration.github.io/gazebo-gh-pages/#!/osrf/gazebo/pull-requests/2470), though support for setting the parameter via a world file was requested but not implemented.

I request that we add the near/far clipping distance as parameters for the MinimalScene camera.

## Alternatives considered

The parameters could be configured interactively via a GUI widget or an ignition-transport interface, but I would prefer specifying it in XML.

## Implementation suggestion

The following is an example of the current XML configuration parameters for a `MinimalScene` plugin instance:
~~~
    <plugin filename="MinimalScene">
      <engine>ogre</engine>
      <scene>banana</scene>
      <ambient_light>1.0 0 0</ambient_light>
      <background_color>0 1 0</background_color>
      <camera_pose>1 2 3 0 0 1.57</camera_pose>
    </plugin>
~~~

I would propose adding the following parameters to mirror the SDFormat camera clipping distance parameter structure:

~~~
      <camera_clip>
        <near>0.1</near>
        <far>5000</far>
      </camera_pose>
~~~

~~~
    <plugin filename="MinimalScene">
      <engine>ogre</engine>
      <scene>banana</scene>
      <ambient_light>1.0 0 0</ambient_light>
      <background_color>0 1 0</background_color>

      <camera_clip>
        <near>0.1</near>
        <far>5000</far>
      </camera_pose>

      <camera_pose>1 2 3 0 0 1.57</camera_pose>
    </plugin>
~~~

## Additional context

Although setting these parameters from SDFormat was never implemented in gazebo, a Gazebo plugin for this purpose was created in the following public repository:

* https://github.com/nasa/irg_open/tree/ros2/irg_gazebo_plugins/src/IRGNearFarClipPlugin

scpeters, 2021-11-02:
I've added the Fortress label since `MinimalScene` as added in fortress

chapulina, 2021-11-02:
> a Gazebo plugin for this purpose was created in the following public repository:

Another reference, the SimSlides plugin also has an implementation for both Gazebo classic and Ignition:

https://github.com/chapulina/simslides/blob/1dcd393fb9622043cdf90c68ade8e8af5d48195c/common/Common.cc#L24-L35

jennuine, 2021-11-03:
PR #309 allows you to configure near/far clip distance in xml. Should I work on extending ign-gazebo's view angle plugin to configure it through the GUI?

scpeters, 2021-11-03:
> PR #309 allows you to configure near/far clip distance in xml. Should I work on extending ign-gazebo's view angle plugin to configure it through the GUI?

sounds good to me 👍

iche033, 2021-11-03:
related issue: https://github.com/ignitionrobotics/ign-gazebo/issues/1154

PR #309 (2021-11-03): Set near/far camera clipping distance
# 🎉 New feature

Closes #308

## Summary
Added ability to configure the user camera's near/far clipping distance in the `MinimalScene` plugin from xml. If none is set, the default near `0.01` and far `1000` is used.

## Test it

Add `<camera_clip>` to the `MinimalScene` plugin config. For example,
```xml
<camera_clip>
  <near>0.1</near>
  <far>5000</far>
</camera_clip>
```

to [minimal_scene.sdf](https://github.com/ignitionrobotics/ign-gazebo/blob/35ec2a75f6d0522bb840e0b704768864cfeb4ec4/examples/worlds/minimal_scene.sdf#L30-L42)

## Checklist
- [x] Signed [... cut]
closed 2021-11-03 by jennuine