gazebosim/gz-gui #310: Provide reusable QML components with common widgets

chapulina, 2021-11-04:
Widgets created with Ignition GUI often have duplicate code for interfaces to vectors, poses, colors, etc. It would be easier to construct various plugins if we offered these building blocks out of the box. It would also help users to write widgets that have the same look and feel.

## Desired behavior

When I'm writing a plugin, I'd like to import Ignition GUI's resources and immediately include common components. For example:

```
...
import "qrc:/qml"
...
IgnPose3d {
  id: myPose
  xValue: posX
  ...
}
```

## Alternatives considered

Keep using custom components for each plugin.

## Implementation suggestion

See the implementation for [IgnSpinBox](https://github.com/ignitionrobotics/ign-gui/blob/ign-gui6/include/ignition/gui/qml/IgnSpinBox.qml), which is currently used in multiple plugins.

## Additional context

Igntiion Gazebo's [ComponentInspector](https://github.com/ignitionrobotics/ign-gazebo/tree/ign-gazebo6/src/gui/plugins/component_inspector) plugin already implements a lot of these. The work should consist in porting them to `ign-gui` and making them as general as possible.

zflat, 2021-11-22:
I am pretty sure that if a `qmldir` file (https://doc.qt.io/archives/qt-6.0/qtqml-modules-qmldir.html) was added to the `include/ignition/gui/qml` directory and also added to the `resources.qrc` the `qrc` path to can be added to the QML load path. Then common components can be loaded with import statements as QML modules. 

However, with QML modules, the name of the folder containing the `qmldir` is the name of the module so it would make sense to change QML files to be located at something like `include/ignition/gui/qml/ignition_gui_common` and put the `qmldir` file into the `ignition_gui_common` directory.

With the right setup, this wold be possible from QML:
```
import ignitiion_gui_common 1.0 as CommonComponents
Item {
  CommonComponents.SpinBox{
    //...
  }
  //...
}
```

Basically, use `qmldir` files to define importable QML modules.

zflat, 2021-11-24:
@chapulina I just opened a draft pull request with changes like the ones I described in https://github.com/ignitionrobotics/ign-gui/issues/310#issuecomment-975089924

If this looks like the right approach, more files can be added to the QML module and import statements can be updated where the module will be used.

chapulina, 2022-07-27:
We got reusable vector, pose and color widgets. That's enough to close this ticket for now. As we develop more plugins, if we identify widgets that are being recreated multiple times, we should consider following the pattern of the new `Gz` widgets and create reusable components for them.

PR #319 (2021-12-02): use qmldir to define QML module with IgnSpinBox

# 🎉 New feature

Closes #310 

## Summary

This change adds a qmldir file that defines an importable QML module for common QML components implemented in `include/ignition/gui/qml/`.

With this change, it will be possible to import QML components using module import statements instead of file path.

Updating import statements will look like this:

```diff
-import "qrc:/qml"
+import ignition.gui 1.0
```

## Test it

* Build and run the application `ign run`
* Add the "Teleop" plugin
* Verify that the plugin loads and that the `IgnSpinBox` is part  [... cut]
closed 2022-07-27 by chapulina