gazebosim/gz-gui #314: Add Metal support to MinimalScene and Qt Applications

srmainwaring, 2021-11-11:
## Desired behavior

The MinimalScene plugin is not currently supported on macOS. This feature request is to allow MinimalScene to work on macOS when using the `ogre2` render engine by adding support for Metal.

## Alternatives considered

#### Use OpenGL on macOS

Approaches to using OpenGL on maOS have been discussed in a number of issues:

- https://github.com/ignitionrobotics/ign-rendering/pull/463 investigates running `ign gui` and `ign gazebo` on macOS using ogre-next2.1 and OpenGL. There is a method that will work for Ignition Edifice but there are difficulties in supporting this approach in later releases.

- https://github.com/ignitionrobotics/ign-rendering/issues/422 examines options for running the ignition-rendering examples on macOS using ogre-next2.2. Continuing to use OpenGL on macOS requires fall-back options for OpenGL to be implemented in ogre-next2.2, some progress has been made on this but a full implementation is not straightforward. 

It was suggested that providing support for Metal in Ignition may require less work than providing fall-back support for OpenGL and has the advantage of being supported by Apple.

## Implementation suggestion

### Preparation

Preparatory work to support Metal in ignition-rendering is included in the following PRs:

- https://github.com/ignitionrobotics/ign-rendering/pull/463
- https://github.com/ignitionrobotics/ign-rendering/pull/467
- https://github.com/ignitionrobotics/ign-rendering/pull/478

The upstream interface required from Ogre is available in:

https://github.com/OGRECave/ogre-next/pull/237

### Prototype example

A possible approach is prototyped in the `simple_demo_qml` example which has been extended to provide support for either OpenGL or Metal in this branch:

https://github.com/srmainwaring/ign-rendering/tree/feature/metal-simple_demo_qml

The method involves using the Qt scene graph backend for Metal provided by setting

```c++
QQuickWindow::setSceneGraphBackend(QSGRendererInterface::MetalRhi);
```

and using the Metal render system when loading the `ogre2` render engine.

Code specific to Metal must be compiled as Objective-C/C++. To prevent macOS specific code in the interface while retaining the option to still use OpenGL on macOS, a number of small interface classes have been introduced that allow the classes `IgnRenderer`, `RenderThread`, `TextureNode` and `RenderWindowItem`  in the MinimalScene plugin to dynamically load the appropriate render system.

### Draft implementation

A draft implementation that allows Metal to be used with `ign gui` and `ign gazebo` is available in the following branches:

- https://github.com/srmainwaring/ign-gui/tree/feature/ign-gui7-metal
- https://github.com/srmainwaring/ign-gazebo/tree/feature/ign-gazebo7-metal

The application in the draft hardcodes the render system to be Metal, however the MinimalScene plugin defaults to using OpenGL and uses the plugin XML to override the render system with the element:

```xml
<plugin filename="MinimalScene" name="3D View">
  <ignition-gui>
    <title>3D View</title>
    <property type="bool" key="showTitleBar">false</property>
    <property type="string" key="state">docked</property>
  </ignition-gui>
  <engine>ogre2</engine>
  <scene>scene</scene>
  <ambient_light>0.4 0.4 0.4</ambient_light>
  <background_color>0.8 0.8 0.8</background_color>
  <camera_pose>-6 0 6 0 0.5 0</camera_pose>
  <render_system>metal</render_system>
</plugin>
```

Mixing the OpenGL render system and Qt Metal scene graph back end is not permitted.

## Additional context

Ignition Gazebo on macOS with Metal, including GPU sensor support, is working using the branches posted above.

The following examples use Ignition on macOS / Metal with ArduPilot SITL and an extension of Brian Gerkey's port of the [ArduPilotGazebo plugin to Ignition](https://github.com/ArduPilot/ardupilot_gazebo/tree/ignition-garden) (for further details on this usage see [Ignition Gazebo + SITL support](https://discuss.ardupilot.org/t/ignition-gazebo-sitl-support/76120))

[image]

[image]

ahcorde, 2021-11-19:
I added `<render_system>metal</render_system>` in my `gui.config` as you suggested [here](https://github.com/ignitionrobotics/ign-rendering/pull/463#issuecomment-973979017). 

I can see this trace: `[Dbg] [Application.cc:105] Qt using Metal rendering interface` but it's still not working. uhhmmm any other tip ?

```bash 
Msg] Ignition Gazebo GUI    v7.0.0~pre1
[Dbg] [Application.cc:87] Initializing application.
[Dbg] [Application.cc:105] Qt using Metal rendering interface
[GUI] [Dbg] [Application.cc:456] Create main window
[GUI] [Wrn] [Application.cc:698] [QT] Populating font family aliases took 655 ms. Replace uses of missing font family "Roboto" with one that exists to avoid this cost. 
[GUI] [Wrn] [Application.cc:698] [QT] qrc:/qml/StyleDialog.qml:112:3: QML Connections: Implicitly defined onFoo properties in Connections are deprecated. Use this syntax instead: function onFoo(<arguments>) { ... }
[GUI] [Wrn] [Application.cc:698] [QT] qrc:/qml/StyleDialog.qml:105:3: QML Connections: Implicitly defined onFoo properties in Connections are deprecated. Use this syntax instead: function onFoo(<arguments>) { ... }
[GUI] [Wrn] [Application.cc:698] [QT] qrc:/qml/StyleDialog.qml:98:3: QML Connections: Implicitly defined onFoo properties in Connections are deprecated. Use this syntax instead: function onFoo(<arguments>) { ... }
[GUI] [Wrn] [Application.cc:698] [QT] qrc:qml/Main.qml:83:3: QML Connections: Implicitly defined onFoo properties in Connections are deprecated. Use this syntax instead: function onFoo(<arguments>) { ... }
[GUI] [Wrn] [Application.cc:698] [QT] qrc:/qml/PluginMenu.qml:27:3: QML Connections: Implicitly defined onFoo properties in Connections are deprecated. Use this syntax instead: function onFoo(<arguments>) { ... }
[GUI] [Dbg] [PathManager.cc:66] Requesting resource paths through [/gazebo/resource_paths/get]
[GUI] [Wrn] [Application.cc:698] [QT] file::/Gazebo/GazeboDrawer.qml:241:3: QML Connections: Implicitly defined onFoo properties in Connections are deprecated. Use this syntax instead: function onFoo(<arguments>) { ... }
[GUI] [Wrn] [Application.cc:698] [QT] file::/Gazebo/GazeboDrawer.qml:147:3: QML Dialog: Binding loop detected for property "implicitHeight"
[GUI] [Wrn] [Application.cc:698] [QT] file::/Gazebo/GazeboDrawer.qml:147:3: QML Dialog: Binding loop detected for property "implicitHeight"
[GUI] [Dbg] [Gui.cc:149] GUI requesting list of world names. The server may be busy downloading resources. Please be patient.
[GUI] [Dbg] [PathManager.cc:55] Received resource paths.
[GUI] [Dbg] [Gui.cc:208] Requesting GUI from [/world/default/gui/info]...
[GUI] [Dbg] [GuiRunner.cc:110] Requesting initial state from [/world/default/state]...
[GUI] [Msg] Loading config [/Users/ahcorde/.ignition/gazebo/7/gui.config]
[GUI] [Dbg] [Application.cc:336] Loading plugin [MinimalScene]
[GUI] [Dbg] [Application.cc:692] [QT] Creating ign-renderering interface for OpenGL
[GUI] [Dbg] [Application.cc:692] [QT] Creating ign-renderering interface for OpenGL
[GUI] [Dbg] [Application.cc:692] [QT] Creating render thread interface for OpenGL
[GUI] [Dbg] [Application.cc:692] [QT] Creating ign-renderering interface for Metal
[GUI] [Dbg] [Application.cc:692] [QT] Creating render thread interface for Metal
[GUI] [Msg] Added plugin [3D View] to main window
[GUI] [Msg] Loaded plugin [MinimalScene] from path [/Users/ahcorde/ignition_fortress/install/lib/ign-gui-7/plugins/libMinimalScene.dylib]
[GUI] [Dbg] [Application.cc:336] Loading plugin [EntityContextMenuPlugin]
[GUI] [Wrn] [Application.cc:698] [QT] file::/EntityContextMenuPlugin/EntityContextMenuPlugin.qml:33:5: QML Connections: Implicitly defined onFoo properties in Connections are deprecated. Use this syntax instead: function onFoo(<arguments>) { ... }
[GUI] [Msg] Added plugin [EntityContextMenu] to main window
[GUI] [Msg] Loaded plugin [EntityContextMenuPlugin] from path [/Users/ahcorde/ignition_fortress/install/lib/ign-gazebo-7/plugins/gui/libEntityContextMenuPlugin.dylib]
[GUI] [Dbg] [Application.cc:336] Loading plugin [GzSceneManager]
[GUI] [Msg] Added plugin [Scene Manager] to main window
[GUI] [Msg] Loaded plugin [GzSceneManager] from path [/Users/ahcorde/ignition_fortress/install/lib/ign-gazebo-7/plugins/gui/libGzSceneManager.dylib]
[GUI] [Dbg] [Application.cc:336] Loading plugin [InteractiveViewControl]
[GUI] [Msg] Camera view controller topic advertised on [/gui/camera/view_control]
[GUI] [Msg] Added plugin [] to main window
[GUI] [Msg] Loaded plugin [InteractiveViewControl] from path [/Users/ahcorde/ignition_fortress/install/lib/ign-gui-7/plugins/libInteractiveViewControl.dylib]
[GUI] [Dbg] [Application.cc:336] Loading plugin [CameraTracking]
[GUI] [Msg] Added plugin [Camera tracking] to main window
[GUI] [Msg] Loaded plugin [CameraTracking] from path [/Users/ahcorde/ignition_fortress/install/lib/ign-gui-7/plugins/libCameraTracking.dylib]
[GUI] [Dbg] [Application.cc:336] Loading plugin [MarkerManager]
[GUI] [Msg] Listening to stats on [/world/default/stats]
[GUI] [Msg] Added plugin [Marker Manager] to main window
[GUI] [Msg] Loaded plugin [MarkerManager] from path [/Users/ahcorde/ignition_fortress/install/lib/ign-gui-7/plugins/libMarkerManager.dylib]
[GUI] [Dbg] [Application.cc:336] Loading plugin [SelectEntities]
[GUI] [Msg] Added plugin [Select entities] to main window
[GUI] [Msg] Loaded plugin [SelectEntities] from path [/Users/ahcorde/ignition_fortress/install/lib/ign-gazebo-7/plugins/gui/libSelectEntities.dylib]
[... cut ...]
#2    Object "libignition-tools-backward.dylib", at 0x1058af5a8, in backward::StackTraceImpl<backward::system_tag::darwin_tag>::load_from(void*, unsigned long, void*, void*) + 56
#1    Object "libignition-tools-backward.dylib", at 0x1058af701, in backward::StackTraceImpl<backward::system_tag::darwin_tag>::load_here(unsigned long, void*, void*) + 129
#0    Object "libignition-tools-backward.dylib", at 0x1058b0f60, in unsigned long backward::details::unwind<backward::StackTraceImpl<backward::system_tag::darwin_tag>::callback>(backward::StackTraceImpl<backward::system_tag::darwin_tag>::callback, unsigned long) + 32
Segmentation fault: 11
```

srmainwaring, 2021-11-19:
It's a bit odd. Something is still attempting to load the `RenderSystem_GL3Plus` render window. Can you try disabling all the plugins in `gui.config` except MinimalScene and run an empty ignition gazebo server session.

```bash
$ ign gazebo -v4 -s -r
```
I always need to exclude the transform control, as there are issue with it on load that I have not yet resolved. Other plugins seem to load ok from start but eliminating them may help narrow down the problem.

srmainwaring, 2021-11-19:
@ahcorde - thanks for your patience. I'm not sure what's going on so perhaps take a step back to check that ign-rendering with QML is working. Ah - I've just realised that you may not have all the metal dependencies if you are using ign-rendering/main (it has the changed interface to get the Metal GPU texture but not the changes from the render engine to use them).

Here is the current list of libraries I am using to run Gazebo (ign-rendering and above):

- https://github.com/srmainwaring/ign-rendering/tree/feature/metal-simple_demo_qml
- https://github.com/srmainwaring/ign-gui/tree/feature/ign-gui7-metal
- https://github.com/ignitionrobotics/ign-sensors
- https://github.com/srmainwaring/ign-gazebo/tree/feature/ign-gazebo6_to_7-metal

```bash
$ colcon graph
ignition-cmake2       + ***************
ignition-tools         +    *  *.  ..**
ignition-math6          + *****.*******
ignition-utils1          +* *..**.*..**
ignition-common4          +  ** *******
ignition-msgs8             + * *   ****
sdformat12                  +   *   **.
ignition-fuel_tools7         +       *.
ignition-plugin1              + ****.**
ignition-transport11           +   ****
ignition-physics5               +    *.
ignition-rendering7              + ***.
ignition-tutorials                +    
ignition-gui7                      + **
ignition-sensors6                   +*.
ignition-gazebo7                     +*
ignition-launch5                      +
```

1. The `feature/metal-simple_demo_qml` includes the changes from `feature/ign-rendering6-metal-macos` and the interface change in `main`.
2. Branch feature/ign-gui7-metal includes additional events and feature/ign-gazebo6_to_7-metal is rebased on the forward port 6 to 7 that fixes some threading issues that were causing random segfaults in the GUI which macOS seems sensitive to. 

### `simple_demo_qml`

Check the `simple_demo_qml` example is working. The example uses a similar rendering workflow to ign-gui, so if this is not working then Gazebo is unlikely to work either:

```bash
# ./src/ign-rendering/examples/simple_demo_qml/build
$ cmake .. && make
$ ./simple_demo_qml metal
```
[image]

### `ign gazebo`

If `simple_demo_qml` is working, try the basic ign gazebo example again.

ahcorde, 2021-11-19:
The problem was the rendering branch, with the [branch](https://github.com/srmainwaring/ign-rendering/tree/feature/metal-simple_demo_qml) that you mentioned the `simple_demo_qml` code is not segfaulting but the window is completely white:

[image]

```
Setting RenderSystem: "opengl"
Setting RenderSystem: "metal"
Creating RenderThread for Metal render system

Initialising Metal RenderThread
[Msg] Loading plugin [ignition-rendering-ogre2]
Creating resource group General
Creating resource group Internal
Creating resource group Autodetect
SceneManagerFactory for type 'DefaultSceneManager' registered.
Registering ResourceManager for type Material
Registering ResourceManager for type Mesh
Registering ResourceManager for type Mesh2
Registering ResourceManager for type OldSkeleton
MovableObjectFactory for type 'ParticleSystem' registered.
ArchiveFactory for archive type FileSystem registered.
ArchiveFactory for archive type Zip registered.
ArchiveFactory for archive type EmbeddedZip registered.
DDS codec registering
FreeImage version: 3.18.0
This program uses FreeImage, a free, open source image library supporting all common bitmap formats. See http://freeimage.sourceforge.net for details
Supported formats: bmp,ico,jpg,jif,jpeg,jpe,jng,koa,iff,lbm,mng,pbm,pbm,pcd,pcx,pgm,pgm,png,ppm,ppm,ras,tga,targa,tif,tiff,wap,wbmp,wbm,psd,psb,cut,xbm,xpm,gif,hdr,g3,sgi,rgb,rgba,bw,exr,j2k,j2c,jp2,pfm,pct,pict,pic,3fr,arw,bay,bmq,cap,cine,cr2,crw,cs1,dc2,dcr,drf,dsc,dng,erf,fff,ia,iiq,k25,kc2,kdc,mdc,mef,mos,mrw,nef,nrw,orf,pef,ptx,pxn,qtk,raf,raw,rdc,rw2,rwl,rwz,sr2,srf,srw,sti,x3f,webp,jxr,wdp,hdp
OITD codec registering
Registering ResourceManager for type HighLevelGpuProgram
MovableObjectFactory for type 'Decal' registered.
MovableObjectFactory for type 'InternalCubemapProbe' registered.
MovableObjectFactory for type 'Entity' registered.
MovableObjectFactory for type 'Item' registered.
MovableObjectFactory for type 'Light' registered.
MovableObjectFactory for type 'Rectangle2Dv2' registered.
MovableObjectFactory for type 'BillboardSet' registered.
MovableObjectFactory for type 'ManualObject2' registered.
MovableObjectFactory for type 'BillboardChain' registered.
MovableObjectFactory for type 'RibbonTrail' registered.
MovableObjectFactory for type 'WireAabb' registered.
*-*-* OGRE Initialising
*-*-* Version 2.2.6 (Cerberus)
OverlayElementFactory for type Panel registered.
OverlayElementFactory for type BorderPanel registered.
OverlayElementFactory for type TextArea registered.
Registering ResourceManager for type Font
Loading library /usr/local/opt/ogre2.2/lib/OGRE-2.2/OGRE/RenderSystem_GL3Plus.dylib
Installing plugin: GL 3+ RenderSystem
OpenGL 3+ Rendering Subsystem created.
Plugin successfully installed
Loading library /usr/local/opt/ogre2.2/lib/OGRE-2.2/OGRE/Plugin_ParticleFX.dylib
Installing plugin: ParticleFX
Particle Emitter Type 'Point' registered
Particle Emitter Type 'Box' registered
Particle Emitter Type 'Ellipsoid' registered
Particle Emitter Type 'Cylinder' registered
Particle Emitter Type 'Ring' registered
Particle Emitter Type 'HollowEllipsoid' registered
Particle Affector Type 'LinearForce' registered
Particle Affector Type 'ColourFader' registered
Particle Affector Type 'ColourFader2' registered
Particle Affector Type 'ColourImage' registered
Particle Affector Type 'ColourInterpolator' registered
Particle Affector Type 'Scaler' registered
Particle Affector Type 'Rotator' registered
Particle Affector Type 'DirectionRandomiser' registered
Particle Affector Type 'DeflectorPlane' registered
Plugin successfully installed
Loading library /usr/local/opt/ogre2.2/lib/OGRE-2.2/OGRE/RenderSystem_Metal.dylib
Installing plugin: Metal RenderSystem
[... 115 lines cut ...]
Can't assign material scene::Material(65511) because this Material does not exist. Have you forgotten to define it in a .material script?
Can't assign material scene::Material(65522) because this Material does not exist. Have you forgotten to define it in a .material script?
Can't assign material scene::Material(65500) because this Material does not exist. Have you forgotten to define it in a .material script?
Can't assign material scene::Material(65495) because this Material does not exist. Have you forgotten to define it in a .material script?
Can't assign material scene::Material(65495) because this Material does not exist. Have you forgotten to define it in a .material script?
Can't assign material scene::Material(65484) because this Material does not exist. Have you forgotten to define it in a .material script?
Can't assign material scene::Material(65480) because this Material does not exist. Have you forgotten to define it in a .material script?
Can't assign material scene::Material(65476) because this Material does not exist. Have you forgotten to define it in a .material script?
Can't assign material scene::Material(65484) because this Material does not exist. Have you forgotten to define it in a .material script?
Can't assign material scene::Material(65480) because this Material does not exist. Have you forgotten to define it in a .material script?
Can't assign material scene::Material(65476) because this Material does not exist. Have you forgotten to define it in a .material script?
Can't assign material scene::Material(65484) because this Material does not exist. Have you forgotten to define it in a .material script?
[... cut ...]
Can't assign material scene::Material(65448) because this Material does not exist. Have you forgotten to define it in a .material script?
Shader 100000000VertexShader_vs compiled successfully.
Shader 100000001VertexShader_vs compiled successfully.
Shader 100000001PixelShader_ps compiled successfully.
Shader 100000002VertexShader_vs compiled successfully.
Shader 100000002PixelShader_ps compiled successfully.
Shader 100000003VertexShader_vs compiled successfully.
Shader 100000003PixelShader_ps compiled successfully.
```

srmainwaring, 2021-11-19:
That's better! (sort of). This might be the aliasing issue we were seeing before on your MacBook vs my MacPro. Try setting the AntiAliasing in `IgnitionRendererMetal.mm` to either `0` or `4`.  

```
  CameraPtr camera = _scene->CreateCamera("camera");
  camera->SetLocalPosition(0.0, 0.0, 0.0);
  camera->SetLocalRotation(0.0, 0.0, 0.0);
  camera->SetImageWidth(800);
  camera->SetImageHeight(600);
  camera->SetAntiAliasing(2);
  camera->SetAspectRatio(800.0/600.0);
  camera->SetHFOV(IGN_PI / 2);
  root->AddChild(camera);
```

ahcorde, 2021-11-19:
I tried with `0`,  `1`, `4` and `8`. But it's still white

srmainwaring, 2021-11-19:
Ok - I'll need to see if I can replicate that on my MacBook. The only other thing I can think of at this stage is that your version of Ogre2.2 does not include the patch that actually makes the texture available (there is a default API in Ogre::TextureGPU that returns a null texture - so you wouldn't see a crash).

ahcorde, 2021-11-19:
My current Ogre2.2 version is this one (it's installed with brew):

```bash
osrf/simulation/ogre2.2: stable 2.2.6+20211021~312bf40 (bottled), HEAD
Scene-oriented 3D engine written in c++
https://www.ogre3d.org/
/usr/local/Cellar/ogre2.2/2.2.6+20211021~312bf40 (839 files, 16.0MB) *
  Poured from bottle on 2021-11-19 at 15:53:45
From: https://github.com/osrf/homebrew-simulation/blob/HEAD/Formula/ogre2.2.rb
License: MIT
==> Dependencies
Build: cmake ✘
Required: doxygen ✔, freeimage ✔, freetype ✔, libx11 ✔, libzzip ✔, rapidjson ✔, tbb ✔
==> Options
--HEAD
        Install HEAD version
```

By the way, thank you for your support and great work! I'm super excited to make it work!

srmainwaring, 2021-11-19:
That might be the problem.

This patch https://github.com/OGRECave/ogre-next/commit/b7187a55a9ad5ba65ed24d1c212d1749833923ac has been not been back-ported to the `ogre-next/v2-2` branch.

The `osrf/simulation/ogre2.2.rb` formula references: https://github.com/OGRECave/ogre-next/archive/312bf406a77244afe230930e67e3e5d52a119507.tar.gz which I've just checked does not include the changes to  `./RenderSystem/Metal/OgreMetalTextureGpu.h` and `./RenderSystem/Metal/OgreMetalTextureGpu.h`. There should be an override for the method `getCustomAttribute`.

There's two approaches - one would be to get the Ogre maintainers to backport the commit to `ogre-next/v2-2` and update the Formula, the other is to apply the patch directly in the Formula. The latter might be quicker for OSRF to implement.

srmainwaring, 2021-11-19:
Here you go: add these lines to the ogre2.2 Formula `/usr/local/Homebrew/Library/Taps/osrf/homebrew-simulation/Formula/ogre2.2.rb`:

```ruby
  patch do
    # implement override for MetalTextureGpu::getCustomAttribute
    url "https://github.com/OGRECave/ogre-next/commit/b7187a55a9ad5ba65ed24d1c212d1749833923ac.patch?full_index=1"
    sha256 "38975001bfa903194565ed0bf411cf29857cd5b2f0f71a651d64543f610c4ff6"
  end
```

Then reinstall `ogre2.2` building from source:

```bash
brew install --build-from-source ogre2.2
==> Downloading https://github.com/scpeters/ogre-next/commit/3486b772b35b3e5f8cf
Already downloaded: /Users/rhys/Library/Caches/Homebrew/downloads/affb4b48e6407594691f192f9711a1e382e176e98aa910fb664db37e99f0006f--3486b772b35b3e5f8cf6da5b7a41e6c195408d7d.patch
==> Downloading https://github.com/OGRECave/ogre-next/commit/b7187a55a9ad5ba65ed
Already downloaded: /Users/rhys/Library/Caches/Homebrew/downloads/fac46ab2da55d1627a4d1d69ba571df022fca8345e84d24c467e0b53ef6ed3c3--b7187a55a9ad5ba65ed24d1c212d1749833923ac.patch
==> Downloading https://github.com/OGRECave/ogre-next/archive/312bf406a77244afe2
Already downloaded: /Users/rhys/Library/Caches/Homebrew/downloads/4601c24fe78df4d35670cfb87e9c879040c189c5a3c4a12a5c885a124826859c--ogre-next-312bf406a77244afe230930e67e3e5d52a119507.tar.gz
==> Installing ogre2.2 from osrf/simulation
==> Patching
==> Applying 3486b772b35b3e5f8cf6da5b7a41e6c195408d7d.patch
patching file CMake/Utils/OgreConfigTargets.cmake
patching file CMakeLists.txt
==> Applying b7187a55a9ad5ba65ed24d1c212d1749833923ac.patch
patching file RenderSystems/Metal/include/OgreMetalTextureGpu.h
Hunk #1 succeeded at 102 (offset -4 lines).
patching file RenderSystems/Metal/src/OgreMetalTextureGpu.mm
Hunk #1 succeeded at 385 (offset -13 lines).
==> cmake .. -DOGRE_LIB_DIRECTORY=lib/OGRE-2.2 -DOGRE_BUILD_LIBS_AS_FRAMEWORKS=O
==> make install
🍺  /usr/local/Cellar/ogre2.2/2.2.6+20211021~312bf40: 840 files, 16.0MB, built in 1 minute
```

Where we can see the Metal textures have been patched. For reference here is a copy of the updated Formula:

[ogre2.2.rb.zip](https://github.com/ignitionrobotics/ign-gui/files/7571586/ogre2.2.rb.zip)

I've rebuilt ign-rendering against this version of ogre2.2 (replacing my from source build), and it is working.

iche033, 2021-11-19:
sure let's see if the patch can be backported upstream. If not, we can add the patch to the ogre 2.2 formula.

@darksylinc, can  OGRECave/ogre-next@b7187a5 be backported to the `v2-2` branch?

srmainwaring, 2021-11-30:
@ahcorde thanks for bumping ignitiion-rendering6 into main. I've added a [wiki page](https://github.com/srmainwaring/ign-rendering/wiki) to my fork of ignition-rendering documenting how to install ignition gazebo with Metal support (to save hunting through issues for the details).

I see there are [new instructions for installing ignition garden](https://ignitionrobotics.org/docs/garden/install_osx_src), so I'll check I've got the right dependencies and see if I can get the ign-gui and ign-gazebo changes into draft PRs for discussion.

darksylinc, 2022-08-07:
Is there a reason this ticket is still open? Is it not working?

I am developing the Vulkan backend and it's pretty much using @srmainwaring 's work for Metal.

One huge gotcha I detected is that this call:

`QQuickWindow::setSceneGraphBackend`

Needs to happen extremely early (before Qt has initialized its RHI) which makes it very hard to make an optional toggle in the XML because if I saw things correctly, the XML files are loaded too late.

Supporting Metal/Vulkan toggles are at CLI level or environment variables are a no-brainer since `Application::Application` in ign-gui/src/Application.cc has access to all the required info.

However toggles living XML files are a different problem.

I see that @srmainwaring 's work lacks a way of informing from the GUI to the MinimalScene plugin which backend is being used; which I am fixing via:

```cpp
Application::Application( ... )
{
....
this->dataPtr->mainWin->setProperty("renderEngineBackendApiName",
                                            "vulkan");
}
```

And then adding

srmainwaring, 2022-08-07:
> Is there a reason this ticket is still open? Is it not working? What's missing?

@darksylinc #323 partially closed this and the updates to the ogre2.2 formula in osrf/simulation complete this this for gz-gui / garden. It's all running and I've been using Metal on amd64 and arm64 macs for Gazebo work since the changes were merged, so I'll close this issue.

To use Metal with gz-sim we still need https://github.com/gazebosim/gz-sim/pull/1225 (or at least a minimal version that updates the ruby launch scripts so the gui can be started).

The command line change to select the backend would be a nice add - could this be factored out into a separate PR from the vulkan work?

closed 2022-08-07 by srmainwaring

darksylinc, 2022-08-07:
> The command line change to select the backend would be a nice add - could this be factored out into a separate PR from the vulkan work?

Right now I'm still figuring things out so I don't have time for. It's definitely not much work:

```diff
diff --git a/src/Application.cc b/src/Application.cc
index 8de4ea58..94f1dd9e 100644
--- a/src/Application.cc
+++ b/src/Application.cc
@@ -118,8 +118,34 @@ Application::Application(int &_argc, char **_argv, const WindowType _type)
   // format.setRenderableType(QSurfaceFormat::OpenGL);
   // QSurfaceFormat::setDefaultFormat(format);
 #else
-  // Otherwise use OpenGL
-  gzdbg << "Qt using OpenGL graphics interface" << std::endl;
+  // Otherwise use OpenGL (or Vulkan when supported and requested)
+
+  const bool useVulkan = false;  // TODO(anyone)
+
+  if (useVulkan)
+  {
+    gzdbg << "Qt using Vulkan graphics interface" << std::endl;
+
+#  ifdef GZ_USE_VULKAN_DEBUG_EXT
+    qputenv("QT_VULKAN_INSTANCE_EXTENSIONS",
+            "VK_EXT_debug_report;VK_EXT_debug_utils");
+#  endif
+    qputenv("QT_VULKAN_DEVICE_EXTENSIONS",
+            "VK_KHR_maintenance2;VK_EXT_shader_subgroup_vote;"
+            "VK_EXT_shader_viewport_index_layer;"
+#  ifdef GZ_USE_VULKAN_DEBUG_EXT
+            ";VK_EXT_debug_marker"
+#  endif
+    );
+
+#  if QT_VERSION >= QT_VERSION_CHECK(5, 15, 2)
+    QQuickWindow::setSceneGraphBackend(QSGRendererInterface::VulkanRhi);
+#  endif
+  }
+  else
+  {
+    gzdbg << "Qt using OpenGL graphics interface" << std::endl;
+  }
 #endif
 
   // Configure console
@@ -149,7 +175,17 @@ Application::Application(int &_argc, char **_argv, const WindowType _type)
   if (_type == WindowType::kMainWindow)
   {
     if (!this->InitializeMainWindow())
+    {
       gzerr << "Failed to initialize main window." << std::endl;
+    }
+    else
+    {
+      if (useVulkan)
+      {
+        this->dataPtr->mainWin->setProperty("renderEngineBackendApiName",
+                                            "vulkan");
+      }
+    }
   }
   else if (_type == WindowType::kDialog)
   {
diff --git a/src/Helpers.cc b/src/Helpers.cc
index 52c93a36..95b54a61 100644
--- a/src/Helpers.cc
+++ b/src/Helpers.cc
@@ -195,6 +195,20 @@ std::string gz::gui::renderEngineName()
   return renderEngineNameVariant.toString().toStdString();
 }
 
+/////////////////////////////////////////////////
+std::string gz::gui::renderEngineBackendApiName()
+{
+  auto win = App()->findChild<MainWindow *>();
+  if (nullptr == win)
+    return {};
+
+  auto renderEngineNameVariant = win->property("renderEngineBackendApiName");
+  if (!renderEngineNameVariant.isValid())
+    return {};
+
+  return renderEngineNameVariant.toString().toStdString();
+}
+
 /////////////////////////////////////////////////
 const QString gz::gui::qmlQrcImportPath()
 {
diff --git a/src/plugins/minimal_scene/MinimalScene.cc b/src/plugins/minimal_scene/MinimalScene.cc
index 72952db8..929c6be5 100644
--- a/src/plugins/minimal_scene/MinimalScene.cc
+++ b/src/plugins/minimal_scene/MinimalScene.cc
@@ -696,7 +696,19 @@ void GzRenderer::NewMouseEvent(const common::MouseEvent &_e)
 RenderThread::RenderThread()
 {
   // Set default graphics API to OpenGL
-  this->SetGraphicsAPI(rendering::GraphicsAPI::OPENGL);
+  const std::string backendApiName = gz::gui::renderEngineBackendApiName();
+  if (backendApiName == "vulkan")
+  {
+    this->SetGraphicsAPI(rendering::GraphicsAPI::VULKAN);
+  }
+  else if (backendApiName == "metal")
+  {
+    this->SetGraphicsAPI(rendering::GraphicsAPI::METAL);
+  }
+  else
+  {
+    this->SetGraphicsAPI(rendering::GraphicsAPI::OPENGL);
+  }
 
   RenderWindowItem::Implementation::threads << this;
   qRegisterMetaType<RenderSync*>("RenderSync*");
@@ -1243,6 +1255,20 @@ void MinimalScene::LoadConfig(const tinyxml2::XMLElement *_pluginElem)
                 << std::endl;
     }
 
+    const std::string backendApiName = gz::gui::renderEngineBackendApiName();
+    if (backendApiName == "vulkan")
+    {
+      renderWindow->SetGraphicsAPI(rendering::GraphicsAPI::VULKAN);
+    }
+    else if (backendApiName == "metal")
+    {
+      renderWindow->SetGraphicsAPI(rendering::GraphicsAPI::METAL);
+    }
+    else
+    {
+      renderWindow->SetGraphicsAPI(rendering::GraphicsAPI::OPENGL);
+    }
+
     elem = _pluginElem->FirstChildElement("graphics_api");
     if (nullptr != elem && nullptr != elem->GetText())
     {
```