gazebosim/gz-gui #358: What is the planned relationship between MinimumScene and TransportSceneManager?

Levi-Armstrong, 2022-02-21:
I noticed that MinimumScene has the following, but then I came across TransportSceneManager which seems to manage the same set of topics and service. Is the plan that this functionality will be removed from MinimumScene and TransportSceneManager should be used going forward or is there another intended purpose of TransportSceneManager?

``` c++
    /// \brief Scene service. If not empty, a request will be made to get the
    /// scene information using this service and the renderer will populate the
    /// scene based on the response data
    public: std::string sceneService;

    /// \brief Scene pose topic. If not empty, a node will subcribe to this
    /// topic to get pose updates of objects in the scene
    public: std::string poseTopic;

    /// \brief Ign-transport deletion topic name
    public: std::string deletionTopic;

    /// \brief Ign-transport scene topic name
    /// New scene messages will be published to this topic when an entities are
    /// added
    public: std::string sceneTopic;
```

ahcorde, 2022-02-22:
I think these methods and attributes are there for some historical reason. Because are not used at all. I created a [PR](https://github.com/ignitionrobotics/ign-gui/pull/359) to remove then

Levi-Armstrong, 2022-02-24:
Thank you for the support.

PR #359 (2022-03-01): Removed unused code in MinimalScene plugin

# 🦟 Bug fix

Fixes #358

## Summary

This code is not used and it should be part of the TransportSceneManager

## Checklist
- [x] Signed all commits for DCO
- [ ] Added tests
- [ ] Updated documentation (as needed)
- [ ] Updated migration guide (as needed)
- [x] `codecheck` passed (See [contributing](https://ignitionrobotics.org/docs/all/contributing#contributing-code))
- [x] All tests passed (See [test coverage](https://ignitionrobotics.org/docs/all/contributing#test-coverage))
- [x] While waiting for a review on your PR, please help review [another [... cut]
closed 2022-03-01 by ahcorde