gazebosim/gz-gui #396: Support qt6 before May 2023

scpeters, 2022-05-10:
## Desired behavior

The last version of qt5 listed as an LTS is 5.15, and its support window ends in May 2023:

* https://endoflife.date/qt

We should update the package to support Qt6.

## Alternatives considered

## Implementation suggestion

## Additional context

chapulina, 2022-05-11:
Thanks for bringing this up. It looks like Qt6 is only available on [Ubuntu from Jammy](https://packages.ubuntu.com/jammy/qt6-base-dev) and Qt5 is available on [Ubuntu from Bionic to Kinetic](https://packages.ubuntu.com/jammy/qtbase5-dev).

When it comes to Ubuntu, it looks to me that we could keep using Qt5 for Jammy even past its EOL date. But other platforms like Homebrew may be cutting support immediately when it EOLs - do you know what's happening there @scpeters ?

For stable versions (Citadel and Fortress), we shouldn't change the default Qt version. The most we can consider is also supporting compilation from source with Qt6. I wouldn't prioritize this work but we could review PRs from the community.

For Garden, since we'll be supporting both [Focal and Jammy](https://gazebosim.org/docs/garden/install#supported-platforms), we wouldn't be able to use only Qt6. So we either:

1. support 5 (for Focal) and 6 (for Jammy) side-by-side (this may add a lot of maintenance burden and slow down development)
1. stick to Qt5 (how does this affect Homebrew?)
1. drop Focal support and go full on Qt6 (this may not be too terrible to do for a non-LTS version)

We have lots of other things going on right now, so depending on how complicated the migration to Qt6 is, we may not have resources to put into it and may be forced to go with option 2.

jennuine, 2023-11-03:
Closing since migration is being tracked here https://github.com/gazebosim/gz-gui/issues/586

closed 2023-11-03 by jennuine