gazebosim/gz-gui #482: Can't open Nav Sat Map plugin (crashes)

ijnek, 2022-09-05:
## Environment
* OS Version: Ubuntu22.04
* Source or binary build? Binary (version: 6.6.1)

## Description
* Expected behavior: Nav Sat Map plugin opens correctly
* Actual behavior: Ignition Gazebo crashes with the following messages

```sh
[GUI] [Dbg] [Application.cc:426] Loading plugin [NavSatMap]
[GUI] [Wrn] [Application.cc:797] [QT] file::/NavSatMap/NavSatMap.qml:160:3: QML Connections: Implicitly defined onFoo properties in Connections are deprecated. Use this syntax instead: function onFoo(<arguments>) { ... }
[GUI] [Wrn] [Application.cc:797] [QT] QObject::connect(QQuickWindow, QDeclarativeGeoMap): invalid nullptr parameter
[GUI] [Msg] Added plugin [Navigation satellite map] to main window
[GUI] [Msg] Loaded plugin [NavSatMap] from path [/usr/lib/x86_64-linux-gnu/ign-gui-6/plugins/libNavSatMap.so]
[GUI] [Wrn] [Application.cc:797] [QT] QGeoTileProviderOsm: Tileserver disabled at  QUrl("http://maps-redirect.qt.io/osm/5.8/satellite")
[GUI] [Wrn] [Application.cc:797] [QT] QGeoTileFetcherOsm: all providers resolved
```

## Steps to reproduce

1. ``ign gazebo -v 4``
2. Go to add a Nav Sat Map plugin from the right-top menu
3. Crashes, when Nav Sat Map plugin is clicked

## Output
[image]

jennuine, 2022-09-06:
Hi @ijnek I have tried both fortress from source and binaries and am not able to reproduce the crash (I've tried launching `ign gazebo -v 4` then adding the plugin manually and also `ign gazebo -v 4 spherical_coordinates.sdf`). I suggest double checking your installation and make sure it's set up correctly.  An option is to uninstall then reinstall fortress: https://gazebosim.org/docs/fortress/install_ubuntu

jennuine, 2022-09-06:
I just realized you're on `jammy` not `focal`, re-opening issue since this wasn't tested appropriately

vincentrou, 2022-10-11:
I can confirm, I have the same issue with navsatmap gui plugin with same config (gazebo Fortress and Ubuntu 22.04)

azeey, 2023-03-07:
There's some discussion [here](https://www.qt.io/blog/2017/03/09/provisioning-openstreetmap-providers-in-qtlocation) about a change in map providers which might be causing the issue. I tried playing with the configuration options mentioned there and in https://forum.sailfishos.org/t/using-qml-map-osm-not-working/7261 with no luck

jennuine, 2023-03-07:
I was able to get a little further with the below patch but the GUI will eventually freeze. Unfortunately, I don't have bandwidth to continue looking at this but here are the steps in case anyone does (On Jammy, using Gazebo Garden):

1. Apply below patch, rebuild
2. `gz sim -v 4 -r spherical_coordinates.sdf`
3. Add the `Nav Sat Map` from the plugin menu
4. Translate the box somewhere, this is when the GUI freezes

```patch
diff --git a/src/plugins/navsat_map/NavSatMap.qml b/src/plugins/navsat_map/NavSatMap.qml
index 3e7659b2..3145e1ed 100644
--- a/src/plugins/navsat_map/NavSatMap.qml
+++ b/src/plugins/navsat_map/NavSatMap.qml
@@ -18,8 +18,8 @@ import QtQuick 2.9
 import QtQuick.Controls 2.2
 import QtQuick.Layouts 1.3
 import QtQuick.Controls.Material 2.1
-import QtLocation 5.6
-import QtPositioning 5.6
+import QtLocation 5.15
+import QtPositioning 5.15
 
 Item {
   id: navSatMap
@@ -92,6 +92,16 @@ Item {
   Plugin {
     id: mapPlugin
     name: "osm"
+
+    PluginParameter {
+      name: "osm.mapping.providersrepository.disabled"
+      value: "true"
+    }
+
+    PluginParameter {
+      name: "osm.mapping.providersrepository.address"
+      value: "http://maps-redirect.qt.io/osm/5.6/"
+    }
   }
 
   Map {
@@ -159,7 +169,7 @@ Item {
 
   Connections {
     target: NavSatMap
-    onNewMessage: {
+    function onNewMessage(_latitudeDeg, _longitudeDeg) {
       latitude = _latitudeDeg
       longitude = _longitudeDeg
     }
```

iche033, 2023-03-08:
possibly related Qt issues:
* https://bugreports.qt.io/browse/QTBUG-103592
* https://bugreports.qt.io/browse/QTBUG-95638 - mentions bug is fixed in 5.15.11

JaouadROS, 2023-03-08:
@jennuine  I've tried your patch and I've got the same output, the GUI freezes.
@iche033 5.15.11 looks like a commercial release. If it does solve the issue, what do you suggest to try it out?

iche033, 2023-03-09:
You could try other plugins that are listed in this page:
https://doc.qt.io/qt-5/qtlocation-index.html#plugin-references-and-parameters

I don't know how well they work though.

iche033, 2023-03-09:
the `mapboxgl` plugin works for me. I created a PR: https://github.com/gazebosim/gz-gui/pull/520