gazebosim/gz-gui #484: Use Migration.md instructions for replacing GzScene3D in Garden lead to a crash

j-rivero, 2022-09-09:
## Environment
* OS Version: Focal
* Binaries in prerelease for Garden

## Description

While working on https://github.com/gazebosim/garden-tutorial-party/issues/2015 I was updating the GzScene3D plugin on https://github.com/gazebosim/docs/blob/master/garden/tutorials/sdf_worlds/world_demo.sdf#L23-L36. I checked the instructions in `Migration.md` and replaced by with https://github.com/gazebosim/docs/commit/d249a113ef0d2b238eb8ba609d0542044701409e, mostly TransportSceneManager and MinimalScene.

If I used that modified file, the result is:
[image]
(nothing displayed).

If I try to interact with the Scene, the simulator crashes:
```
terminate called after throwing an instance of 'Ogre::InvalidParametersException'
  what():  OGRE EXCEPTION(2:InvalidParametersException): Texture 'scene::RenderTexture(65526)0': Invalid settings! in TextureGpu::checkValidSettings at /home/jenkins/workspace/ogre-2.3-debbuilder/repo/OgreMain/src/OgreTextureGpu.cpp (line 451)
Stack trace (most recent call last) in thread 1108751:
#9    Object "[0xffffffffffffffff]", at 0xffffffffffffffff, in 
#8    Object "/lib/x86_64-linux-gnu/libc.so.6", at 0x7f6411e37132, in clone
#7    Object "/lib/x86_64-linux-gnu/libpthread.so.0", at 0x7f6411cfd608, in 
#6    Object "/usr/lib/x86_64-linux-gnu/libQt5Core.so.5", at 0x7f640b98296d, in 
#5    Object "/usr/lib/x86_64-linux-gnu/libQt5Core.so.5", at 0x7f640b980e8a, in qTerminate()
#4    Object "/usr/lib/x86_64-linux-gnu/libstdc++.so.6", at 0x7f640d8dd3f6, in std::terminate()
#3    Object "/usr/lib/x86_64-linux-gnu/libstdc++.so.6", at 0x7f640d8dd38b, in 
#2    Object "/usr/lib/x86_64-linux-gnu/libstdc++.so.6", at 0x7f640d8d1910, in 
#1    Object "/lib/x86_64-linux-gnu/libc.so.6", at 0x7f6411d3a858, in abort
#0    Object "/lib/x86_64-linux-gnu/libc.so.6", at 0x7f6411d5b00b, in gsignal
```

Note that the instructions in Migration.md are different from the behavior applied automatically if the GzScene3D is still in the sdf file:
```
[GUI] [Wrn] [Gui.cc:460] The [GzScene3D] GUI plugin has been removed since Garden. Loading the following plugins instead:
* GzSceneManager
* InteractiveViewControl
* CameraTracking
* MarkerManager
* SelectEntities
* EntityContextMenuPlugin
* Spawn
* VisualizationCapabilities
```

Not sure if I'm doing something wrong.

## Steps to reproduce

1. wget https://raw.githubusercontent.com/gazebosim/docs/master/garden/tutorials/sdf_worlds/world_demo.sdf
2. gz sim --verbose 4 world_demo.sdf 

## Output
```
gz sim --verbose 2 garden/tutorials/sdf_worlds/world_demo.sdf
[Wrn] [LocalCache.cc:105] Server directory does not exist [/home/jrivero/.gz/fuel/fuel.gazebosim.org]
[GUI] [Wrn] [Application.cc:845] [QT] file::/Gazebo/GazeboDrawer.qml:147:3: QML Dialog: Binding loop detected for property "implicitHeight"
[GUI] [Wrn] [Application.cc:845] [QT] file::/Gazebo/GazeboDrawer.qml:147:3: QML Dialog: Binding loop detected for property "implicitHeight"
[Wrn] [LocalCache.cc:105] Server directory does not exist [/home/jrivero/.gz/fuel/fuel.gazebosim.org]
[Wrn] [Component.hh:144] Trying to serialize component with data type [N3sdf3v135WorldE], which doesn't have `operator<<`. Component will not be serialized.
[GUI] [Wrn] [Application.cc:845] [QT] file::/EntityTree/EntityTree.qml:148:7: QML ToolButton: Detected anchors on an item that is managed by a layout. This is undefined behavior; use Layout.alignment instead.
[GUI] [Wrn] [Component.hh:189] Trying to deserialize component with data type [N3sdf3v135WorldE], which doesn't have `operator>>`. Component will not be deserialized.
[GUI] [Wrn] [Application.cc:845] [QT] file::/Gazebo/GazeboDrawer.qml:147:3: QML Dialog: Binding loop detected for property "implicitHeight"
libEGL warning: DRI2: failed to create dri screen
libEGL warning: DRI2: failed to create dri screen
[GUI] [Wrn] [Application.cc:845] [QT] file::/Gazebo/GazeboDrawer.qml:147:3: QML Dialog: Binding loop detected for property "implicitHeight"
```

jennuine, 2022-09-09:
Hmm, when using from source it doesn't crash for me but the tutorials ([sdf_worlds.md](https://github.com/gazebosim/docs/blob/master/garden/sdf_worlds.md#scene-3d-plugin) and [sdf](https://github.com/gazebosim/docs/blob/master/garden/tutorials/sdf_worlds/world_demo.sdf#L23-L36)) need to be updated to use the `MinimalScene` plugin since `GzScene3d` is deprecated.

@j-rivero Can you try replacing this section: https://github.com/gazebosim/docs/blob/master/garden/tutorials/sdf_worlds/world_demo.sdf#L23-L36

with this instead: 
https://github.com/gazebosim/gz-sim/blob/ff1c82b41e548dfdc8076374f9500db2df2c35a1/examples/worlds/minimal_scene.sdf#L29-L128

does the crash still occur?

j-rivero, 2022-09-12:
> Hmm, when using from source it doesn't crash for me

Sorry I did not give all the details on what I was doing. here is a quick gif:
[image]

> @j-rivero Can you try replacing this section: https://github.com/gazebosim/docs/blob/master/garden/tutorials/sdf_worlds/world_demo.sdf#L23-L36
>
> with this instead: [...]

Nop! It works as expected. We might need to update the Migration.md document to reflect this. I'll try to send a PR.

> the tutorials ([sdf_worlds.md](https://github.com/gazebosim/docs/blob/master/garden/sdf_worlds.md#scene-3d-plugin) and [...]

I'll take care of patching them.

closed 2023-08-21 by ahcorde