gazebosim/gz-gui #489: Program crashes when you load a client configuration

gaileyleseman, 2022-09-12:
## Bug Decription
* Expected behavior: You can load a client configuration through the GUI. It can handle overrides such that there are not multiple instances (for example of `GzSceneManager`) at the same time. 
* Actual behavior: When you load a client configuration the simulator crashes. 

## Steps to reproduce

1. Gazebo Garden 7.0.0~pre1 binary installation on Ubuntu 22.04
2. `gz sim shapes.sdf`
3.  `Save client configuration as` and save the file.
4.  `Load client configuration` and load the file from the previous step. 

## Output
```
gailey@zbook:~$ gz sim shapes.sdf
libEGL warning: DRI2: failed to create dri screen
libEGL warning: DRI2: failed to create dri screen
^[
(gz sim gui:199567): Gtk-CRITICAL **: 01:14:02.023: gtk_image_set_from_gicon: assertion 'GTK_IS_IMAGE (image)' failed
[GUI] [Err] [GzSceneManager.cc:104] Only one GzSceneManager is supported at a time.
[GUI] [Err] [SelectEntities.cc:485] Only one SelectEntities plugin is supported at a time.
[GUI] [Err] [Spawn.cc:175] Only one Spawn plugin is supported at a time.
[GUI] [Err] [VisualizationCapabilities.cc:2823] Only one Visualization capabilities plugin is supported at a time.
terminate called after throwing an instance of 'Ogre::ItemIdentityException'
  what():  OGRE EXCEPTION(4:ItemIdentityException): A texture with name 'scene::RenderTexture(65526)0' already exists. (Real tex name: 'scene::RenderTexture(65526)0') in TextureGpuManager::createTexture at ./OgreMain/src/OgreTextureGpuManager.cpp (line 385)
Stack trace (most recent call last):
#31   Object "/usr/lib/x86_64-linux-gnu/libgz-sim7-gz.so.7.0.0~pre1", at 0x7f470feedc4f, in runGui
#30   Object "/lib/x86_64-linux-gnu/libgz-sim7-gui.so.7", at 0x7f470fd42088, in gz::sim::v7::gui::runGui(int&, char**, char const*, char const*, int, char const*)
#29   Object "/lib/x86_64-linux-gnu/libQt5Core.so.5", at 0x7f470ee3bce3, in QCoreApplication::exec()
#28   Object "/lib/x86_64-linux-gnu/libQt5Core.so.5", at 0x7f470ee3374a, in QEventLoop::exec(QFlags<QEventLoop::ProcessEventsFlag>)
#27   Object "/lib/x86_64-linux-gnu/libQt5Core.so.5", at 0x7f470ee8e0a7, in QEventDispatcherGlib::processEvents(QFlags<QEventLoop::ProcessEventsFlag>)
#26   Object "/lib/x86_64-linux-gnu/libglib-2.0.so.0", at 0x7f470c9dd3c2, in g_main_context_iteration
#25   Object "/lib/x86_64-linux-gnu/libglib-2.0.so.0", at 0x7f470ca346f7, in 
#24   Object "/lib/x86_64-linux-gnu/libglib-2.0.so.0", at 0x7f470c9dfd1a, in g_main_context_dispatch
#23   Object "/lib/x86_64-linux-gnu/libQt5Core.so.5", at 0x7f470ee8ea56, in 
#22   Object "/lib/x86_64-linux-gnu/libQt5Core.so.5", at 0x7f470ee37f16, in QCoreApplicationPrivate::sendPostedEvents(QObject*, int, QThreadData*)
#21   Object "/lib/x86_64-linux-gnu/libQt5Core.so.5", at 0x7f470ee34e29, in QCoreApplication::notifyInternal2(QObject*, QEvent*)
#20   Object "/lib/x86_64-linux-gnu/libQt5Widgets.so.5", at 0x7f470e58e712, in QApplicationPrivate::notify_helper(QObject*, QEvent*)
#19   Object "/lib/x86_64-linux-gnu/libQt5Core.so.5", at 0x7f470ee6240d, in QObject::event(QEvent*)
#18   Object "/usr/lib/x86_64-linux-gnu/gz-gui-7/plugins/libMinimalScene.so", at 0x7f46dc5dbce3, in gz::gui::plugins::RenderWindowItem::Ready()
#17   Object "/usr/lib/x86_64-linux-gnu/gz-gui-7/plugins/libMinimalScene.so", at 0x7f46dc5db8f0, in gz::gui::plugins::RenderThread::Initialize[abi:cxx11]()
#16   Object "/usr/lib/x86_64-linux-gnu/gz-gui-7/plugins/libMinimalScene.so", at 0x7f46dc5e50ed, in gz::gui::plugins::RenderThreadRhiOpenGL::Initialize[abi:cxx11]()
#15   Object "/usr/lib/x86_64-linux-gnu/gz-gui-7/plugins/libMinimalScene.so", at 0x7f46dc5dda30, in gz::gui::plugins::GzRenderer::Initialize[abi:cxx11]()
#14   Object "/usr/lib/x86_64-linux-gnu/gz-rendering-7/engine-plugins/libgz-rendering-ogre2.so", at 0x7f465c88c52d, in 
#13   Object "/usr/lib/x86_64-linux-gnu/gz-rendering-7/engine-plugins/libgz-rendering-ogre2.so", at 0x7f465c91e256, in gz::rendering::v7::Ogre2RenderTarget::PreRender()
#12   Object "/usr/lib/x86_64-linux-gnu/gz-rendering-7/engine-plugins/libgz-rendering-ogre2.so", at 0x7f465c91aa5a, in 
#11   Object "/usr/lib/x86_64-linux-gnu/gz-rendering-7/engine-plugins/libgz-rendering-ogre2.so", at 0x7f465c91a950, in gz::rendering::v7::Ogre2RenderTarget::RebuildImpl()
#10   Object "/usr/lib/x86_64-linux-gnu/gz-rendering-7/engine-plugins/libgz-rendering-ogre2.so", at 0x7f465c91c1b9, in gz::rendering::v7::Ogre2RenderTarget::BuildTargetImpl()
#9    Object "/usr/lib/x86_64-linux-gnu/OGRE-2.3/libOgreNextMain.so.2.3.1", at 0x7f465c540a43, in Ogre::TextureGpuManager::createTexture(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, Ogre::GpuPageOutStrategy::GpuPageOutStrategy, unsigned int, Ogre::TextureTypes::TextureTypes, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, unsigned int, unsigned int)
#8    Object "/usr/lib/x86_64-linux-gnu/OGRE-2.3/libOgreNextMain.so.2.3.1", at 0x7f465c2d91ff, in 
#7    Object "/usr/lib/x86_64-linux-gnu/OGRE-2.3/libOgreNextMain.so.2.3.1", at 0x7f465c2b597b, in 
#6    Object "/lib/x86_64-linux-gnu/libstdc++.so.6", at 0x7f47107ed557, in __cxa_throw
#5    Object "/lib/x86_64-linux-gnu/libstdc++.so.6", at 0x7f47107ed2f6, in std::terminate()
#4    Object "/lib/x86_64-linux-gnu/libstdc++.so.6", at 0x7f47107ed28b, in 
#3    Object "/lib/x86_64-linux-gnu/libstdc++.so.6", at 0x7f47107e1bfd, in 
#2    Object "/lib/x86_64-linux-gnu/libc.so.6", at 0x7f4714f6d7f2, in abort
#1    Object "/lib/x86_64-linux-gnu/libc.so.6", at 0x7f4714f87475, in raise
#0    Object "/lib/x86_64-linux-gnu/libc.so.6", at 0x7f4714fdba7c, in pthread_kill
Aborted (Signal sent by tkill() 199567 1000)
```
* running on real hardware

## Environment
* OS Version: Ubuntu 22.04
* Build: Binary 7.0.0~pre1
* Rendering plugin: `gz-rendering-ogre2`
* GPU information 

    ```bash
    Tue Sep 13 00:34:32 2022       
    +-----------------------------------------------------------------------------+
[... 59 lines cut ...]

```
$ ps aux | grep Xorg

gailey      2056  0.5  0.8 27200740 144020 tty2  Rl+  sep10  20:48 /usr/lib/xorg/Xorg vt2 -displayfd 3 -auth /run/user/1000/gdm/Xauthority -nolisten tcp -background none -noreset -keeptty -novtswitch -verbose 3
gailey     48272  0.0  0.0  17868  2328 pts/0    S+   00:46   0:00 grep --color=auto Xorg
```

```
$ sudo env LANG=C X -version  

X.Org X Server 1.21.1.3
X Protocol Version 11, Revision 0
Current Operating System: Linux zbook 5.15.0-47-generic #51-Ubuntu SMP Thu Aug 11 07:51:15 UTC 2022 x86_64
Kernel command line: BOOT_IMAGE=/boot/vmlinuz-5.15.0-47-generic root=UUID=bb4da8d4-c255-46ee-bb34-4395db58cee0 ro quiet splash vt.handoff=7
xorg-server 2:21.1.3-2ubuntu2.1 (For technical support please see http://www.ubuntu.com/support) 
Current version of pixman: 0.40.0
	Before reporting problems, check http://wiki.x.org
	to make sure that you have the latest version.
```

### Ogre2 Log
[ogre2.log](https://github.com/gazebosim/gz-gui/files/9552243/ogre2.log)

</details>

tejalbarnwal, 2022-09-16:
This happens with Ubuntu 20.04, Binary installation.
## Environment specifications:
- OS Version: Ubuntu 20.04
- Build: Binary
- GPU Information: 
```
Fri Sep 16 11:27:32 2022       
+-----------------------------------------------------------------------------+
| NVIDIA-SMI 515.65.01    Driver Version: 515.65.01    CUDA Version: 11.7     |
|-------------------------------+----------------------+----------------------+
| GPU  Name        Persistence-M| Bus-Id        Disp.A | Volatile Uncorr. ECC |
| Fan  Temp  Perf  Pwr:Usage/Cap|         Memory-Usage | GPU-Util  Compute M. |
|                               |                      |               MIG M. |
|===============================+======================+======================|
|   0  NVIDIA GeForce ...  Off  | 00000000:01:00.0 Off |                  N/A |
| N/A   44C    P8    N/A /  N/A |      9MiB /  3072MiB |      0%      Default |
|                               |                      |                  N/A |
+-------------------------------+----------------------+----------------------+
                                                                               
+-----------------------------------------------------------------------------+
| Processes:                                                                  |
|  GPU   GI   CI        PID   Type   Process name                  GPU Memory |
|        ID   ID                                                   Usage      |
|=============================================================================|
|    0   N/A  N/A       936      G   /usr/lib/xorg/Xorg                  4MiB |
|    0   N/A  N/A      1462      G   /usr/lib/xorg/Xorg                  4MiB |
+-----------------------------------------------------------------------------+
```

azeey, 2022-09-26:
I just tested this on Fortress and got the same crash, so it's not just a Garden issue.

jennuine, 2022-09-26:
Same here, I tried in Citadel and although it doesn't crash the 3d scene becomes black 
[image]

azeey, 2022-09-26:
@gaileyleseman, I think you can still use saved configurations using `gz sim --gui-config <path_to_config_file>`. This issue seems to happen only if you've already been running a simulation and then try to load a config file.

calmelo, 2024-05-08:
> Same here, I tried in Citadel and although it doesn't crash the 3d scene becomes black [image]

Have you resolved the problem?

azeey, 2024-05-08:
No, this issue hasn't been resolved yet. We don't have anyone currently working on it, so any help would be appreciated.