gazebosim/gz-gui #503: UNIT_Teleop_TEST fails on macOS Big Sur

scpeters, 2022-11-15:
## Environment
* OS Version: macOS Big Sur (both Glados and Ultron CI machines). I can't reproduce the failure on my laptop running macOS Monterey
* Source or binary build? source build of ign-gui7 branch (most recently https://github.com/gazebosim/gz-gui/commit/17bd485d38888b97f8c54ef6fb17038618af009d)
* If this is a GUI or sensor rendering bug, describe your GPU and rendering system. Otherwise delete this section. 
    - Rendering plugin: [ogre | ogre2]. 
      - [ ] Sensor rendering error. 
      - [ ] GUI rendering error. 
    - EGL headless mode:
      - [ ] Running in EGL headless mode 
    - Generally, mention all circumstances that might affect rendering capabilities: 
      - [ ] running on a dual GPU machine (integrated GPU + discrete GPU)
      - [ ] running on a multi-GPU machine (it has multiple discrete GPUs)
      - [ ] running on real hardware
      - [ ] running in virtual machine
      - [ ] running in Docker/Singularity
      - [ ] running remotely (e.g. via SSH)
      - [ ] running in a cloud
      - [ ] using VirtualGL, XVFB, Xdummy, XVNC or other indirect rendering utilities
      - [ ] GPU is concurrently used for other tasks
        - [ ] desktop acceleration
        - [ ] video decoding (i.e. a playing Youtube video)
        - [ ] video encoding
        - [ ] CUDA/ROCm computations (Tensorflow, Torch, Caffe running)
        - [ ] multiple simulators running at the same time
      - [ ] other...
    - Rendering system info:
      - On Linux, provide the outputs of the following commands:
          ```bash
          LANG=C lspci -nn | grep VGA  # might require installing pciutils
          echo "$DISPLAY"
          LANG=C glxinfo -B | grep -i '\(direct rendering\|opengl\|profile\)'  # might require installing mesa-utils package
          ps aux | grep Xorg
          sudo env LANG=C X -version  # if you don't have root access, try to tell the version of Xorg e.g. via package manager
          ```
      - On Windows, run `dxdiag` and report the GPU-related information.
      - On Mac OS, open a terminal and type `system_profiler SPDisplaysDataType`. Copy the output here.
        
    - [ ] Please, attach the ogre.log or ogre2.log file from  `~/.gz/rendering` 

<details>

```
# paste log here
```

</details>

## Description
* Expected behavior: `UNIT_Teleop_TEST` should pass
* Actual behavior: it has been failing since 17bd485d38888b97f8c54ef6fb17038618af009d  was merged https://build.osrfoundation.org/view/ign-garden/job/ignition_gui-ci-gz-gui7-homebrew-amd64/18/testReport/(root)/UNIT_Teleop_TEST/test_ran/history/

## Steps to reproduce

1. Run the following CI build: https://build.osrfoundation.org/view/ign-garden/job/ignition_gui-ci-gz-gui7-homebrew-amd64
2.
3.

## Output

### console output

~~~
glados:build jenkins$ lldb bin/UNIT_Teleop_TEST 
(lldb) target create "bin/UNIT_Teleop_TEST"
Current executable set to '/Users/jenkins/workspace/ignition_gui-ci-gz-gui7-homebrew-amd64/build/bin/UNIT_Teleop_TEST' (x86_64).
(lldb) r
Process 63527 launched: '/Users/jenkins/workspace/ignition_gui-ci-gz-gui7-homebrew-amd64/build/bin/UNIT_Teleop_TEST' (x86_64)
Running main() from /Users/jenkins/workspace/ignition_gui-ci-gz-gui7-homebrew-amd64/ign-gui/test/gtest_vendor/src/gtest_main.cc
[==========] Running 2 tests from 1 test suite.
[----------] Global test environment set-up.
[... 355 lines cut ...]
    frame #0: 0x00007fff2066f94e libsystem_kernel.dylib`__workq_kernreturn + 10
    frame #1: 0x00007fff206a04c1 libsystem_pthread.dylib`_pthread_wqthread + 414
    frame #2: 0x00007fff2069f42f libsystem_pthread.dylib`start_wqthread + 15
* thread #24, name = 'QSGRenderThread', stop reason = EXC_BAD_ACCESS (code=1, address=0x3f800008)
  * frame #0: 0x00000001009ca525 QtQuick`QSGBatchRenderer::Renderer::updateMaterialDynamicData(QSGBatchRenderer::ShaderManagerShader*, QSGMaterialRhiShader::RenderState&, QSGMaterial*, QVarLengthArray<QRhiShaderResourceBinding, 8>*, QSGBatchRenderer::Batch const*, int, int) + 565
    frame #1: 0x00000001009cafb4 QtQuick`QSGBatchRenderer::Renderer::prepareRenderMergedBatch(QSGBatchRenderer::Batch*, QSGBatchRenderer::Renderer::PreparedRenderBatch*) + 1076
    frame #2: 0x00000001009cd7f5 QtQuick`QSGBatchRenderer::Renderer::renderBatches() + 1157
    frame #3: 0x00000001009cfb46 QtQuick`QSGBatchRenderer::Renderer::render() + 1958
    frame #4: 0x00000001009b6b4b QtQuick`QSGRenderer::renderScene(QSGBindable const&) + 539
    frame #5: 0x00000001009b691a QtQuick`QSGRenderer::renderScene(unsigned int) + 58
    frame #6: 0x0000000100a7e77a QtQuick`QQuickWindowPrivate::renderSceneGraph(QSize const&, QSize const&) + 954
    frame #7: 0x0000000100a25a6c QtQuick`___lldb_unnamed_symbol2564$$QtQuick + 2076
    frame #8: 0x0000000100a26caf QtQuick`___lldb_unnamed_symbol2573$$QtQuick + 287
    frame #9: 0x0000000101f14440 QtCore`___lldb_unnamed_symbol241$$QtCore + 320
    frame #10: 0x00007fff206a38fc libsystem_pthread.dylib`_pthread_start + 224
    frame #11: 0x00007fff2069f443 libsystem_pthread.dylib`thread_start + 15
  thread #25, name = 'CVDisplayLink'
    frame #0: 0x00007fff20670cce libsystem_kernel.dylib`__psynch_cvwait + 10
    frame #1: 0x00007fff206a3e7f libsystem_pthread.dylib`_pthread_cond_wait + 1352
    frame #2: 0x00007fff275f6f97 CoreVideo`CVDisplayLink::waitUntil(unsigned long long) + 229
    frame #3: 0x00007fff275f64a0 CoreVideo`CVDisplayLink::runIOThread() + 482
    frame #4: 0x00007fff206a38fc libsystem_pthread.dylib`_pthread_start + 224
    frame #5: 0x00007fff2069f443 libsystem_pthread.dylib`thread_start + 15
(lldb)
~~~

iche033, 2022-11-15:
I can't reproduce the crash on my mac too but saw the same Qt output about shader preparation failure. The test passed on my machine.

scpeters, 2022-11-15:
> I can't reproduce the crash on my mac too but saw the same Qt output about shader preparation failure. The test passed on my machine.

what version of macOS did you use?

iche033, 2022-11-16:
I'm on macOS Monterey as well.

azeey, 2023-04-07:
I was able to reproduce this on my Bigsur machine and determined that the crashes started after qt5 was updated to 5.15.7, so I've created a PR to disable the test.
Looking at the qt5 [release note](https://code.qt.io/cgit/qt/qtreleasenotes.git/about/qt/5.15.7/release-note.md), [QTBUG-55204](https://bugreports.qt.io/browse/QTBUG-55204) seems to be since it mentions shader effects, but I could be totally off the mark.