gazebosim/gz-gui #542: GzSpinBox causes the Qt Scene Graph to crash on macOS

srmainwaring, 2023-05-31:
## Environment
* OS Version: macOS, `qt@5 5.15.8_3`
* Source or binary build? either

## Description
* Expected behavior: GUI plugins that use GzSpinBox should work correctly.
* Actual behavior: The GUI application crashes. The error occurs on the Qt scene graph render thread.

## Steps to reproduce

1.  Update and upgrade `brew`. This will install qt@5 5.15.8_3. 
2. Run `gz sim -g shapes.sdf`
3. Open the Component Inspector (if not already opened)

## Output

See https://github.com/gazebosim/gz-sim/issues/1987.

The problem appears to be a conflict between QtQuickControls 1.4 and QtQuickControls 2.2. Both versions have a SpinBox but the behaviour is very different. The version in QtQuickControls 2.2 is for integer values only.

The QML documentation suggests an approach to use the controls for floats, but it is badly flawed and will not work for the use cases needed in `gz-sim`. A reimplementation of a spin box for doubles is needed. Docs and discussion:

-  https://doc.qt.io/qt-5/qml-qtquick-controls2-spinbox.html#validator-prop
- https://stackoverflow.com/questions/43406830/how-to-use-float-in-a-qml-spinbox

This is a serious bug on macOS as it causes an application crash on start up for any of the default examples (because the default `gui.config` includes the component inspector which indirectly loads the spin box).

azeey, 2023-09-22:
@srmainwaring I just tested this on an Intel macOS running Monterey, but I'm not getting a segfault at all. The spinbox doesn't seem to work properly (it's blank), but it doesn't crash. qt@5 has been upgraded to 5.15.10. Can you give it another try?

srmainwaring, 2024-02-08:
> Can you give it another try?

@azeey - doing some book-keeping! Not seeing this at all in current Garden and Harmonic running Sonoma on M1 with latest brew - so suspect it was transient.

closed 2024-02-08 by srmainwaring