gazebosim/gz-gui #586: Migrate Qt5 to Qt6

mjcarroll, 2023-10-10:
Tracking issue for tasks required to migrate from Qt5 to Qt6.

- [ ] Update MinimalScene for new APIs getting native access to VkImage and Metal equivalent.
- [ ] QtGraphicalEffects is deprecated and has some incompatibilities: https://doc.qt.io/qt-6/qtgraphicaleffects5-index.html

mjcarroll, 2024-01-29:
Clazy checks: https://github.com/gazebosim/gz-gui/pull/584

mjcarroll, 2024-01-29:
Bulk of the migrations for gui: https://github.com/gazebosim/gz-gui/pull/596

azeey, 2024-07-12:
Here's the list of qt6 packages I needed to install:

- qml6-module-qt-labs-folderlistmodel
- qml6-module-qt-labs-platform
- qml6-module-qt-labs-settings
- qml6-module-qtcharts
- qml6-module-qtpositioning
- qml6-module-qtqml-models
- qml6-module-qtquick-controls
- qml6-module-qtquick-dialogs
- qml6-module-qtquick-layouts
- qml6-module-qtquick-templates
- qml6-module-qtquick-window
- qml6-module-qtquick
- qt6-base-dev
- qt6-declarative-dev
- libqt6quickcontrols2-6
- qml6-module-qtcore
- qml6-module-qtqml-workerscript
- qml6-module-qt5compat-graphicaleffects
- libqt6svg6

azeey, 2024-07-16:
I've added a fix in the gz-sim branch https://github.com/gazebosim/gz-sim/pull/2305/commits/a2652ac19b906587f441c40f29c886ceda7dd0d3 which should help with the communication between the server and GUI ECMs. With that, I see glimmers of hope :)
[Screen recording 2024-07-16 14.18.43.webm](https://github.com/user-attachments/assets/8281d76d-4bce-46b8-8ffd-445ec0aebaaa)

caguero, 2024-07-17:
I was able to visualize the 3D screen even without your changes. It was hidden at the very left covered by other collapsed window plugins.

[image]

azeey, 2025-08-25:
Implemented by https://github.com/gazebosim/gz-sim/pull/2832, https://github.com/gazebosim/gz-gui/pull/666 and a few other auxiliary PRs.