gazebosim/gz-gui #618: VMWare Ubuntu: Warning: Ignoring XDG_SESSION_TYPE=wayland on Gnome. Use QT_QPA_PLATFORM=wayland to run on Wayland anyway.

Ryanf55, 2024-05-02:
## Environment
* OS Version: Ubuntu 22.04 running in VMWare on Windows 10
* Source or binary build?
 Binary - Harmonic
    - Rendering plugin: [ogre | ogre2]. 
      - [ ] Sensor rendering error. 
      - [ ] GUI rendering error. 
    - EGL headless mode:
      - [ ] Running in EGL headless mode 
    - Generally, mention all circumstances that might affect rendering capabilities: 
      - [ ] running on a dual GPU machine (integrated GPU + discrete GPU)
      - [ ] running on a multi-GPU machine (it has multiple discrete GPUs)
      - [ ] running on real hardware
      - [x] running in virtual machine
      - [ ] running in Docker/Singularity
      - [ ] running remotely (e.g. via SSH)
      - [ ] running in a cloud
      - [ ] using VirtualGL, XVFB, Xdummy, XVNC or other indirect rendering utilities
      - [ ] GPU is concurrently used for other tasks
        - [ ] desktop acceleration
        - [ ] video decoding (i.e. a playing Youtube video)
        - [ ] video encoding
        - [ ] CUDA/ROCm computations (Tensorflow, Torch, Caffe running)
        - [ ] multiple simulators running at the same time
      - [ ] other...
    - Rendering system info:
      - On Linux, provide the outputs of the following commands:
          ```bash
          LANG=C lspci -nn | grep VGA  # might require installing pciutils
         00:0f.0 VGA compatible controller [0300]: VMware SVGA II Adapter [15ad:0405]
          echo "$DISPLAY"
          :0
          LANG=C glxinfo -B | grep -i '\(direct rendering\|opengl\|profile\)'  # might require installing mesa-utils package
            direct rendering: Yes
                Preferred profile: core (0x1)
                Max core profile version: 4.3
                Max compat profile version: 4.3
                Max GLES1 profile version: 1.1
                Max GLES[23] profile version: 3.1
                OpenGL vendor string: VMware, Inc.
                OpenGL renderer string: SVGA3D; build: RELEASE;  LLVM;
                OpenGL core profile version string: 4.3 (Core Profile) Mesa 23.2.1-1ubuntu3.1~22.04.2
                OpenGL core profile shading language version string: 4.30
                OpenGL core profile context flags: (none)
                OpenGL core profile profile mask: core profile
                OpenGL version string: 4.3 (Compatibility Profile) Mesa 23.2.1-1ubuntu3.1~22.04.2
                OpenGL shading language version string: 4.30
                OpenGL context flags: (none)
                OpenGL profile mask: compatibility profile
                OpenGL ES profile version string: OpenGL ES 3.1 Mesa 23.2.1-1ubuntu3.1~22.04.2
                OpenGL ES profile shading language version string: OpenGL ES GLSL ES 3.10

          ps aux | grep Xorg
              ryan        3996  0.0  0.0   9080  2432 pts/0    S+   00:04   0:00 grep --color=auto Xorg

          sudo env LANG=C X -version  # if you don't have root access, try to tell the version of Xorg e.g. via package manager
            X.Org X Server 1.21.1.4
            X Protocol Version 11, Revision 0
            Current Operating System: Linux ub22d17 6.5.0-28-generic #29~22.04.1-Ubuntu SMP PREEMPT_DYNAMIC Thu Apr  4 14:39:20 UTC 2 x86_64
            Kernel command line: BOOT_IMAGE=/boot/vmlinuz-6.5.0-28-generic root=UUID=4e184a09-a38b-4a10-a09f-cf3f949d586c ro find_preseed=/preseed.cfg auto noprompt priority=critical locale=en_US quiet splash
            xorg-server 2:21.1.4-2ubuntu1.7~22.04.10 (For technical support please see http://www.ubuntu.com/support) 
            Current version of pixman: 0.40.0
          ```
      - On Windows, run `dxdiag` and report the GPU-related information.
      - On Mac OS, open a terminal and type `system_profiler SPDisplaysDataType`. Copy the output here.
        
    - [ ] Please, attach the ogre.log or ogre2.log file from  `~/.gz/rendering` 

</details>

[... 177 lines cut ...]
[Wrn] [Model.hh:69] Skipping serialization / deserialization for models with //pose/@relative_to attribute.
[Wrn] [Component.hh:144] Trying to serialize component with data type [N3sdf3v145WorldE], which doesn't have `operator<<`. Component will not be serialized.
[GUI] [Msg] Added plugin [Component inspector] to main window
[GUI] [Msg] Loaded plugin [ComponentInspector] from path [/usr/lib/x86_64-linux-gnu/gz-sim-8/plugins/gui/libComponentInspector.so]
[GUI] [Wrn] [Application.cc:908] [QT] file::/EntityTree/EntityTree.qml:148:7: QML ToolButton: Detected anchors on an item that is managed by a layout. This is undefined behavior; use Layout.alignment instead.
[GUI] [Msg] Added plugin [Entity tree] to main window
[GUI] [Msg] Loaded plugin [EntityTree] from path [/usr/lib/x86_64-linux-gnu/gz-sim-8/plugins/gui/libEntityTree.so]
[GUI] [Msg] Using server control service [/server_control]
[GUI] [Wrn] [Component.hh:189] Trying to deserialize component with data type [N3sdf3v145WorldE], which doesn't have `operator>>`. Component will not be deserialized.
[GUI] [Wrn] [Model.hh:98] Unable to deserialize sdf::Model
[GUI] [Wrn] [Model.hh:98] Unable to deserialize sdf::Model
[GUI] [Wrn] [Model.hh:98] Unable to deserialize sdf::Model
[GUI] [Wrn] [Model.hh:98] Unable to deserialize sdf::Model
[GUI] [Wrn] [Model.hh:98] Unable to deserialize sdf::Model
[GUI] [Wrn] [Application.cc:908] [QT] file::/EntityTree/EntityTree.qml:148:7: QML ToolButton: Detected anchors on an item that is managed by a layout. This is undefined behavior; use Layout.alignment instead.
[GUI] [Wrn] [Application.cc:908] [QT] file::/WorldStats/WorldStats.qml:53:3: QML RowLayout: Binding loop detected for property "x"
[GUI] [Wrn] [Application.cc:908] [QT] file::/EntityContextMenuPlugin/EntityContextMenuPlugin.qml:52:3: QML RenderWindowOverlay: Detected anchors on an item that is managed by a layout. This is undefined behavior; use Layout.alignment instead.
[GUI] [Wrn] [Application.cc:908] [QT] file::/EntityContextMenuPlugin/EntityContextMenuPlugin.qml:67:3: QML EntityContextMenu: Detected anchors on an item that is managed by a layout. This is undefined behavior; use Layout.alignment instead.
[GUI] [Msg] Loading plugin [gz-rendering-ogre2]
[GUI] [Msg] Move to service on [/gui/move_to]
[GUI] [Msg] Follow service on [/gui/follow]
[GUI] [Msg] Move to pose service on [/gui/move_to/pose]
[GUI] [Msg] Camera pose topic advertised on [/gui/camera/pose]
[GUI] [Msg] Follow offset service on [/gui/follow/offset]
```

Ryanf55, 2024-05-02:
Ironically, running inside a VSode docker devcontainer solves it:
[image]

Here's the stripped version with all the random platform-specific stuff removed. To me, `"LIBGL_ALWAYS_SOFTWARE": "1",` is very suspicious that our devcontainer doesn't use the GPU and that's why it works.
```json
// See https://aka.ms/vscode-remote/devcontainer.json for format details.
{
   "name": "Development Container",
   "build": {
      "context": "..",
      "dockerfile": "../.docker/Dockerfile",
      "target": "dev",
      "args": {
         "DOCKER_GID": "${localEnv:DOCKER_GID}",
      }
   },
   "mounts": [
      "source=/var/run/docker.sock,target=/var/run/docker.sock,type=bind",
      "source=${localEnv:HOME}/.ssh/config,target=/home/avas/.ssh/config,type=bind",
      "source=${localEnv:HOME}/.ssh/known_hosts,target=/home/avas/.ssh/known_hosts,type=bind"
   ],
   "workspaceMount": "source=${localWorkspaceFolder},target=/ws,type=bind",
   "workspaceFolder": "/ws",
   "runArgs": [
      "--network=host",
      "--cap-add=SYS_PTRACE",
      "--security-opt=seccomp:unconfined",
      "--security-opt=apparmor:unconfined",
      "--volume=/tmp/.X11-unix:/tmp/.X11-unix",
      "--ipc=host",
   ],
   "containerEnv": {
      "DISPLAY": "${localEnv:DISPLAY}", // Needed for GUI try ":0" for windows
      "WAYLAND_DISPLAY": "${localEnv:WAYLAND_DISPLAY}",
      "XDG_RUNTIME_DIR": "${localEnv:XDG_RUNTIME_DIR}",
      "PULSE_SERVER": "${localEnv:PULSE_SERVER}",
      "LIBGL_ALWAYS_SOFTWARE": "1", // Needed for software rendering of opengl
      "DOCKER_GID": "${localEnv:DOCKER_GID}",
      "ARCH": "${localEnv:ARCH}"
   },
   // Set *default* container specific settings.json values on container create.
   "customizations": {
      "vscode": {
         "settings": {
            "terminal.integrated.defaultProfile.linux": "bash",
            "terminal.integrated.profiles.linux": {
               "bash": { "path": "/bin/bash" }
            }
         }
      }
   }
}

```

I'm not sure what to

Ryanf55, 2024-05-02:
Relates to https://github.com/gazebosim/gz-sim/issues/1841

traversaro, 2024-05-02:
I guess this is also related to https://github.com/gazebosim/gz-sim/issues/1492, in particular https://github.com/gazebosim/gz-sim/issues/1492#issuecomment-1146663381#issuecomment-1146663381 . I think it would be useful if you could check the test mentioned in that comment to understand which OpenGL extension are available in VmWare GPU's driver. If the problem is indeed the missing extensions, I guess you can try to use `ogre` instead of `ogre2` rendering engine. For debugging, it would also be useful to report the `ogre2.log` file from `~/.gz/rendering`.

azeey, 2024-05-06:
Can you try setting `QT_QPA_PLATFORM=xcb`? (from https://github.com/ros2/rviz/issues/847#issuecomment-1506502560)

Ryanf55, 2024-05-06:
> Can you try setting `QT_QPA_PLATFORM=xcb`? (from [ros2/rviz#847 (comment)](https://github.com/ros2/rviz/issues/847#issuecomment-1506502560))

No change in behavior

Ryanf55, 2024-05-06:
> I guess this is also related to [gazebosim/gz-sim#1492](https://github.com/gazebosim/gz-sim/issues/1492), in particular [gazebosim/gz-sim#1492 (comment)](https://github.com/gazebosim/gz-sim/issues/1492#issuecomment-1146663381) . I think it would be useful if you could check the test mentioned in that comment to understand which OpenGL extension are available in VmWare GPU's driver. If the problem is indeed the missing extensions, I guess you can try to use `ogre` instead of `ogre2` rendering engine. For debugging, it would also be useful to report the `ogre2.log` file from `~/.gz/rendering`.

```
$ glxinfo | grep GL_ARB_shading_language_420pack
glxinfo | grep GL_ARB_texture_storage

# Optional but really good to have
glxinfo | grep GL_ARB_buffer_storage
glxinfo | grep GL_ARB_texture_buffer_range
glxinfo | grep GL_ARB_base_instance

# misc, just in case. In fact it may be better if this one returns nothing
glxinfo | grep GL_ARB_multi_draw_indirect
    GL_ARB_shader_texture_lod, GL_ARB_shading_language_420pack, 
    GL_ARB_shading_language_420pack, GL_ARB_shading_language_include, 
    GL_ARB_texture_storage, GL_ARB_texture_storage_multisample, 
    GL_ARB_texture_stencil8, GL_ARB_texture_storage, 
    GL_ARB_texture_storage_multisample, GL_ARB_texture_swizzle, 
    GL_ARB_blend_func_extended, GL_ARB_buffer_storage, 
    GL_ARB_buffer_storage, GL_ARB_clear_buffer_object, GL_ARB_clear_texture, 
    GL_ARB_texture_buffer_range, GL_ARB_texture_compression_bptc, 
    GL_ARB_texture_buffer_range, GL_ARB_texture_compression, 
    GL_ARB_ES3_compatibility, GL_ARB_arrays_of_arrays, GL_ARB_base_instance, 
    GL_ARB_arrays_of_arrays, GL_ARB_base_instance, GL_ARB_blend_func_extended, 
    GL_ARB_map_buffer_range, GL_ARB_multi_bind, GL_ARB_multi_draw_indirect, 
    GL_ARB_map_buffer_range, GL_ARB_multi_bind, GL_ARB_multi_draw_indirect,
```

traversaro, 2024-05-06:
A lot of extensions indeed are missing. Can you report the full output of `glxinfo`?

traversaro, 2024-05-06:
Actually the extensions are there, I misunderstood your output. Also the `GL_ARB_copy_image` extension that was missing for WSLg (https://github.com/gazebosim/gz-sim/issues/920#issuecomment-1470925812) is present. At this point, it could be really useful to also get the `ogre2.log` file that could contain information on the missing extension, if a missing extension is indeed the problem.

Ryanf55, 2024-05-06:
> Actually the extensions are there, I misunderstood your output. Also the `GL_ARB_copy_image` extension that was missing for WSLg ([gazebosim/gz-sim#920 (comment)](https://github.com/gazebosim/gz-sim/issues/920#issuecomment-1470925812)) is present. At this point, it could be really useful to also get the `ogre2.log` file that could contain information on the missing extension, if a missing extension is indeed the problem.

[ogre2_log.txt](https://github.com/gazebosim/gz-gui/files/15226679/ogre2_log.txt)

azeey, 2024-05-07:
I see in your logs
```
Warning: Ignoring XDG_SESSION_TYPE=wayland on Gnome. Use QT_QPA_PLATFORM=wayland to run on Wayland anyway.
qt.qpa.plugin: Could not find the Qt platform plugin "wayland" in ""
This application failed to start because no Qt platform plugin could be initialized. Reinstalling the application may fix this problem.

Available platform plugins are: eglfs, linuxfb, minimal, minimalegl, offscreen, vnc, xcb.
```

What was the error log after doing `export QT_QPA_PLATFORM=xcb` and running `gz sim`?

traversaro, 2024-05-08:
Just to tackle another lead, the `ogre2.log` contains some interesting errors:
~~~
15:17:16: OGRE EXCEPTION(3:RenderingAPIException): Fixed Function pipeline is no longer allowed nor supported. The material scene::Material(65518) must use shaders in HlmsLowLevel::calculateHashFor at ./OgreMain/src/OgreHlmsLowLevel.cpp (line 160)
15:17:16: OGRE EXCEPTION(3:RenderingAPIException): Fixed Function pipeline is no longer allowed nor supported. The material scene::Material(65518) must use shaders in HlmsLowLevel::calculateHashFor at ./OgreMain/src/OgreHlmsLowLevel.cpp (line 160)
15:17:16: Couldn't apply datablock '[Hash 0x1dc5a795]' to this renderable. Using default one. Check previous log messages to see if there's more information.
15:17:16: OGRE EXCEPTION(3:RenderingAPIException): Fixed Function pipeline is no longer allowed nor supported. The material scene::Material(65518) must use shaders in HlmsLowLevel::calculateHashFor at ./OgreMain/src/OgreHlmsLowLevel.cpp (line 160)
15:17:16: OGRE EXCEPTION(3:RenderingAPIException): Fixed Function pipeline is no longer allowed nor supported. The material scene::Material(65518) must use shaders in HlmsLowLevel::calculateHashFor at ./OgreMain/src/OgreHlmsLowLevel.cpp (line 160)
15:17:16: Couldn't apply datablock '[Hash 0x1dc5a795]' to this renderable. Using default one. Check previous log messages to see if there's more information.
15:17:16: Can't assign material scene::Material(65466) because this Material does not exist. Have you forgotten to define it in a .material script?
15:17:16: OGRE EXCEPTION(3:RenderingAPIException): Fixed Function pipeline is no longer allowed nor supported. The material Default/TransGreen must use shaders in HlmsLowLevel::calculateHashFor at ./OgreMain/src/OgreHlmsLowLevel.cpp (line 160)
15:17:16: OGRE EXCEPTION(3:RenderingAPIException): Fixed Function pipeline is no longer allowed nor supported. The material Default/TransGreen must use shaders in HlmsLowLevel::calculateHashFor at ./OgreMain/src/OgreHlmsLowLevel.cpp (line 160)
15:17:16: Couldn't apply datablock '[Hash 0x1a9fb58f]' to this renderable. Using default one. Check previous log messages to see if there's more information.
15:17:16: Vertex Shader: 100000001VertexShader_vs
Fragment Shader: 100000001PixelShader_ps
 GLSL validation result : 
active samplers with a different type refer to the same texture image unit
15:17:16: WARNING: Deleting mapped buffer without having it unmapped. This is often sign of a resource leak or a bad pattern. Umapping the buffer for you...
15:17:16: WARNING: Deleting mapped buffer without having it unmapped. This is often sign of a resource leak or a bad pattern. Umapping the buffer for you...
15:17:16: WARNING: Deleting mapped buffer without having it unmapped. This is often sign of a resource leak or a bad pattern. Umapping the buffer for you...
15:17:16: WARNING: Deleting mapped buffer without having it unmapped. This is often sign of a resource leak or a bad pattern. Umapping the buffer for you...
~~~

Just to double check, can you report also the `ogre2.log` when running with `LIBGL_ALWAYS_SOFTWARE=1`, so we can diff them?

Ryanf55, 2024-05-09:
`export QT_QPA_PLATFORM=xcb`
[ogre2_QT_QPA_PLATFORM_xcb .txt](https://github.com/gazebosim/gz-gui/files/15267967/ogre2_QT_QPA_PLATFORM_xcb.txt)

`export LIBGL_ALWAYS_SOFTWARE=1`
[ogre2_LIBGL_ALWAYS_SOFTWARE_1.txt](https://github.com/gazebosim/gz-gui/files/15267999/ogre2_LIBGL_ALWAYS_SOFTWARE_1.txt)

azeey, 2024-05-09:
I was hoping to get the logs of `gz sim -v4 shapes.sdf` since the issue might be with Qt.

Ryanf55, 2024-05-10:
Here's the console log for each scenario.
[console_QT_QPA_PLATFORM_xcb .txt](https://github.com/gazebosim/gz-gui/files/15276594/console_QT_QPA_PLATFORM_xcb.txt)
[console_LIBGL_ALWAYS_SOFTWARE_1.txt](https://github.com/gazebosim/gz-gui/files/15276595/console_LIBGL_ALWAYS_SOFTWARE_1.txt)

jmssa, 2025-02-25:
Hi, I am currently facing the same issue. Do you happen to have had any further developments? Thanks in advance.

Ryanf55, 2025-03-01:
> Hi, I am currently facing the same issue. Do you happen to have had any further developments? Thanks in advance.

If you are facing the same issue, please put a thumbs up on the issue so the Gazebo team is aware of how many people run into this. It is unresolved.