gazebosim/gz-gui #627: libGrid3D.so is installed with RUNPATH set to the build location

wentasah, 2024-07-15:
## Environment
* OS Version: NixOS (not relevant)
* Source or binary build? Both (branch `gz=gui8`)
  
  

## Description
* Expected behavior: `libGrid3D.so` is not installed with `RUNPATH` set.
* Actual behavior: `libGrid3D.so` is has `RUNPATH` set.

## Steps to reproduce

Run the following and observe the output:

```
$ objdump -x /usr/lib/x86_64-linux-gnu/gz-gui-8/plugins/libGrid3D.so|grep PATH
RUNPATH              /home/jenkins/workspace/gz-gui8-debbuilder/build/gz-gui-8.3.0/obj-x86_64-linux-gnu/lib:
$ objdump -x /usr/lib/x86_64-linux-gnu/gz-gui-8/plugins/libGridConfig.so|grep PATH
# no output
```
`libGrid3D.so` has RUNPATH set, while other plugins don't.

## Discussion

I came across this problem when building Gazebo with the Nix package manager in [this project](https://github.com/lopsided98/nix-ros-overlay). Nix builds are automatically checked for inappropriate RUNPATH and fail if one is detected.

The reason why `RUNPATH` is set for `libGrid3D.so` is that it just a copy of `libGridConfig.so`, but installed differently ([see here](https://github.com/gazebosim/gz-gui/blob/036df090d5fb9323617ad186156e295a85e38421/src/plugins/grid_config/CMakeLists.txt#L11)). Due to the fact that it is installed as a file rather than a library, CMake doesn't remove `RUNPATH` setting during installation.

I tried to figure out, how to convince CMake to install a library under a different name and remove `RUNPATH` settings, however I didn't find anything straightforward (except building the library twice). `RENAME` keyword is ignored when installing library targets and I don't see a way to do post-installation changes.

We can work around this in the downstream project, but fixing it here might benefit other users too.

azeey, 2024-07-22:
Thanks for flagging this. We likely won't be getting around to this anytime soon, but I've added the `help wanted` tag

shashank1300, 2025-04-16:
Hey, I'd like to work on this.

wentasah, 2025-04-16:
Great! I'll be happy to test/review the result. Do you know where to start?

shashank1300, 2025-04-16:
I'm looking into it and will get back to you soon, when I find something. For now, I can see that the install() command copies the raw library file which still contains the build directory in its RUNPATH that usually is not cleaned by CMake for manually copied files like this. I'll try modifying the install command to solve this.

shashank1300, 2025-04-18:
Hey @wentasah , sorry it took me a bit long, I was busy with other stuff. I have created a PR with the fix, feel free to test/review it.

PR #676 (2025-08-07): Remove build RUNPATH from installed libGrid3D.so
# 🦟 Bug fix

Fixes #627

## Summary
Installs libGrid3D.so (a legacy alias for libGridConfig.so) by copying the already installed libGridConfig.so using install(CODE ...) instead of install(FILES ...). This ensures the build directory RUNPATH is correctly stripped, matching other installed plugins.

## Checklist
- [x] Signed all commits for DCO
- [ ] Added tests
- [ ] Updated documentation (as needed)
- [ ] Updated migration guide (as needed)
- [ ] Consider updating Python bindings (if the library has them)
- [ ] `codecheck` passed (See [contribu [... cut]
commit 1980224f (2025-08-12): Remove build RUNPATH from installed libGrid3D.so (#676)

Installs libGrid3D.so (a legacy alias for libGridConfig.so)
by copying the already installed libGridConfig.so using install(CODE ...)
instead of install(FILES ...). This ensures the build directory
RUNPATH is correctly stripped, matching other installed plugins.

Fixes #627