gazebosim/gz-gui #628: GUI camera doesn't keep manual orientation changes when in Follow entity mode

NDevDrone, 2024-07-18:
## Environment
* OS Version: Ubuntu 22.04
* Source or binary build?
Garden, both source and binary. 

## Description
* Expected behavior: Rotating camera in follow mode should change orientation of the tracking camera.
* Actual behavior: The rotation changes but then returns to the original follow orientation. 
## Steps to reproduce

1. Follow entity.
2. Rotate or pan GUI camera view.
3. Sit in despair as you watch the camera rotate back to its default, as you can't find a means to lock new camera orientation while following anywhere within the GUI. 

## Output
N/A

azeey, 2024-07-18:
I've transferred this issue to gz-gui since it's not a documentation issue. 

Also, there have been some changes to the camera tracking plugin. Is this still an issue?

NDevDrone, 2024-07-18:
Hi @azeey , sorry I haven't been up to speed on developments of the camera tracking plugin. Would you mind sharing where they are reported ? Great news if so!

azeey, 2024-07-22:
Sure, you can find the it in the [changelog](https://github.com/gazebosim/gz-gui/blob/gz-gui8/Changelog.md).

NDevDrone, 2024-07-22:
Thanks azeey, that's pretty great to see. Looking forward to trying it out.