gazebosim/gz-gui #641: `gz gui` does not work on Windows

azeey, 2024-09-13:
## Environment
* OS Version: Windows
* Source or binary build? source, gz-gui9

## Description
* Expected behavior: `gz gui` to work on windows
* Actual behavior: Fails with 
```
<internal:C:/Users/.../miniconda3/envs/gz-ws/lib/ruby/3.3.0/rubygems/core_ext/kernel_require.rb>:136:in `require': cannot load such file -- C:/Users/kimbe/Development/simulation/gazebo/gz-ws/install/lib/ruby/gz/cmdgui9 (LoadError)
        from <internal:C:/Users/.../miniconda3/envs/gz-ws/lib/ruby/3.3.0/rubygems/core_ext/kernel_require.rb>:136:in `require'
        from C:/Users/kimbe/Development/simulation/gazebo/gz-ws/install/bin//gz:310:in `<main>'
```
See https://github.com/gazebosim/gazebo_test_cases/issues/1321

TheCrazyT, 2024-12-22:
well my manual workaround was downloading:
https://raw.githubusercontent.com/gazebosim/gz-gui/d110e5b288e8b580b28c0a7256d0e1f0e0c402e2/src/cmd/cmdgui.rb.in
renamed it as "cmdgui9.rb" (placed in "c:\Users\[YOUR USERNAME]\anaconda3\envs\gazebo\Library\lib\ruby\gz\" folder) and modified the lines:

```
LIBRARY_NAME = '../../../bin/gz-gui9.dll'
LIBRARY_VERSION = '9.0.0'
```

not so shure what package should normaly do that. (maybe "libgz-gui9" ?)

traversaro, 2024-12-22:
> well my manual workaround was downloading:
> https://raw.githubusercontent.com/gazebosim/gz-gui/d110e5b288e8b580b28c0a7256d0e1f0e0c402e2/src/cmd/cmdgui.rb.in
> renamed it as "cmdgui9.rb" (placed in "c:\Users\[YOUR USERNAME]\anaconda3\envs\gazebo\Library\lib\ruby\gz\" folder) and modified the lines:
> 
> ```
> LIBRARY_NAME = '../../../bin/gz-gui9.dll'
> LIBRARY_VERSION = '9.0.0'
> ```
> 
> not so shure what package should normaly do that. (maybe "libgz-gui9" ?)
> 
> **Update:**
> Oh well, but I guess it won't be useful, since https://github.com/gazebosim/gz-sim/issues/168 still seems to block using windows version for anything useful.

What is your use case? gz sim -s and gz sim -g should work fine for the usual simulation workflow.

TheCrazyT, 2024-12-23:
> > well my manual workaround was downloading:
> > https://raw.githubusercontent.com/gazebosim/gz-gui/d110e5b288e8b580b28c0a7256d0e1f0e0c402e2/src/cmd/cmdgui.rb.in
> > renamed it as "cmdgui9.rb" (placed in "c:\Users[YOUR USERNAME]\anaconda3\envs\gazebo\Library\lib\ruby\gz" folder) and modified the lines:
> > ```
> > LIBRARY_NAME = '../../../bin/gz-gui9.dll'
> > LIBRARY_VERSION = '9.0.0'
> > ```
> > 
> > 
> >     
> >       
> >     
> > 
> >       
> >     
> > 
> >     
> >   
> > not so shure what package should normaly do that. (maybe "libgz-gui9" ?)
> > **Update:**
> > Oh well, but I guess it won't be useful, since [gazebosim/gz-sim#168](https://github.com/gazebosim/gz-sim/issues/168) still seems to block using windows version for anything useful.
> 
> What is your use case? gz sim -s and gz sim -g should work fine for the usual simulation workflow.

Actually I didn't know you could use both (-g and -s) and thought only the "-s" will work. (I assumed that "-g" is the default)
But you are correct, that way it is working.