gazebosim/gz-gui #643: Plugins tutorial should be improved to be more new user friendly

azeey, 2024-09-16:
* **Test enviornment**
    ```text
    CPU: Ryzen 5 5600x, Ram: 32GB @ 3200 Mhz, GPU: RTX 3080
    Clean copy of Ubuntu 24.04 Humble
    ROS 2 Jazzy Jalisco full desktop
    ```
* **Summary**
     * In my humble opinion this tutorial may need to be presented in context of a more hands-on example. The current version was hard to follow, specially as a newcomer to Gazebo.
    * Add a note stating if built from source, ```GZ_GUI_PLUGIN_PATH``` can be set as follows
    ```bash
    cd ~/workspace/src/gz-gui
    export GZ_GUI_PLUGIN_PATH=${HOME}/workspace/src/gz-gui
    echo $GZ_GUI_PLUGIN_PATH
    gz gui -s ImageDisplay
    ```
    * Tutorial does not show how to set `GZ_GUI_PLUGIN_PATH`. Nor it states the user had to change direcotry into ```src/gz-gui``` directory before this command would work correctly
    

[image]

[image]

* **Issues**
    * Users needs to change directory to ```~/workspace/src/gz-gui```. Without this step, invoking any of example commands with ```gz gui -c``` prefix throws ```XMLError``` error.
    * The tutorial page has multiple ```\subpage``` references that leads to nowhere.
    * Commands for Topic Viewer, Topic Stats, Requesters and Responders did not work.

[image]

_Originally posted by @Mechazo11 in https://github.com/gazebosim/gazebo_test_cases/issues/1323#issuecomment-2333043686_

anshium, 2025-03-27:
Interested to take this up.

iche033, 2025-03-27:
sure, please go ahead!

kscottz, 2025-03-27:
@ahcorde, heads up.

kscottz, 2025-03-31:
@anshium How are things going here? Do you need some help?

anshium, 2025-03-31:
I am sorry, I was unwell and occupied with a few important things in the last few days. I would soon create a PR.

Apologies for the delay.

PR #671 (2025-04-25): feat: make plugins tutorial more new user friendly
# Fix

Fixes #643

## Summary
Makes the plugin tutorial more new user friendly.

I have made the plugin tutorial more conversational and explicit. I have also added more images to support the tutorial.

## Checklist
- [x] Signed all commits for DCO
- [N/A] Added tests
- [x] Updated documentation (as needed)
- [N/A] Updated migration guide (as needed)
- [N/A] Consider updating Python bindings (if the library has them)
- [x] `codecheck` passed (See [contributing](https://gazebosim.org/docs/all/contributing#contributing-code))
- [N/A ] All tests  [... cut]