gazebosim/gz-gui #659: OpenGL context 3.3+ not found unless requesting 3.3 version

tho-, 2025-03-11:
Hi,

Using gz-gui 9.0.0 / gz-rendering 9.0.0 / ogre2.3.3 all built from source on NetBSD 10.1, amd64/intel with graphics  hardware as reported by this glxinfo excerpt: 
```OpenGL vendor string: Intel Open Source Technology Center
OpenGL renderer string: Mesa DRI Intel(R) HD Graphics 4600 (HSW GT2)
OpenGL core profile version string: 4.5 (Core Profile) Mesa 21.3.7
   Max core profile version: 4.5
   Max compat profile version: 3.0
```

ogre2 fails to initialize its context with the message:
`OGRE EXCEPTION(3:RenderingAPIException): OpenGL 3.3 is not supported. Please update your graphics card drivers.`
and ogre2.log shows
```GL Version = 3.0.0.0
GL_VERSION = 3.0 Mesa 21.3.7
```
(it then segfaults later, by the way, but that's a different issue)

If I apply this small patch right after the QApplication constructor then it works perfectly fine:
```
--- src/Application.cc~	2024-09-27 01:38:17.000000000 +0200
+++ src/Application.cc	2025-03-10 01:16:16.297511872 +0100
@@ -96,6 +96,9 @@
 {
   gzdbg << "Initializing application." << std::endl;
 
+  QSurfaceFormat format;
+  format.setVersion(3, 3);
+  QSurfaceFormat::setDefaultFormat(format);
   this->setOrganizationName("Gazebo");
   this->setOrganizationDomain("gazebosim.org");
   this->setApplicationName("Gazebo GUI");
```

And in `.gz/redering/ogre2.log` there is the exepect message
```RenderSystem Name: OpenGL 3+ Rendering Subsystem
Driver Version: 4.5.0.0
```

I'm not sure wether this fix is correct in terms of genericity, e.g. if it should rather belong to gz-rendering/ogre2, or at least be protected by an 'OGRE2' ifdef or something. I tried various different tricks, patching in different places near or around ogre2 initialization but the above solution that was the only one that actually worked. 

Let me know if that makes sense to you,
Cheers
Anthony

iche033, 2025-03-11:
Instead of the above patch, can you take a look see if any of the steps for troubleshooting rendering issues works for you? https://gazebosim.org/docs/latest/troubleshooting/#unable-to-create-the-rendering-window

tho-, 2025-03-11:
> Instead of the above patch, can you take a look see if any of the
> steps for troubleshooting rendering issues works for you?
> https://gazebosim.org/docs/latest/troubleshooting/#unable-to-create-the-rendering-window

Sure, but none of them fix the issue. In particular:

* `update your computer’s OpenGL version`: Mesa is already the latest
  version, as well as the intel Xorg driver etc.
* `export LIBGL_DRI3_DISABLE=1` has no effect.
* `export LIBGL_ALWAYS_SOFTWARE=1` kinda works, of course without
  hardware acceleration it's not usable in practice.
* `export MESA_GL_VERSION_OVERRIDE=3.3` fixes the OGRE2 error, but
  then the rendering is completely buggy (weird textures, or garbage).
* `use Ogre 1` should also work (not tested), but then this would not
  fix the issue :)

Let me know if you need more feedback from my side.

iche033, 2025-03-13:
we actually had the surface format set to 3.3 before in https://github.com/gazebosim/gz-sim/pull/339, but that plugin was replaced by the [minimal_scene](https://github.com/gazebosim/gz-gui/tree/gz-gui9/src/plugins/minimal_scene) in gz-gui, which no longer has the surface format setting. I think we can add that back in here around here https://github.com/gazebosim/gz-gui/blob/db69c56f4be1bc8ee8427c283f6f4cb8799962da/src/plugins/minimal_scene/MinimalScene.cc#L1211

Just curious, could you see if `MESA_GL_VERSION_OVERRIDE=3.3FC` (with the `FC`) makes any difference?

tho-, 2025-03-13:
Indeed, patching like so also works:
```
--- src/plugins/minimal_scene/MinimalScene.cc~  2024-09-27 01:38:17.000000000 +0200
+++ src/plugins/minimal_scene/MinimalScene.cc   2025-03-13 10:21:14.079635782 +0100
@@ -1207,8 +1207,12 @@
       // and makeCurrent down below while setting up our own context.
       current->doneCurrent();
 
+      auto surfaceFormat = QSurfaceFormat();
+      surfaceFormat.setMajorVersion(3);
+      surfaceFormat.setMinorVersion(3);
+
       this->dataPtr->renderThread->SetContext(new QOpenGLContext());
-      this->dataPtr->renderThread->Context()->setFormat(current->format());
+      this->dataPtr->renderThread->Context()->setFormat(surfaceFormat);
       this->dataPtr->renderThread->Context()->setShareContext(current);
       this->dataPtr->renderThread->Context()->create();
 
```

> Just curious, could you see if MESA_GL_VERSION_OVERRIDE=3.3FC (with
> the FC) makes any difference?

Interestingly, with 3.3FC it also works fine.