gazebosim/gz-gui #661: UNIT_MainWindow_TEST fails: can't get the expected value

XINJIANGMO, 2025-03-13:
## Environment
* OS Version:  Ubuntu 24.04
* Source or binary build?  source gz-gui9

## Description
* Expected behavior:  test passed
* Actual behavior:  test failed

## Steps to reproduce
1. cd workspace/build/gz-gui9/bin
2. ./UNIT_MainWindow_TEST

## Output
```
[ RUN      ] MainWindowTest.ApplyConfig
 [debug] [Application.cc:97] [GUI] Initializing application.
 [debug] [Application.cc:178] [GUI] Qt using OpenGL graphics interface
 [debug] [Application.cc:685] [GUI] Create main window
 [warning] [Application.cc:933] [GUI] [QT] file:///usr/lib/x86_64-linux-gnu/qt5/qml/QtQuick/Dialogs/DefaultFileDialog.qml:413:17: QML ToolButton: Binding loop detected for property "implicitHeight"
 [warning] [Application.cc:933] [GUI] [QT] file:///usr/lib/x86_64-linux-gnu/qt5/qml/QtQuick/Dialogs/DefaultFileDialog.qml:309:21: QML Button: Binding loop detected for property "implicitHeight"
 [warning] [Application.cc:933] [GUI] [QT] file:///usr/lib/x86_64-linux-gnu/qt5/qml/QtQuick/Dialogs/DefaultFileDialog.qml:309:21: QML Button: Binding loop detected for property "implicitHeight"
 [warning] [Application.cc:933] [GUI] [QT] file:///usr/lib/x86_64-linux-gnu/qt5/qml/QtQuick/Dialogs/DefaultFileDialog.qml:413:17: QML ToolButton: Binding loop detected for property "implicitHeight"
 [warning] [Application.cc:933] [GUI] [QT] file:///usr/lib/x86_64-linux-gnu/qt5/qml/QtQuick/Dialogs/DefaultFileDialog.qml:309:21: QML Button: Binding loop detected for property "implicitHeight"
 [warning] [Application.cc:933] [GUI] [QT] file:///usr/lib/x86_64-linux-gnu/qt5/qml/QtQuick/Dialogs/DefaultFileDialog.qml:309:21: QML Button: Binding loop detected for property "implicitHeight"
 [warning] [Application.cc:933] [GUI] [QT] file:///usr/lib/x86_64-linux-gnu/qt5/qml/QtQuick/Controls/ToolBar.qml:146:9: QML QQuickItem*: Binding loop detected for property "layoutHeight"
 [warning] [Application.cc:933] [GUI] [QT] file:///usr/lib/x86_64-linux-gnu/qt5/qml/QtQuick/Controls/ToolBar.qml:146:9: QML QQuickItem*: Binding loop detected for property "layoutHeight"
 [warning] [Application.cc:933] [GUI] [QT] file:///usr/lib/x86_64-linux-gnu/qt5/qml/QtQuick/Dialogs/DefaultFileDialog.qml:413:17: QML ToolButton: Binding loop detected for property "implicitHeight"
[warning] [Application.cc:933] [GUI] [QT] file:///usr/lib/x86_64-linux-gnu/qt5/qml/QtQuick/Dialogs/DefaultFileDialog.qml:309:21: QML Button: Binding loop detected for property "implicitHeight"
 [warning] [Application.cc:933] [GUI] [QT] file:///usr/lib/x86_64-linux-gnu/qt5/qml/QtQuick/Dialogs/DefaultFileDialog.qml:309:21: QML Button: Binding loop detected for property "implicitHeight"
 [warning] [Application.cc:933] [GUI] [QT] file:///usr/lib/x86_64-linux-gnu/qt5/qml/QtQuick/Dialogs/DefaultFileDialog.qml:413:17: QML ToolButton: Binding loop detected for property "implicitHeight"
 [warning] [Application.cc:933] [GUI] [QT] file:///usr/lib/x86_64-linux-gnu/qt5/qml/QtQuick/Dialogs/DefaultFileDialog.qml:309:21: QML Button: Binding loop detected for property "implicitHeight"
 [warning] [Application.cc:933] [GUI] [QT] file:///usr/lib/x86_64-linux-gnu/qt5/qml/QtQuick/Dialogs/DefaultFileDialog.qml:309:21: QML Button: Binding loop detected for property "implicitHeight"
/home/momo/workspace/gazebo/src/gz-gui/src/MainWindow_TEST.cc:768: Failure
Expected equality of these values:
  c.width
    Which is: 300
  100
/home/momo/workspace/gazebo/src/gz-gui/src/MainWindow_TEST.cc:769: Failure
Expected equality of these values:
  c.height
    Which is: 300
  200
```

anshium, 2025-03-19:
I tested this out and have some observations:

1. The test always fails on a configurable tiling manager DE (I use Hyprland on Ubuntu) - they do not allow the ApplyConfig to be used properly unless we force the gui to use X11.
2. The test passed most of the times when I switched to GNOME.

# Screenshots:

## 1. On Hyprland

### Tile Config - 1
[image]

### Tile Config - 2

[image]

### Tile Config - 3

[image]

## 2. With GNOME

[image]

## However, running repeatedly on GNOME for 20-30 times did cause an odd failed case.

[image]

(This failure was more frequent with Wayland)

XINJIANGMO, 2025-03-20:
That's a good point!  I didn't use manager DE to test it, I used GNOME. But something different from you, I used  GNOME but failed every time!

exMachina316, 2025-03-31:
> That's a good point! I didn't use manager DE to test it, I used GNOME. But something different from you, I used GNOME but failed every time!

Yes it fails for me too on GNOME with wayland consistently (the 10 times I tried it).